nix-community/robotnix #3: Failure to build for google-walleye

samueldr, 2020-04-04:
I'm still looking into it, but as builds take quite some time, I wanted to first report in case you can see the issue from the logs.

```
2020-04-04 05:53:24 - add_img_to_target_files - INFO    : done.

#### build completed successfully (02:22:48 (hh:mm:ss)) ####

installing
cp: cannot stat '/build/out/target/product/generic/obj/PACKAGING/target_files_intermediates/aosp_walleye-target_files-2020.03.16.18.zip': No such file or directory
builder for '/nix/store/s8yn77mx4mw766hv0c7bfch9w3smsl9g-robotnix-aosp_walleye-2020.03.16.18.drv' failed with exit code 1
cannot build derivation '/nix/store/lm4hpkl5kd3kjmfjrx5jxsghzxlhlrpz-walleye-img-2020.03.16.18.zip.drv': 1 dependencies couldn't be built
error: build of '/nix/store/lm4hpkl5kd3kjmfjrx5jxsghzxlhlrpz-walleye-img-2020.03.16.18.zip.drv' failed
```

It seems to build successfully, and only fails at the very end, while trying to copy the files. This makes me think that `buildProduct` or `buildNumber` is somehow wrong.

danielfullmer, 2020-04-04:
This is another issue with the confusing naming for Pixel 2 (XL) devices in AOSP. The error occurs very early in the build process during the `choosecombo` command because I didn't properly include certain source directories for walleye:

```
Which product would you like? [aosp_arm] aosp_walleye
build/make/core/product_config.mk:223: error: Can not locate config makefile for product "aosp_walleye".
```

(I should probably see if I can force it to fail here if it can't find the product 

I just pushed 4deb9c57824f30f41030f111b6642cd602e7dd79 which should hopefully fix this. I started a build for walleye, and I'll update this issue when it finishes.

samueldr, 2020-04-04:
♥ restarting the build and validating. I assumed something related to walleye/taimen/wahoo/whatever naming :)

samueldr, 2020-04-05:
As expected, it now builds.

closed 2020-04-05 by samueldr

danielfullmer, 2020-04-05:
That's great to hear. I'd love to hear if it even boots on walleye!