nix-community/robotnix #4: Failure to boot with "init: Failed to mount required partitions early ..." (google-walleye)

samueldr, 2020-04-05:
```
dy: 5537653297, waiting time: [    7.280703] c5      1 md: Skipping autodetection of RAID arrays. (raid=autodetect will force)
[    7.280740] c5      1 device-mapper: init: attempting early device configuration.                      
[    7.288468] c5      1 device-mapper: init: adding target '0 5159992 verity 1 PARTUUID=00000016-0000-0000-0000-000000000000 PARTUUID=00000016-0000-0000-0000-000000000000 4096 4096 644999 644999 sha1 c1f328bce953
413028bfc51c9009ea868ae1f610 1bf48e3dd3f9d2616e6d3d6ae0c99ae9fbad59e2 10 restart_on_corruption ignore_zero_blocks use_fec_from_device PARTUUID=00000016-0000-0000-0000-000000000000 fec_roots 2 fec_blocks 650080 fec
_start 650080'                                                                                            
[    7.312197] c5      1 device-mapper: init: dm-0 is ready
[    7.337256] c5      1 EXT4-fs (dm-0): couldn't mount as ext3 due to feature incompatibilities
[    7.339597] c5      1 EXT4-fs (dm-0): couldn't mount as ext2 due to feature incompatibilities
[    7.348845] c4      1 EXT4-fs (dm-0): mounted filesystem without journal. Opts: (null)
[    7.356618] c4      1 VFS: Mounted root (ext4 filesystem) readonly on device 252:0.
[    7.366994] c4      1 Freeing unused kernel memory: 8640K
[    7.421595] c0     80 msm-dwc3 a800000.ssusb: dwc3_msm_resume: exiting lpm
[    7.426940] c0     80 msm-dwc3 a800000.ssusb: DWC3 exited from low power mode
[    7.457180] c7      1 init: init first stage started!
[    7.458729] c7      1 init: [libfs_mgr]ReadDefaultFstab(): failed to find device default fstab
[    7.461358] c7      1 init: Using Android DT directory /proc/device-tree/firmware/android/             
[    7.481352] c7      1 init: [libfs_mgr]superblock s_max_mnt_count:65535,/dev/block/platform/soc/1da4000.ufshc/by-name/persist
[    7.481443] c7      1 init: [libfs_mgr]Filesystem on /dev/block/platform/soc/1da4000.ufshc/by-name/persist was not cleanly shutdown; state flags: 0x1, incompat feature flags: 0x46
[    7.492165] c6      1 EXT4-fs (sdd3): Ignoring removed nomblk_io_submit option
[    7.508657] c6      1 EXT4-fs (sdd3): recovery complete                                
[    7.514732] c6      1 EXT4-fs (sdd3): mounted filesystem with ordered data mode. Opts: errors=remount-ro,nomblk_io_submit
[    7.519896] c6      1 init: [libfs_mgr]check_fs(): mount(/dev/block/platform/soc/1da4000.ufshc/by-name/persist,/persist,ext4)=0: Success
[    7.531654] c6      1 init: [libfs_mgr]check_fs(): unmount(/persist) succeeded
[    7.543356] c6      1 init: [libfs_mgr]Running /system/bin/e2fsck on /dev/block/sdd3
[    7.551397] c6      1 e2fsck: executing /system/bin/e2fsck failed: No such file or directory
[    7.551397] c6      1                                                                                  
[    7.558238] c6      1 e2fsck: e2fsck terminated by exit(255)
[    7.558238] c6      1                          
[    7.568812] c6      1 init: [libfs_mgr]e2fsck returned status 0xff00
[    7.578215] c6      1 EXT4-fs (sdd3): mounted filesystem with ordered data mode. Opts: barrier=1       
[    7.583338] c6      1 init: [libfs_mgr]__mount(source=/dev/block/platform/soc/1da4000.ufshc/by-name/persist,target=/persist,type=ext4)=0: Success
[    7.595558] c7      1 init: [libfs_avb]Returning avb_handle with status: Success
[    7.605226] c4      1 init: [libfs_avb]Hashtree descriptor not found: vendor
[    7.612493] c4      1 init: Failed to setup verity for '/vendor': No such file or directory
[    7.619515] c4      1 init: Failed to mount /vendor: No such file or directory                         
[    7.627652] c5      1 init: Failed to mount required partitions early ...
[    7.639590] c5      1 init: #00 pc 0000000000066f1c  /system/bin/init (android::init::InitFatalReboot()+92)
[    7.641756] c5      1 init: #01 pc 000000000009272c  /system/bin/init (android::init::InitAborter(char const*)+44)
[    7.651301] c5      1 init: #02 pc 000000000000c650  /system/lib64/libbase.so (android::base::LogMessage::~LogMessage()+608)
[    7.661713] c5      1 init: #03 pc 000000000003eecc  /system/bin/init (android::init::FirstStageMain(int, char**)+5292)
[    7.673088] c5      1 init: #04 pc 00000000000230f4  /system/bin/init (main+148)
[    7.683576] c5      1 init: #05 pc 000000000007d844  /system/lib64/bootstrap/libc.so (__libc_init+108)
[    7.691226] c5      1 init: Reboot ending, jumping to kernel
[    7.700400] c5    378 tas2557s 7-004d: tas2557_fw_ready:
[    7.706140] c5    378 tas2557s 7-004d: tas2557s_PG21_uCDSP.bin firmware is not loaded.
[    7.711447] c5     50 drv2624 6-005a: drv2624_firmware_load, ERROR!! firmware not found
[    7.719528] c5      1 mnh_pci 0000:01:00.0: ioummap pcie_config
[    7.727063] c5      1 mnh_pci 0000:01:00.0: ioummap config
[    7.732956] c5      1 mnh_pci 0000:01:00.0: ioummap ddr                                                
[    7.738764] c5      1 sd 0:0:0:3: [sdd] Synchronizing SCSI cache
[    7.745432] c6      1 sd 0:0:0:0: [sda] Synchronizing SCSI cache
[    7.750044] c5      1 mdss_fb_release_all: try to close unopened fb 1! from pid:1 name:init
[    7.755896] c5      1 mdss_fb_release_all: try to close unopened fb 0! from pid:1 name:init
[    7.764154] c5      1 nxp,ptn36241g soc:ptn36241g: ptn36241g_shutdown - shutdown ptn36241g
[    7.779060] c0      1 reboot: Restarting system with command 'bootloader'
[    7.780619] c0      1 Going down for restart now
[    7.788110] c0      1 
[    7.792144] c0      1 Restarting 1972-07-21 00:35:58.589314791 UTC
[    7.794327] c0      1 Restarting Linux version 4.4.200-g11566f2bfb0e (android-build@xpcd9.ams.corp.google.com) (Android (5484270 based on r353983c) clang version 9.0.3 (https://android
```

I wonder if I could have accidentally my vendor partition while playing about with Mobile NixOS.

I am waiting on a download of the [factory image from Google](https://developers.google.com/android/images#walleye) and will flash *only* the `vendor` partition from that zip.

danielfullmer, 2020-04-05:
`fastboot update -w <img.zip>` should be sufficient to write all the partitions. The image should include the vendor partition as well. In fact, the build process for robotnix downloads the factory image for your device, runs it through [android-prepare-vendor](https://github.com/anestisb/android-prepare-vendor), and includes that in the AOSP build.

samueldr, 2020-04-05:
I don't have a `vendor` partition in the .zip for walleye, but I did not build from the tippy-top of master, where I see there are more changes.

`configuration.nix`:
```
{
  ccache.enable = true;
  device = "walleye";
  flavor = "vanilla";
}
```

Built using:

```
nix-build --arg configuration ./configuration.nix -A img
```

List of files in the zip:

```
-rw-r--r-- 1 samuel users         14 Jan  1  2009 android-info.txt
-rw-r--r-- 1 samuel users   33554432 Jan  1  2009 boot.img
-rw-r--r-- 1 samuel users    8388608 Jan  1  2009 dtbo.img
-rw-r--r-- 1 samuel users 1098785264 Jan  1  2009 system.img
-rw-r--r-- 1 samuel users   72241432 Jan  1  2009 system_other.img
-rw-r--r-- 1 samuel users    4374940 Jan  1  2009 userdata.img
-rw-r--r-- 1 samuel users       4096 Jan  1  2009 vbmeta.img
```

danielfullmer, 2020-04-05:
:disappointed: Yes, this seems to be a real issue. The crosshatch builds that I frequently do seem to include a vendor.img. It looks like this will take some more debugging.

(The other changes I've made recently shouldn't affect this.)

samueldr, 2020-04-05:
Tried flashing the full factory image, that one boots, flashing on top, using `update -w` doesn't boot, same issue. It looks like it may not be an issue with the vendor partition.

danielfullmer, 2020-04-05:
I don't think AVB (android verified boot) would allow mounting google's vendor partition because it is not signed with the same keys as the rest of the system. We should still include vendor in our build since IIRC the dm-verity hashtree for `vendor` gets included in the `vbmeta.img` we build (or something like that). This message in the bootlog is probably the issue: `[libfs_avb]Hashtree descriptor not found: vendor`

samueldr, 2020-04-05:
Oh, this makes sense. I have not had the chance to actually learn about AVB, so this escaped me totally. All my previous personal devices didn't have AVB at all (I think), and my new one has no custom ROM scene, so I'm running the OEM ROM.

Tell me if there's anything I can do to help on that issue.

danielfullmer, 2020-04-05:
This should hopefully be fixed by: https://github.com/danielfullmer/robotnix/commit/b0f4ee7f7457709e1d0a580664c6e112571637cd

I've built an `img` for `{device = "walleye"; flavor = "vanilla";}` and verified that it has a `vendor.img` in the resulting zip using that latest commit on master: 3d611dae3f957d5e6919f8a12cde59b3fdde04d0

samueldr, 2020-04-05:
Confirmed working on my `google-walleye`. Thanks!

closed 2020-04-05 by samueldr