nix-community/robotnix #5: Undiagnosed boot failure for walleye; Signal 9 sent to zygote

samueldr, 2020-04-05:
[Related IRC discussion](https://logs.nix.samueldr.com/nixos-aarch64/2020-04-05#1586125101-1586127072;)

 * https://gist.github.com/samueldr/f720ce000ee2a4f3e2fcbc48accd0212

It is unknown what the issue is. What can be observed is that the boot stays at the "android" logo.

danielfullmer, 2020-04-06:
These lines stick out to me:
```
[   26.187003] c7    779 binder: release 1206:2993 transaction 5752 out, still active
[   26.187033] c4    868 binder_alloc: 1206: binder_alloc_buf, no vma
[   26.187037] c4    868 binder: 868:868 transaction failed 29189/-3, size 0-0 line 3238
[   26.197885] c4      1 init: Service 'vendor.folio_daemon' (pid 868) exited with status 1
[   26.197895] c4      1 init: Sending signal 9 to service 'vendor.folio_daemon' (pid 868) process group...
[   26.198039] c4      1 libprocessgroup: Successfully killed process cgroup uid 1000 pid 868 in 0ms
[   26.199032] c4      1 init: starting service 'vendor.folio_daemon'...
[   26.200266] c5      1 init: Service 'zygote' (pid 811) received signal 9
[   26.200277] c5      1 init: Sending signal 9 to service 'zygote' (pid 811) process group...
```
In particular, I notice that folio_daemon exits with status 1. The code is here: https://android.googlesource.com/device/google/wahoo/+/refs/heads/master/folio_daemon/main.cpp
I wonder if it's exiting due to failure or if it's being killed ealier.  There are a bunch of ALOG instructions that adb, which might give us more information.

So I recommend trying a build with `variant="eng";` so that we can get adb logs as well.

samueldr, 2020-04-06:
It stays stuck on the "android" logo in `"eng"` mode, and adb is not able to see the device. I left it sit for more than 10 minutes the first time around.

Though, here's a fresh boot's log:

 * https://gist.github.com/samueldr/df2c8a683de2e758df8775dba9e598a5

danielfullmer, 2020-04-06:
So the only other thing I can think of at the moment to try is to comment out these lines:
https://github.com/danielfullmer/robotnix/blob/3d611dae3f957d5e6919f8a12cde59b3fdde04d0/pkgs/android-prepare-vendor/android-prepare-vendor.nix#L59-L63

It's a hack that I recently added for grapheneos.  (I don't have a convenient way of applying this conditionally at the moment,)

When building grapheneos _without_ this hack, IIRC it has a similar failure, with this line:
`binder_alloc_buf, no vma`
and then `zygote` crashing shortly afterward.

So maybe building vanilla _with_ this hack somehow causes it? I tried this yesterday with marlin but couldn't reproduce the issue. This could also be because marlin doesn't receive monthly updates from google anymore, so it is working off older AOSP source.

danielfullmer, 2020-04-07:
I've just made this patch conditional on using GrapheneOS in this commit: https://github.com/danielfullmer/robotnix/commit/b2c36d8032721c31a981e674b3e69debbe66080a

(Also now in master is the April 2020 update for vanilla AOSP, as well as a fix for `walleye` not getting the default usability improvements that other devices get for vanilla)

samueldr, 2020-04-07:
Disabling the patch manually didn't seem to work, but I didn't investigate that build. I already had a fresh proper build from the top of master (d0348d143b5226b3bb9084970aedae3fdf4a4b42) done, so I tried it directly:

```
{
  ccache.enable = true;
  device = "walleye";
  flavor = "vanilla";
}
```

And it works. I'm not sure whether it's removing that hack, or the update that did the trick, but at least it's working now. Now I can tinker a bit more, when time permits. Though, you now know it's been verified booting on `walleye` :).

closed 2020-04-07 by samueldr

danielfullmer, 2020-04-07:
@samueldr Thanks for the feedback! It's great to know it works on devices other than mine!

danielfullmer, 2020-04-09:
Just an added note here in case anyone comes across a similar issue:

I think this was actually fixed by d0348d143b5226b3bb9084970aedae3fdf4a4b42 which enables some usability improvements for `walleye`, including using a modern chromium for webview, getting rid of the old browser, etc.

I was able to reproduce this while experimenting with building generic `aosp_x86` for use with the emulator. It failed in a similar way until I enabled these usability improvements.  I was surprised these settings fixed the problem. Maybe the included prebuilt chromium is just broken?