nix-community/robotnix #6: google-walleye: graphene flavour has bad CRC for system.img / unexpected size

samueldr, 2020-04-06:
Using `fastboot update`

```
extracting android-info.txt (0 MB) to RAM...
--------------------------------------------       
Bootloader Version...: mw8998-002.0081.00
Baseband Version.....: g8998-00020-1912122233
Serial Number........: HT79F1A02235
--------------------------------------------       
Checking product                                   OKAY [  0.000s]
Checking version-bootloader                        OKAY [  0.000s]
Checking version-baseband                          OKAY [  0.000s]
extracting boot.img (32 MB) to disk... took 0.163s
archive does not contain 'boot.sig'
archive does not contain 'boot_other.img'
extracting dtbo.img (8 MB) to disk... took 0.027s
archive does not contain 'dtbo.sig'
archive does not contain 'dt.img'
archive does not contain 'odm.img'
archive does not contain 'product.img'
archive does not contain 'product-services.img'
archive does not contain 'recovery.img'
archive does not contain 'super.img'
extracting system.img (1498 MB) to disk...W/ziparchive(24987): Zip: Unexpected size 1571434824 (declared) vs 1571434829 (actual)
fastboot: error: 
failed to extract 'system.img': I/O error
```

Using `unzip`:

```
Archive:  /nix/store/pxd0yijmnwrhchlfgbbq1xdivfrg4qfd-walleye-img-2020.03.16.18.zip
  inflating: android-info.txt        
  inflating: boot.img                
  inflating: dtbo.img                
  inflating: system.img               bad CRC 177a6df6  (should be b7fc3d8d)
  inflating: userdata.img            
  inflating: vbmeta.img              
  inflating: vendor.img              
```

* * *

Just for kicks, I tried flashing it manually

```
 $ fastboot flash system system.img 
Invalid sparse file format at unknown block FFFF at 1338949828
Invalid sparse file format at unknown block 0C00 at 1338949840
Invalid sparse file format at unknown block 0C00 at 1338949852
Invalid sparse file format at unknown block 65CF at 1338949864
Invalid sparse file format at unknown block 0807 at 1338949876
Invalid sparse file format at unknown block 0000 at 1338949888
Invalid sparse file format at unknown block 0000 at 1338949900
Invalid sparse file format at unknown block 0000 at 1338949912
Invalid sparse file format at unknown block 6F67 at 1338949924
Invalid sparse file format at unknown block 6870 at 1338949936
Invalid sparse file format at unknown block 642F at 1338949948
Invalid sparse file format at unknown block 6D75 at 1338949960
error: write_sparse_skip_chunk: don't care size 1035841869 is not a multiple of the block size 4096
error: write_sparse_skip_chunk: don't care size 500257101 is not a multiple of the block size 4096
Sending sparse 'system_a' 1/3 (522948 KB)          error: write_sparse_skip_chunk: don't care size 1035841869 is not a multiple of the block size 4096
error: write_sparse_skip_chunk: don't care size 1035841869 is not a multiple of the block size 4096
OKAY [ 13.096s]
Writing sparse 'system_a' 1/3                      OKAY [  0.000s]
Sending sparse 'system_a' 2/3 (522968 KB)          error: write_sparse_skip_chunk: don't care size 500257101 is not a multiple of the block size 4096
error: write_sparse_skip_chunk: don't care size 500257101 is not a multiple of the block size 4096
FAILED (remote: 'buffer overreads occured due to invalid sparse header')
Finished. Total time: 30.340s
```

* * *

This happened at revision `3d611dae3f957d5e6919f8a12cde59b3fdde04d0`. This can be verified without the device since `unzip` is failing. I wonder if it could be transient failure that shouldn't happen that caused corruption.

danielfullmer, 2020-04-06:
Hmm. This zip is written in the final step, so maybe try `--check` to see if you get a different output? Mine was reproducible and didn't have this problem.

I'm trying to reproduce the output hash you get:
`/nix/store/pxd0yijmnwrhchlfgbbq1xdivfrg4qfd-walleye-img-2020.03.16.18.zip`

But using `--arg configuration {device="walleye"; flavor="vanilla"; ccache.enable=true;}'`:
`/nix/store/gnn5zd0gam8192aa4bvh5k6s70isnvr7-walleye-img-2020.03.16.18.zip`

And with `--arg configuration {device="walleye"; flavor="vanilla"; ccache.enable=false;}'`:
`/nix/store/11c5b2gc1nnmwsg7bbdbipway028ip5n-walleye-img-2020.03.16.18.zip`

Were you using a different configuration? Maybe there are some drv-level impurities somewhere....
This is against 3d611dae3f957d5e6919f8a12cde59b3fdde04d0 as well.

samueldr, 2020-04-06:
`flavor = "graphene"`, I should have left the configuration in the issue.

```
{
  ccache.enable = true;
  device = "walleye";
  source.buildNumber = "QQ2A.200305.002";
  buildNumber = "2020.03.16.18";
  flavor = "grapheneos";
}
```

And oops I see I left buildNumber from trying `pixel` previously... I'm rebuilding without those.

danielfullmer, 2020-04-06:
OK. I got the same output hash, but my final zip doesn't have this problem:
```
Archive:  /nix/store/pxd0yijmnwrhchlfgbbq1xdivfrg4qfd-walleye-img-2020.03.16.18.zip
  inflating: android-info.txt        
  inflating: boot.img                
  inflating: dtbo.img                
  inflating: system.img              
  inflating: userdata.img            
  inflating: vbmeta.img              
  inflating: vendor.img              
```

If your rebuild has the same issue then I guess we have to consider things like underlying filesystem you're using, memory/hardware errors, etc.  If you do have the same problem, could you can post the zip somewhere and I can try to diffoscope the output. Also useful to compare would be the intermediate build products which you can get by using `-A build.android` instead of `-A img`

samueldr, 2020-04-07:
After a rebuild, without the `buildNumber` configs, 

```
{
  ccache.enable = true;
  device = "walleye";
  flavor = "grapheneos";
}
```

no broken zip, and It boots. I'll chalk it up to bad luck and corruption.

closed 2020-04-07 by samueldr