nix-community/robotnix #7: LineageOS support

Atemu, 2020-05-05:
It's the biggest and most popular FOSS Android fork, it would be great if we could build it using Nix!

I'm not too familiar with the details but it should be rather similar to AOSP, have you experimented with it at all?

It does not officially support your crosshatch unfortunately but there's official LOS17.1 (Android 10) for your marlin.

samueldr, 2020-05-05:
I coincidentally was looking into that this week-end, and have had *some* success in getting a LinageOS ROM built using `robotnix`.

I'm preparing a Draft PR with a list of issues I've faced, and some notes about how to handle this.

I think the *main* issue LineageOS will face is that there is no proper "tagged release", this means that we cannot use the same repository revision as another LineageOS build has. So what `robotnix` can target is pretty limited to "only whatever is on the tip of the dependencies at the moment we generate the repo manifest".

Though, this **does** mean that a `robotnix`-built ROM *is* possibly reproducible, what won't be is getting the same exact deps as a nightly LineageOS build.

danielfullmer, 2020-05-05:
Looking briefly at the [build instructions for marlin](https://wiki.lineageos.org/devices/marlin/build), it doesn't look terribly different from what I'm currently doing.  My only preference would be that it is better to extract the proprietary blobs from an `img/ota` provided from the vendor, instead of relying on external repositories like [TheMuppets](https://github.com/TheMuppets), if possible.

@samueldr If you want, feel free to send a WIP PR--even if it doesn't currently work, it might still get me started experimenting lineageos marlin a little easier.

samueldr, 2020-05-05:
See #8.

danielfullmer, 2020-05-06:
Based partially on @samueldr 's PR I was able to get a working lineageos build running on `marlin`. It ended up being much easier than I expected, with no serious workarounds required. With some luck, we could add the "lineageos" flavor and all the `android-17.1` devices it supports. (Older versions are possible as well, but let's start here)

closed 2020-08-24 by Atemu