nix-community/robotnix #23: update microG GmsCore to v0.2.11.202414

petabyteboy, 2020-07-27:
Hi there,

I tried to do this update myself by changing the version number and sha256 hash, but the resulting image didn't boot. Maybe you have an idea on how to debug this issue or how to update microG :slightly_smiling_face: 

Thanks in advance

danielfullmer, 2020-07-27:
Thanks for reporting! I just reverted the commit I made upgrading MicroG. I didn't test well enough before commiting. :frowning:  I just signed the apk and `adb install`-ed it on a device which already had the older version of microg working. That seems to be working fine, and reboots properly as well.

However, I can now reproduce your issue in the emulator (`nix-build ... -A build.emulator`) for ffa20da0d22f9988f3e4dc9b4ab2f0b67854f607:
```nix
{ device="x86"; flavor="vanilla"; }
```
works fine, but
```nix
{ device="x86"; flavor="vanilla"; microg.enable=true; }
```
fails to boot.

Attached is the serial output from the emulator. I'll try to take a closer look at this later tonight.
[microg.txt](https://github.com/danielfullmer/robotnix/files/4984274/microg.txt)

danielfullmer, 2020-07-30:
This issue should be fixed by 27cef20b35edda3eb9b6e9a6866e824994911024

Starting the emulator with the `-logcat` option in addition to `-show-kernel` gives better output for debugging. Here's the relevant output:
```
...
07-30 04:22:18.530  2335  2335 W PackageManager: Privileged permission android.permission.UPDATE_APP_OPS_STATS for package com.google.android.gms - not in privapp-permissions whitelist
07-30 04:22:18.530  2335  2335 I PackageManager: Un-granting permission android.permission.UPDATE_APP_OPS_STATS from package com.google.android.gms (protectionLevel=274 flags=0x20883e45)
...
07-30 04:22:18.533  2335  2335 E System  : ******************************************
07-30 04:22:18.533  2335  2335 E System  : ************ Failure starting system services
07-30 04:22:18.533  2335  2335 E System  : java.lang.IllegalStateException: Signature|privileged permissions not in privapp-permissions whitelist: {com.google.android.gms: android.permission.UPDATE_APP_OPS_STATS}
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.pm.permission.PermissionManagerService.systemReady(PermissionManagerService.java:2970)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.pm.permission.PermissionManagerService.access$100(PermissionManagerService.java:122)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.pm.permission.PermissionManagerService$PermissionManagerServiceInternalImpl.systemReady(PermissionManagerService.java:3031)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.pm.PackageManagerService.systemReady(PackageManagerService.java:21809)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.SystemServer.startOtherServices(SystemServer.java:1993)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.SystemServer.run(SystemServer.java:513)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.server.SystemServer.main(SystemServer.java:350)
07-30 04:22:18.533  2335  2335 E System  : 	at java.lang.reflect.Method.invoke(Native Method)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
07-30 04:22:18.533  2335  2335 E System  : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:908)
07-30 04:22:18.533  2335  2335 D SystemServerTiming: MakePackageManagerServiceReady took to complete: 303ms
07-30 04:22:18.533  2335  2335 E Zygote  : System zygote died with exception
07-30 04:22:18.533  2335  2335 E Zygote  : java.lang.IllegalStateException: Signature|privileged permissions not in privapp-permissions whitelist: {com.google.android.gms: android.permission.UPDATE_APP_OPS_STATS}
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.pm.permission.PermissionManagerService.systemReady(PermissionManagerService.java:2970)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.pm.permission.PermissionManagerService.access$100(PermissionManagerService.java:122)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.pm.permission.PermissionManagerService$PermissionManagerServiceInternalImpl.systemReady(PermissionManagerService.java:3031)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.pm.PackageManagerService.systemReady(PackageManagerService.java:21809)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.SystemServer.startOtherServices(SystemServer.java:1993)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.SystemServer.run(SystemServer.java:513)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.server.SystemServer.main(SystemServer.java:350)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at java.lang.reflect.Method.invoke(Native Method)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
07-30 04:22:18.533  2335  2335 E Zygote  : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:908)
```

We just needed to add `UPDATE_APP_OPS_STATS` to the `privappPermissions` in the microg module

closed 2020-07-30 by danielfullmer