nix-community/robotnix #25: Incremental / shared builds

danielfullmer, 2020-07-30:
Robotnix currently has a single large derivation called `build.android` which executes the full android build process. This is the step that takes multiple hours.  Most changes to the robotnix configuration will require a full rebuild. `ccache` is available to save some time with certain compilation steps, but it is incomplete (doesn't apply to java, for example), and deprecated within the AOSP project as of Android 10.

It would be preferable if small changes to the robotnix configuration could be built much more quickly by sharing some common intermediate build products.  Adding or removing packages like webview / microg should not require a full recompilation in most cases. One additional possibility is to share build products between multiple devices which share a common architecture. For example, first build a generic `aosp_arm64`, and somehow use some of the intermediate products to build `aosp_crosshatch`.

In theory, any change in configuration (device, etc) could produce intermediate build artifacts that are not appropriate for another configuration. The build dependencies are largely calculated by `soong` in android, which walks the entire android source tree and produces a file for ninja to build.

I currently see a few possibilities:

**Fine-grained incrementalism:**
Using perhaps, [ninja2nix](https://github.com/awakesecurity/ninja2nix), and/or recursive nix (see: [ghc-nix](https://github.com/ocharles/ghc-nix), [nix-ccache](https://github.com/edolstra/nix-ccache))
We could possibly integrate with / patch the RBE or GOMA support already in `soong`. However, the overhead involved with having a derivation for each of the ~11000 build steps would likely overwhelm any savings we could get from incremental / shared builds. The `out/build-aosp_x86.ninja` I have is 314MB by itself.

**Course-grained incrementalism:**
This might involve finding more natural boundaries between the dependencies, and modifying `soong` to produce and build derivations which contain a larger number of steps. This would likely require more extensive modifications to the android build system, which is both more development work and potentially a maintenance issue.

A final potential issue with this is that the intermediate build products can be very large: My `out/soong/.intermediates` is 14.2 GB. `out/target/common` is 1.8GB.

danielfullmer, 2020-07-31:
I've started a new experimental project: `blueprint2nix`, which may be useful here.  The idea is to convert `Android.bp` blueprint files into `.nix` files. Given the input file:
```
// Copyright 2017 The Android Open Source Project

//##########################################################################
// Build e2fsdroid

cc_binary {
    name: "e2fsdroid",
    host_supported: true,
    recovery_available: true,
    defaults: ["e2fsprogs-defaults"],

    srcs: [
        "e2fsdroid.c",
        "block_range.c",
        "fsmap.c",
        "block_list.c",
        "base_fs.c",
        "perms.c",
        "basefs_allocator.c",
    ],
    target: {
        host: {
            static_libs: [
                "libext2_com_err",
                "libext2_misc",
                "libext2fs",
                "libsparse",
                "libz",
                "libcutils",
                "libbase",
                "libselinux",
                "libcrypto",
                "liblog",
            ],
            sanitize: {
                address: false,  // http://b/68387795 - heap overflow in e2fsdroid
            },
        },
        android: {
            shared_libs: [
                "libext2fs",
                "libext2_com_err",
                "libext2_misc",
                "libcutils",
                "libbase",
                "libselinux",
                "libcrypto",
            ],
        },
    },
    stl: "libc++_static",
}

//##########################################################################
// Build ext2simg

cc_binary {
    name: "ext2simg",
    host_supported: true,
    defaults: ["e2fsprogs-defaults"],

    srcs: ["ext2simg.c"],
    shared_libs: [
        "libext2fs",
        "libext2_com_err",
        "libsparse",
        "libz",
    ],
}
```
`blueprint2nix` will produce this output:
```nix
{ cc_binary }:
let

#  Copyright 2017 The Android Open Source Project

# ##########################################################################
#  Build e2fsdroid

e2fsdroid = cc_binary {
    name = "e2fsdroid";
    host_supported = true;
    recovery_available = true;
    defaults = ["e2fsprogs-defaults"];

    srcs = [
        "e2fsdroid.c"
        "block_range.c"
        "fsmap.c"
        "block_list.c"
        "base_fs.c"
        "perms.c"
        "basefs_allocator.c"
    ];
    target = {
        host = {
            static_libs = [
                "libext2_com_err"
                "libext2_misc"
                "libext2fs"
                "libsparse"
                "libz"
                "libcutils"
                "libbase"
                "libselinux"
                "libcrypto"
                "liblog"
            ];
            sanitize = {
                address = false; #  http://b/68387795 - heap overflow in e2fsdroid
            };
        };
        android = {
            shared_libs = [
                "libext2fs"
                "libext2_com_err"
                "libext2_misc"
                "libcutils"
                "libbase"
                "libselinux"
                "libcrypto"
            ];
        };
    };
    stl = "libc++_static";
};

# ##########################################################################
#  Build ext2simg

ext2simg = cc_binary {
    name = "ext2simg";
    host_supported = true;
    defaults = ["e2fsprogs-defaults"];

    srcs = ["ext2simg.c"];
    shared_libs = [
        "libext2fs"
        "libext2_com_err"
        "libsparse"
        "libz"
    ];
};

in { inherit e2fsdroid ext2simg; }
```

This was done with relatively simple modifications to the upstream `printer.go` file to support outputting `nix` instead of `blueprint` files.  To make these `.nix` files buildable, we would need to implement `nix` versions of the `soong` modules used in the blueprint files. (`cc_binary` is a soong module.) There are a large number of [soong modules](https://ci.android.com/builds/latest/branches/aosp-build-tools/targets/linux/view/soong_build.html) (and module options) implemented in go. At the moment, it's unclear if `nix` would be able to handle all the potential complexity.  I think starting small with relatively self-contained host libraries and binaries might be a good place to start. With some luck, it might be possible to cleanly build `adb` and `fastboot` using the nix toolchain (see https://github.com/NixOS/nixpkgs/issues/75603)

(If this experiment is successful, it might point to how a similar approach might work for a `bazel2nix`)

danielfullmer, 2020-08-03:
Success! We can now build `fastboot` using using [blueprint2nix](https://github.com/danielfullmer/blueprint2nix) + [soongnix](https://github.com/danielfullmer/soongnix). Try:
```
nix-build "https://github.com/danielfullmer/soongnix/archive/7a527ceb930ad5de4426b9cfd50384f76fa0c220.tar.gz" -A fastboot
./result/bin/fastboot --help
```

This should download and build only the parts of AOSP that are necessary for fastboot.  This is in contrast to the way it could currently be done in `robotnix`, which would require checking out the entire AOSP source tree.

[soongnix](https://github.com/danielfullmer/soongnix) is the partial Nix implementation of the soong modules. Currently, I've just (hackishly) implemented enough parts of the `cc_*` modules to compile a statically linked fastboot.  (Some other tools work too, like `e2fsck`. Others like `adb` still need more work)

In its current state, each library and executable is a derivation.   It shouldn't be difficult to make the compile step of each `.c` file into its own derivation, however, I'm not certain that will scale to the very large AOSP repo.

(The `out/` subdir of the `soongnix` repo contains the `.nix` files generated by [blueprint2nix](https://github.com/danielfullmer/blueprint2nix))

danielfullmer, 2021-11-22:
Some recent promising ideas:

In theory, we'd like to be able to put all the user-specific customizations into the product partition, have system be generic, and vendor be device-specific.  Then, build these components in separate derivations and somehow merge them into a final image. I'd need to check, but with Android 12 I'd guess AOSP builds for different devices are probably now close to producing (nearly) identical system images.

So the question is then how to merge these separately-built components.   Under `build/make/tools/releasetools` there is a `merge_target_files.py` with the following description:
> This script merges two partial target files packages.
> One input package contains framework files, and the other contains vendor files.
> This script produces a complete, merged target files package:

This looks like its intended to merge target files which contain system/product files with ones that contain vendor files.  It might be possible to adapt to merging 3 target-files containing system, product, and vendor.

If this works, we could probably save a users a lot of compile time by pushing the generic system target files up to cachix. We can't redistribute vendor (target) files, but I doubt the compile time would be high since lots of those are prebuilt anyway.  It might even be possible to distribute precompiled cuttlefish vendor target files that could be combined with the user's customized product target files. This would make testing much more convenient.

Atemu, 2021-11-23:
Something I've thought of that might work:  
Couldn't we manually write plain-old Nix drvs for some parts of the AOSP (i.e. big core libraries) and coerce the AOSP build system into taking the build output of those as its intermediate instead?

The rest would be built as always but we would cache as many parts as we can manage to maintain.

No idea how feasible it is but I wanted to get the idea out there.

vroad, 2022-04-16:
I developed a new feature of robotnix for testing waydroid image changes locally with `debugEnterEnv` script.

https://discourse.nixos.org/t/im-working-on-new-features-of-robotnix-which-makes-testing-android-os-changes-easier/18507

While this doesn't solve the problem with regular robotnix build, it still allows you to edit source files and test changes without losing intermediate build artifacts.

theoparis, 2024-09-27:
> Success! We can now build `fastboot` using using [blueprint2nix](https://github.com/danielfullmer/blueprint2nix) + [soongnix](https://github.com/danielfullmer/soongnix). Try:
> 
> ```
> nix-build "https://github.com/danielfullmer/soongnix/archive/7a527ceb930ad5de4426b9cfd50384f76fa0c220.tar.gz" -A fastboot
> ./result/bin/fastboot --help
> ```
> 
> This should download and build only the parts of AOSP that are necessary for fastboot. This is in contrast to the way it could currently be done in `robotnix`, which would require checking out the entire AOSP source tree.
> 
> [soongnix](https://github.com/danielfullmer/soongnix) is the partial Nix implementation of the soong modules. Currently, I've just (hackishly) implemented enough parts of the `cc_*` modules to compile a statically linked fastboot. (Some other tools work too, like `e2fsck`. Others like `adb` still need more work)
> 
> In its current state, each library and executable is a derivation. It wouldn't be difficult to make the compile step of each `.c` file into its own derivation, however, I'm not certain that will scale to the very large AOSP repo.
> 
> (The `out/` subdir of the `soongnix` repo contains the `.nix` files generated by [blueprint2nix](https://github.com/danielfullmer/blueprint2nix))

I couldn't find any contact information on your github profile so I'm asking you here: Can you tell me how you generated the `repo-android-10.0.0_r40.json` file? I want to update it to Android main/15. I already figured out how to port blueprint2nix to work with the latest version of blueprint from android.googlesource.com.

Atemu, 2024-09-27:
`scripts/mk_repo_file.py`

neobrain, 2025-12-06:
Would it be feasible to integrate soongnix into robotnix and perform this transition incrementally, so that at least individual modules could be cached/reused from the nix store across builds? Or would even the work to get tangible benefits out of this be too much for an individual?