nix-community/robotnix #26: LineageOS on Bacon fails to find vendor makefile

Atemu, 2020-08-26:
```
Which product would you like? [aosp_arm] lineage_bacon
device/oppo/common/common.mk:25: error: _nic.PRODUCTS.[[device/oneplus/bacon/lineage_bacon.mk]]: "vendor/oppo/msm8974-common/msm8974-common-vendor.mk" does not exist.
07:44:55 dumpvars failed with: exit status 1
** Not a valid product: lineage_bacon
```

It seems like Bacon's vendor repo is vendor_oppo instead of vendor_oneplus and is therefore not included in the build. Furthe digging shows that we don't even fetch vendor_oppo in vendor-dirs.json.

Atemu, 2020-08-26:
Figured it out, bacon needs vendor_oppo in addition to vendor_oneplus, PR incoming.

danielfullmer, 2020-08-26:
I wonder what the correct programmatic way to pull in these vendor directories is.  Currently we just use the `oem` attribute from hudson to determine which `vendor/<oem>` repo use. Maybe a better heuristic is include the corresponding `vendor/<oem>` repo any time have a `device/<oem>/...` repo in the build.  Since bacon already includes `device/oppo/msm8974-common` this would bring in `vendor/oppo` as well.

Hopefully soon I'll try to run the whole `build.checkAndroid` process I did a few months ago to determine if we have issues with any other devices as well.

closed 2020-08-27 by danielfullmer

danielfullmer, 2020-08-27:
So originally I didn't want to make a special case for the `bacon` device. However, I checked all the other devices and it appears that this is the only `bacon` and `shamu` have this particular problem.  I still wonder if there is a "correct" way to determine what vendor directories to include.

Atemu, 2020-08-27:
I don't think there is. Officially, the vendor sources are supposed to be pulled from the device manually but nobody does that I think. TheMuppets is a very inofficial thing.

hmenke, 2020-12-21:
Sorry for hijacking this thread.  I also have an old bacon that I wanted to install Robotnix.  So I downloaded Lineage Recovery from https://download.lineageos.org/bacon and used that to sideload my own build.  However, when I use my own keys to sign the build it doesn't boot.  It just stays for 5 minutes at the very first vendor screen and then reboots into recovery saying “Cannot load Android system. Your data may be corrupt. [bla bla bla]”. Signing with test keys it boots without issues.  Do you have any advice?

danielfullmer, 2020-12-21:
@hmenke Try using the `bootImg` attribute from robotnix with a configuration that includes your keys, instead of using the upstream recovery image. I just realized that I haven't disabled some AVB/dm-verity signing code in robotnix for LineageOS devices--even though support for it is likely already partially disabled by upstream LineageOS changes.  (Note that `avbFlags` are still getting included in `modules/signing.nix`.)  Alternatively, you could try setting `signing.avb.enable = false;` and sideloading an OTA produced with that option using the original LineageOS recovery.  I would be interested if either (or both) of these alternatives work.

hmenke, 2020-12-21:
I've already tried `signing.avb.enable = false;` which had no effect on the derivation output.  I'll try building the `bootImg` and flash the Robotnix recovery.

danielfullmer, 2020-12-21:
@hmenke Ah, yes, I forgot that `signing.avb.enable` defaults to `false`....

hmenke, 2020-12-21:
Unfortunately I've rotated the keys in the meantime so I have to rebuild.  Actually I don't even know what kind of signing the bacon supports (maybe dm-verity?) and it is extremely hard to find any documentation on this.  Most search queries just bring up “non-technical” results, i.e. regular users who have random problems flashing official Lineage OS images.  Generally Android related queries seem to always bring up only “light” topics.

hmenke, 2020-12-21:
I've built the `img` and the `otaDir` attributes and extracted `recovery.img` from the `img` output.  Then I flashed `recovery.img` and sideloaded the `otaDir` output (`boot` is flashed as part of the OTA).  This looked good because this time there was no complaint about failed signature verification.  Unfortunately, I still cannot boot.  However, this time I pulled all the boot logs (Lineage Recovery is ADB enabled).  I can send them to you via email if you are interested.

danielfullmer, 2020-12-22:
@hmenke Sure, feel free to follow up via email.  I may have inadvertently made the invalid assumption that all supported LineageOS devices had "recovery as boot".  If that's the case, we ought to fix that, and correct the documentation.

hmenke, 2020-12-29:
@Atemu Since you also have a bacon: Do you use unsigned or signed builds of Robotnix?  When I use the self-signed images I cannot boot and wanted to ask whether you have had a similar experience.

Atemu, 2020-12-30:
I haven't bothered with signing yet as I don't see much of a security benefit in it, sorry.

One thing that could be looked into with (seemingly) Bacon-specific issues is that it's an armv7-only device but I don't think that's an issue here.

hmenke, 2020-12-30:
@Atemu That's fine. Thank you for following up, anyway.  Indeed, I am aware that bacon doesn't support AVB, but I would have assumed that the signature is just ignored.

PR #27 (2020-08-27): Bacon quirk workaround
Fixes #26