nix-community/robotnix #28: LineageOS builds failing for certain devices

danielfullmer, 2020-08-27:
I built the `build.checkAndroid` target for all `lineageos-17.1` devices. Here is a checklist with the failing devices, along with the error message I encountered.

- [ ]  a3xelte, a5xelte, a7xelte, j7elte, s5neolte
  `# mkdir: cannot create directory 'hardware/samsung/slsi': Permission denied`
- [ ]  jasmine_sprout
  `build/make/core/board_config.mk:125: error: No config file found for TARGET_DEVICE jasmine_sprout.`
- [ ]  z2_plus
  `build/make/target/product/telephony_vendor.mk:24: error: _nic.PRODUCTS.[[device/zuk/z2/plus/lineage_z2_plus.mk]]: "device/zuk/z2_plus/device.mk" does not exist.`

Atemu, 2020-08-30:
Could the last one be due to us replacing _s with /s?

danielfullmer, 2020-08-30:
@Atemu Good eye. That's probably what's causing the issue with `jasmine_sprout` as well.

closed 2020-09-05 by danielfullmer

Atemu, 2020-09-05:
Looks like the first one needed to be `samsung_slsi`, so that got fixed too. Great work!

PR #33 (2020-09-05): Refactor LineageOS updater scripts
Instead of relying on the `device_deps.json` from LineageOS's `hudson` being up-to-date, we get this information ourselves from the `lineage.dependencies` files in the individual repos. When including source dirs in the robotnix build, we follow this dependency mapping to include any required directories.

Additionally, the `device-metadata.json` fetcher has been extracted into its own script to easily allow a user to modify the default set of devices to be fetched by `update-device-dirs.py`

Closes #28  (All 110 supported devices can successfully build the ` [... cut]