nix-community/robotnix #29: Documentation: How to add support for a single device?

observer, 2020-08-30:
Hi,

I have a device (potter) that sadly never made it into the official LineageOS device list, however there are actually a bunch of projects still supporting it and these ROMs keep multiple versions of the device tree, vendor files and kernel files, e.g. there is apparently the [Android Ice Cold Project](https://github.com/AICP), which seems to still regularly update these files. 

From what I've read so far on the android build process, it is quite similar regardless on which device you are building, maybe I'm wrong, but as I understood, you only need to somehow put the correct files into the right folders and start the build.

I know you would really like to see a bit of traceability, when it comes to building custom images, but I think some Documentation about how to translate the build process for a single device into robotnix would at the same time be very helpful to get more people started on using it in the first place.

Thanks in Advance!

Atemu, 2020-08-30:
Robotnix itself is (mostly) device-agnostic. The device-specific stuff is contained in `flavors/lineageos/*.json` which are automatically generated from the upstream project infrastructure.

You can edit these manually (add your device's metadata, device- and vendor-dirs) or fork the upstream Hudson, do your changes and point update-devices.py at it.  
We should probably add an option to the script to make this easier.

Some sort of overrides system would probably be the ideal fix for this. See https://github.com/danielfullmer/robotnix/issues/30 (especially the second example).

Atemu, 2020-08-30:
This should make changing the Hudson URL easier: https://github.com/danielfullmer/robotnix/pull/31

danielfullmer, 2020-08-30:
I agree the documentation could use some serious improvement.  I plan to (at least) add some autogenerated docs describing all the available robotnix options, as well as a FAQ for questions like this. In the meantime:

To add/override a source directory, it should be sufficient to just set `source.dirs."<dir_to_override>".src = ...` in your configuration (for example):
```nix
{
  source.dirs."device/apple/iphone".src = pkgs.fetchFromGitHub {
      owner = "apple";
      repo = "iphone";
      rev = "....";
      sha256 = "...";
  };
}
```
Optionally, instead of `pkgs.fetchFromGitHub`, you could use `pkgs.fetchgit` or even just a path to a local directory (for testing).

@Atemu I think forking hudson and using the json updater script is probably overkill if you just want to add support for a single particular device by adding its source dirs.

Atemu, 2020-09-01:
In my case the source dirs exist upstream but are ignored because cheeseburger is officially supported on 16 but not on 17.1. Forking Hudson and modifying targets was the easiest workaround for me.