nix-community/robotnix #30: Override/Overlay system

Atemu, 2020-08-30:
It'd be useful to have a Nixpkgs-like override/overlay system in place to override devices' properties and components of Android without having to alter the JSONs. (Those should only be touched by the update scripts IMO.)

I'm thinking of something like this

```Nix
  (old: { source.dirs."device/oneplus/bacon" = /tmp/bacon; })
```

or

```Nix
  (old: { devices = old.devices // { newDevice = { ... } }; })
```

This would be especially helpful for development as you could, for example, override the src of device A's kernel to be `~/repos/A-kernel/`. Currently this requires pushing your commits to a remote, fetchgit'ing the remote and changing a two hashes in a JSON; not the fastest process.
Ideally you'd be able to override Android components too.

danielfullmer, 2020-08-30:
This should already be possible using the `source.dirs."<dir_to_override>".src` option.  Let me know if that doesn't work properly for you.  I'll close this and leave #29 open until I have proper documentation for this stuff.

closed 2020-08-30 by danielfullmer

Atemu, 2020-08-30:
Wow, that's kinda obvious actually now in hindsight.

How would you go about adding a new device though?

danielfullmer, 2020-08-30:
Unless I'm forgetting something, for an end-user, it should just require setting `device = "newdevice";` and adding the device, kernel, and vendor source trees using `source.dirs`.

Atemu, 2020-09-01:
And no other device metadata is required?

danielfullmer, 2020-09-01:
I don't believe so. The information in `device-metadata.json` is only used to select which directories to include from `device-dirs.json` and `vendor-dirs.json`.

Btw, I'm currently working on modifying the `update-devices.py` script to allow you to select a particular device to update--even if that device isn't in the upstream hudson. This should make it simple for you to use cheeseburger with lineage-17.1. You wouldn't need to fork hudson to use the script, and you wouldn't need to manually track down the device source directories for cheeseburger either.