nix-community/robotnix #32: Build Glodroid variant?

colemickens, 2020-09-02:
Hi,

I was telling samueldr about how I am looking forward to being able to boot Glodroid and mobile-nixos on my Pinephone. They pointed me towards this project. I'm not super well-versed in AOSP to understand where boundaries of projects lie, but maybe you are familiar with Glodroid?

Would it be possible to build Glodroid with Robotnix?

https://glodroid.github.io/
https://github.com/GloDroid/glodroid_manifest

colemickens, 2020-09-02:
I guess it wouldn't be a variant like lineageos/grapheneos, but rather a board/target?

samueldr, 2020-09-02:
AFAIUI it would be both.

danielfullmer, 2020-09-02:
No, I wasn't familiar with Glodroid until you mentioned it.  It certainly looks interesting, and they have goals I like: sticking close to mainline Android, upstreaming their changes, etc.  It definitely should be possible to build with robotnix.  (And it incentivizes me to get a pinephone at some point!)

I've been trying to stick somewhat to the terminology described [here](https://source.android.com/setup/develop/new-device).  So, instead of "variant" , I've been using "flavor" in robotnix for things like vanilla, grapheneos, and lineageos.

I would probably call GloDroid a "flavor" which supports a number of "devices".  One thing supporting calling them a flavor is the fact that they publish their own [manifests](https://github.com/GloDroid/glodroid_manifest) which override a bunch of upstream AOSP repos with patched versions.  They don't do this nearly to the same extent as something like LineageOS or GrapheneOS, but it is still looks different enough that I wouldn't be comfortable with the other option, which would be adding this as a bunch of devices inside the existing vanilla flavor.

Here's a few notes to hopefully get started if you are interested in trying this out, based on reading the documentation [here](https://github.com/GloDroid/glodroid_manifest/blob/master/README.md): You'd need to create a new "repo" json file using `mk-repo-file.py`. Then, in your configuration, add something like `source.dirs = lib.importJSON ./repo-v0.3.0.json;`.  It looks like you'd also need `{ productNamePrefix = ""; device = "pinephone"; }`. Finally, it looks like they want you to build the `images` make target instead of `target-files-package`. So, you'd need to use the `build.mkAndroid` function from `modules/base.nix` to have it build that.

If I have some extra time, I might take a stab at this as well--but I don't have any devices to test.

danielfullmer, 2020-09-03:
I just noticed they have WIP support for RPI3, so if we're lucky I might actually be able to test this on a real device.

danielfullmer, 2020-09-03:
So I was able to get it to build for pinephone, with the configuration file and repo file here: https://gist.github.com/danielfullmer/fe9c7cf8579c9b0ecc373fc456790fa4
This command should be enough to build it pinned to the robotnix revision I tested it on:
```
nix-build "https://gist.github.com/danielfullmer/fe9c7cf8579c9b0ecc373fc456790fa4/archive/8c712df004379537ad1deb754742c83cc2d365ab.tar.gz" -A build.images
```
After a long wait, this will produce `images.tar.gz` under `result`.  I'd presume you'd follow the [instructions](https://github.com/GloDroid/glodroid_manifest) to flash this onto a device.

The support for signed builds in robotnix will not work with this, since it relies on signing target-files, and the glodroid instructions don't seem to consider that.  It probably wouldn't be too hard to reimplement the final steps they have in their `images` target in nix, using the images from a signed target-files-package.

Also, I wasn't able to test this on my RPI3, since it fails to compile for me, which is understandable given the RPI3 support isn't released yet.

danielfullmer, 2021-10-05:
Yes, that sounds right to me.  Let us know if you're able to successfully build and flash an image, or if you run into any other issues.

Mindavi, 2021-11-22:
I should get a (bigger) sd card and a serial cable soon, hopefully I can at least get some logs then.