nix-community/robotnix #34: Android 11 Support

danielfullmer, 2020-09-09:
Tracking issue for Android 11:

Flavors:
- [ ] Vanilla AOSP
      `x86_64` emulator works as of: 8ea39d8b3efb94ca54bb141ce8c87c833b3ae7b7
- [ ] GrapheneOS
- [ ] LineageOS

Apps:
- [ ] MicroG
- [ ] Seedvault
- [ ] Fdroid

Other:
- [ ] Update APK signing: https://gitlab.com/fdroid/fdroidserver/-/issues/827
- [ ] Change `androidVersion` default to `11`

For `vanilla`, we require `android-prepare-vendor` support to be extended to the android 11 images.
`grapheneos` is currently hard at work on android 11, and will also require `android-prepare-vendor`.

petabyteboy, 2020-10-03:
Can confirm this built and booted on my Pixel 4a (sunfish)! Thanks!

danielfullmer, 2020-10-03:
@petabyteboy Thanks for the feedback! Was that for the `vanilla` flavor, or the `grapheneos` flavor?

petabyteboy, 2020-10-03:
It was the `vanilla` flavor. The Pixel 4a isn't supported by GrapheneOS yet as far as I can tell.

danielfullmer, 2020-10-03:
Ah, yes, that slipped my mind!

petabyteboy, 2020-10-03:
I also attempted to get microG working by replacing the patch with one from OmniROM and updating the GmsCore apk to the latest unstable build from https://microg.org/dl/core-en.apk:

```
File: modules/microg.nix
[...]
    source.dirs."frameworks/base".patches = [
      (pkgs.fetchpatch {
        name = "microg.patch";
        url = "https://github.com/omnirom/android_frameworks_base/commit/c60732d87354af6f7928c6dd67d5fb72784b7f0b.patch";
        sha256 = "0ig14flmd9bvcs3gc7y8xwdavgg3f0s7lfppl1snxxfndzx543s4";
      })
    ];
    apps.prebuilt = {
      GmsCore = {
        apk = ../microg-files/core-en.apk;
[...]
```

The patch applies and it compiles, but the microG self-check doesn't agree that it works. Will post screenshots later.

danielfullmer, 2020-10-03:
MicroG was what I planned on working on next, so your work is great timing! Based on a modified version of the patch you referenced (https://gist.github.com/danielfullmer/7ad4d2627bd011ef319bfd26145ad00f), I was able to get MicroG self-check to work, as well as device registration / push notifications, in the emulator. I didn't need to update the microg apk.  PR incoming

petabyteboy, 2020-10-05:
After wiping /data, microG seems to works for me as well! Thanks, this is really awesome.

danielfullmer, 2020-10-06:
I've just changed the default `androidVersion` to `11` in robotnix. (ef6cdeb9accea3bb5805ac7507bb762e6dd42db9)  Vanilla and GrapheneOS should be working well.  I've left Android 10 support in the vanilla flavor for now, but note that it's not receiving monthly updates any more.  I've already removed Android 10 support from the GrapeneOS flavor.

LineageOS will still use Android 10 by default.  If anyone is interested, a good place to contribute here would be to start work on extending the flavor to support the `lineage-18.0` branch.

I do not intend to remove Android 10 support in robotnix the near future.  (This might change--no promises).  When apps like Seedvault begin to only support android 11, I'll keep their android 10 version in the repo for now.  However, keep in mind that the code will be outdated, may contain security issues, and will not receive the same level of basic testing that I will do for Android 11.

One final note: `vanilla` flavor users that are signing their builds will need to re-run `generateKeysScript` to create the new additional keys needed by APEX packages. This should not overwrite any existing keys, but be sure to keep a backup of your keys just in case. (You should be doing this anyway)

petabyteboy, 2020-10-19:
So I tried to build a vanilla Android 11 img for sunfish, this time with signing. I tried both the releaseScript and img targets, and both gave me an error like this:

```
    signing: com.android.vndk.current.apex                     container (packages/modules/vndk/apex/com.android.vndk.current)
           : com.android.vndk.current.apex                     payload   (packages/modules/vndk/apex/com.android.vndk.current.pem)
Traceback (most recent call last):
  File "/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/releasetools/sign_target_files_apks.py", line 1348, in <module>
    main(sys.argv[1:])
  File "/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/releasetools/sign_target_files_apks.py", line 1329, in main
    compressed_extension)
  File "/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/releasetools/sign_target_files_apks.py", line 522, in ProcessTargetFiles
    signing_args=OPTIONS.avb_extra_args.get('apex'))
  File "/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/releasetools/apex_utils.py", line 332, in SignApex
    signing_args)
  File "/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/releasetools/apex_utils.py", line 193, in SignApexPayload
    payload_file, payload_key_path, e))
apex_utils.ApexSigningError: Failed to sign APEX payload /tmp/apex-payload-pZvReo/apex_payload.img with packages/modules/vndk/apex/com.android.vndk.current.pem:
Failed to run command '[u'avbtool', 'add_hashtree_footer', '--do_not_generate_fec', '--algorithm', 'SHA256_RSA4096', '--key', u'packages/modules/vndk/apex/com.android.vndk.current.pem', '--pr
op', 'apex.key:com.android.vndk.v30', '--image', u'/tmp/apex-payload-pZvReo/apex_payload.img', '--salt', '041b9a73ccf530693ebe82982f9a62fd63d1d487a3985f2b58e78db4d83d5088', '--hash_algorithm'
, 'sha256', '--no_hashtree']' (exit code 1):
/nix/store/16kcdpa8p8ss88jgvzmmdgvj5rpaw690-ota-tools/bin/.avbtool: Error getting public key: Can't open packages/modules/vndk/apex/com.android.vndk.current.pem for reading, No such file or d
irectory
139634338489600:error:02001002:system library:fopen:No such file or directory:crypto/bio/bss_file.c:69:fopen('packages/modules/vndk/apex/com.android.vndk.current.pem','r')
139634338489600:error:2006D080:BIO routines:BIO_new_file:no such file:crypto/bio/bss_file.c:76:
unable to load Public Key
```

I did run the generateKeys script multiple times on an empty directory to generate completely fresh keys.

danielfullmer, 2020-10-19:
Hmm, I guess I didn't include all the necessary keys in 4067ee59dfca93a798918334347ece73138db754 . You could try adding these keys to the list that's conditional on `androidVersion >= 11` in `modules/signing.nix`:
```
com.android.vndk.current.key
com.android.vndk.v29.key
com.android.vndk.v28.key
com.android.vndk.v27.key
```
Then re-create and run `generateKeysScript` which should then create these keys.  Google added a ton of new keys, and as far as I can tell, there's no central place where these are all documented.  I've had to grep through all `Android.bp` files looking for `apex_key` modules, and then comparing that with what is included in a built image.

danielfullmer, 2020-10-20:
So I just tried to reproduce your issue with `sunfish` and also with `crosshatch`.  It turns out the set of required keys is conditional on the device somehow, since I needed the `com.android.vndk.current` key for `sunfish` but not for `crosshatch`.  I guess we'll have to track down exactly where this condition is introduced, and determine which keys it affects.

petabyteboy, 2020-10-20:
So, I added the keys you mentioned to `keysToGenerate` in `modules/signing.nix`, re-ran the script and attempted to build with signing again. It's still the same error though, because it doesn't look for these keys in the correct location I suppose.

danielfullmer, 2020-10-20:
Specifically, try this:
```patch
diff --git a/modules/signing.nix b/modules/signing.nix
index 9888cff..03da90b 100644
--- a/modules/signing.nix
+++ b/modules/signing.nix
@@ -60,6 +60,7 @@ in
         "adbd" "art.release" "cellbroadcast" "extservices" "i18n"
         "ipsec" "mediaprovider" "neuralnetworks" "os.statsd" "runtime"
         "permission" "sdkext" "telephony" "tethering" "wifi"
+        "vndk.current" "vndk.v27" "vndk.v28" "vndk.v29"
       ]
     );
 
```

I just tested this last night on `sunfish` so it should be working.

petabyteboy, 2020-10-20:
Thanks, I had added it in the wrong place. Now it works :fireworks:

danielfullmer, 2021-02-16:
Has been working for a while now.

closed 2021-02-16 by danielfullmer