nix-community/robotnix #35: Cannot build LineageOS for dipper

AlienJust, 2020-09-15:
Trying to build LingeageOS for Dipper (Xiaomi Mi 8) with command:

`nix-build "https://github.com/danielfullmer/robotnix/archive/master.tar.gz" --arg configuration '{ device="dipper"; flavor="lineageos"; microg.enable=true; apps.fdroid.enable = true; }' -A img`

but it fails:
```building '/nix/store/84n7gq8ir77q9ip75p5l61f5g37z901s-android_vendor_qcom_opensource_bluetooth-commonsys-intf-2b4cc49.drv'...
exporting https://github.com/LineageOS/android_vendor_qcom_opensource_bluetooth-commonsys-intf (rev 2b4cc49e5e200c339773e08232861803d778d7f4) into /nix/store/wz7l1lcqz5qlp2pajc9yksn165bggg73-android_vendor_qcom_opensource_bluetooth-commonsys-intf-2b4cc49
Initialized empty Git repository in /nix/store/wz7l1lcqz5qlp2pajc9yksn165bggg73-android_vendor_qcom_opensource_bluetooth-commonsys-intf-2b4cc49/.git/
remote: Enumerating objects: 103, done.
remote: Counting objects: 100% (103/103), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 103 (delta 32), reused 101 (delta 30), pack-reused 0
Receiving objects: 100% (103/103), 20.10 KiB | 10.05 MiB/s, done.
Resolving deltas: 100% (32/32), done.
From https://github.com/LineageOS/android_vendor_qcom_opensource_bluetooth-commonsys-intf
 * [new branch]      lineage-17.1 -> origin/lineage-17.1
fatal: reference is not a tree: 2b4cc49e5e200c339773e08232861803d778d7f4
Unable to checkout 2b4cc49e5e200c339773e08232861803d778d7f4 from https://github.com/LineageOS/android_vendor_qcom_opensource_bluetooth-commonsys-intf.
builder for '/nix/store/84n7gq8ir77q9ip75p5l61f5g37z901s-android_vendor_qcom_opensource_bluetooth-commonsys-intf-2b4cc49.drv' failed with exit code 1
building '/nix/store/1lb24v5zmpp7p2sk0a9dzd43qqrhwpwd-otatools-env-usr-multi.drv'...
cannot build derivation '/nix/store/c21j9is640f96ylr7bdmdixhp7gmn296-unpack.sh.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/bb0vqqlacfxsz6ix6k1dipjr4mchqi4q-robotnix-lineage_dipper-2020.09.05.00.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/8x3znyrz2y1xvqm95xd2w660nfjycdg9-dipper-img-2020.09.05.00.zip.drv': 1 dependencies couldn't be built
error: build of '/nix/store/8x3znyrz2y1xvqm95xd2w660nfjycdg9-dipper-img-2020.09.05.00.zip.drv' failed
```

danielfullmer, 2020-09-15:
That commit appears to have disappeared in the LineageOS github repo.  Maybe they did a force-push and github garbage collected the old objects?  I just pushed a small fix in https://github.com/danielfullmer/robotnix/commit/112be5dcbee999c4ccc074f686b635dadce4287c . Hopefully this doesn't happen very often...

Please try again with the latest robotnix commit. To avoid any issues with nix caching and the robotnix tarball url, you should instead do a `git checkout` of robotnix and then run `nix-build ./default.nix ...`.

AlienJust, 2020-09-15:
> That commit appears to have disappeared in the LineageOS github repo. Maybe they did a force-push and github garbage collected the old objects? I just pushed a small fix in [112be5d](https://github.com/danielfullmer/robotnix/commit/112be5dcbee999c4ccc074f686b635dadce4287c) . Hopefully this doesn't happen very often...
> 
> Please try again with the latest robotnix commit. To avoid any issues with nix caching and the robotnix tarball url, you should instead do a `git checkout` of robotnix and then run `nix-build ./default.nix ...`.

Thanks! Build process now passes that part.

AlienJust, 2020-09-15:
Now something wrong with kernel (as I understand):
```
  CC      drivers/staging/qcacld-3.0/../qca-wifi-host-cmn/target_if/green_ap/src/target_if_green_ap.o
  LD      drivers/staging/qcacld-3.0/wlan.o
  LD      drivers/staging/qcacld-3.0/built-in.o
  LD      drivers/staging/built-in.o
  LD      drivers/built-in.o
make[1]: Leaving directory '/build/out/target/product/dipper/obj/KERNEL_OBJ'
make: *** [Makefile:152: sub-make] Error 2
make: Leaving directory '/build/kernel/xiaomi/sdm845'
ninja: build stopped: subcommand failed.
16:30:11 ninja failed with: exit status 1

#### failed to build some targets (48:40 (mm:ss)) ####

installing
cp: cannot stat '/build/out/target/product/dipper/otatools.zip': No such file or directory
builder for '/nix/store/izi336qfvwzx6376mx6bg98iw32syhz3-robotnix-lineage_dipper-2020.09.05.00.drv' failed with exit code 1
cannot build derivation '/nix/store/r6h02wsv0hw5pbdpc1n9jrybi1024dq5-dipper-img-2020.09.05.00.zip.drv': 1 dependencies couldn't be built
error: build of '/nix/store/r6h02wsv0hw5pbdpc1n9jrybi1024dq5-dipper-img-2020.09.05.00.zip.drv' failed
```

Atemu, 2020-09-15:
The actual error happened further up the log

danielfullmer, 2020-09-15:
Yeah, the error is this:
```
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

It's probably some script somewhere trying to get the latest commit or something like that. I'm digging into it now.

danielfullmer, 2020-09-15:
Fix should be in bf07dc2ef0fbb1744f9fc4278a65b06e493c3b6e. It was an upstream LineageOS issue with using (I think) a prebuilt `dtc`. I just updated the LineageOS source to the latest as of today. I ran an entire build with your command line options and it should be working now. Please let me know if it works or doesn't work for you.

AlienJust, 2020-09-16:
Thanks a lot! As the result I've got zip archive with several images. I've flashed images via TWRP recovery to my phone (boot, vendor and system) and it works. Awesome!

closed 2020-09-16 by AlienJust