nix-community/robotnix #36: releaseScript fails with syntax error

Cornu, 2020-09-19:
After following the Readme and creating keys and building the "release script" for sargo (Pixel 3a) running the script fails.

```
$ ./release ./keys/sargo/
./release: line 28: syntax error near unexpected token ||
./release: line 28:  || exit 1
```

Manually removing the newlines before the '|| exit 1' fixes the error. (lines 28, 36 and 44)

danielfullmer, 2020-09-19:
I just pushed some improvements in the latest commits on master that should fix this: 2c76104f2110c149cdcdef7b47b26f87c40e4766 883113139ba8212d5a5017412c1e8c502b7ec8eb 880859df12b7b39036994be7d829ff53460cb143

While it's recommended in the docs, the `releaseScript` isn't part of my normal usage, so it unfortunately isn't tested as often. It's on the TODO to include it in some sort of continuous testing so hopefully I don't break it as often.

Also, please let me know if it works for you. I'm interested in if it works on sargo. Robotnix has been tested on pixel 1-3, but I haven't tried it on anything newer.

Cornu, 2020-09-19:
Thank you for the quick answer and fixes and for robotnix :) The grapheneos build is running on my Pixel 3a. I'm now building the vanilla image and will report back.

One additional question, maybe worth a separate Issue. How are you running the Update server (with NixOs?) or should i just look at the GrapheneOs Update Server?

danielfullmer, 2020-09-19:
Thanks. It's great to hear it works for you so far!

Sure, I need to document this better, but here's what I do:
I have this set in my configuration:
```apps.updater.url = "https://daniel.fullmer.me/android/";```
And I have a server with nginx that serves a directory under `https://daniel.fullmer.me/android/`, which contains these files:
`crosshatch-stable` and `crosshatch-ota_update-2020.09.13.04.zip`.  The `crosshatch-stable` file contains metadata for the updater apps, and contains just the line:
```2020.09.13.04 1599972803 QQ3A.200805.001```
The format is just `<buildNumber> <datetime> <buildID>`.

That's really all that's needed to be hosted at the update url location.  The `otaDir` derivation under `modules/release.nix` will generate a directory containing these files. However, since it involves signing the release inside a nix derivation, it would also require passing the sandbox exception for the keys.  So, it doesn't work conveniently with the `releaseScript` approach.

I also have it on my TODO list to make a NixOS module that would make deploying an update server like this trivial.  I'd also like to make the lineageos updater work as well.

closed 2020-09-20 by danielfullmer

danielfullmer, 2020-10-23:
@Cornu I've just pushed a commit with a small amount of documentation on using the updater  e8db7e4855aa55bcffb82e7e98a94fa288f50d1b.  I forgot in my earlier comment that using `releaseScript` will actually also generate this metadata as well.  So, no matter which approach you use, you shouldn't have to gcreate this metadata yourself.