nix-community/robotnix #41: image signing in nix doesn't work if no sandbox

eyecreate, 2020-09-23:
image signing using nix relies on extra-sandbox-paths to mount a directory at same path. If nix.conf has sandbox disabled(ex. as nix docker container has), then there is no mounting done. Maybe add nix sandbox as requirement for building or try to support accessing the directory without mounting.

danielfullmer, 2020-10-19:
Sorry for taking so long to respond to this issue.  I'm actually kind of conflicted about how I'd want to handle this.  I didn't realize the docker container had the sandbox disabled, and I'd expect a lot of people would want to build in docker.  In principle, it could be easy to handle the case where we don't have a sandbox, because we have `keyStorePath` available already in nix.  However, I actually want to remove `keyStorePath` from the robotnix configuration.  Having it available makes it too easy to accidentally import the keys into your nix store.  `keyStorePath` is currently used just so we can read the _public keys_ from the nix expression to extract some metadata (e.g. key hashes), but I'm planning on moving that logic into the `generateKeysScript`.

I think in the mean-time, I'll just add the nix sandbox as a requirement for signing inside nix.  (Of course, you could still use the `releaseScript` to do it outside of nix)

danielfullmer, 2021-05-09:
The narrow issue of it not working inside a sandbox should be fixed by #101 and by also setting `signing.buildTimeKeyStorePath`. However, there may still be other issues with building inside Docker, e.g. if nested user namespaces don't: https://github.com/danielfullmer/robotnix/issues/94#issuecomment-817421958

PR #101 (2021-05-09): Keys refactor again
Renames `keyStorePath` to `signing.keyStorePath`. Also introduces a `signing.buildTimeKeyStorePath` option, which is useful if we can't use the Nix sandbox to bind-mount our keys under `/keys`.

If you use the Nix sandbox normally, there's no need to use `signing.buildTimeKeyStorePath`, since using it makes the signing derivations contain a string of the path your keys are located  under.

(I needed `signing.buildTimeKeyStorePath` so I could produce a derivation which would sign the build with snakeoil keys under `/nix/store`.)

Closes #41
closed 2021-05-09 by danielfullmer