nix-community/robotnix #45: Android 11 SystemUI Reproducibility

danielfullmer, 2020-10-05:
On `{ device="crosshatch"; flavor="vanilla"; androidVersion=11; }`
`SYSTEM_EXT/priv-app/SystemUI/SystemUI.apk` in `target-files` is not being built reproducibly.

Relevant `diffoscope` output:
```
├── smali/$$ServiceLoaderMethods.smali
│┄ Ordering differences only
│ @@ -6,21 +6,21 @@
│  # direct methods
│  .method public static $load0()Ljava/util/Iterator;
│      .locals 3
│  
│      const/4 v0, 0x1
│  
│      :try_start_0
│ -    new-array v0, v0, [Lkotlinx/coroutines/internal/MainDispatcherFactory;
│ +    new-array v0, v0, [Lkotlinx/coroutines/CoroutineExceptionHandler;
│  
│      const/4 v1, 0x0
│  
│ -    new-instance v2, Lkotlinx/coroutines/android/AndroidDispatcherFactory;
│ +    new-instance v2, Lkotlinx/coroutines/android/AndroidExceptionPreHandler;
│  
│ -    invoke-direct {v2}, Lkotlinx/coroutines/android/AndroidDispatcherFactory;-><init>()V
│ +    invoke-direct {v2}, Lkotlinx/coroutines/android/AndroidExceptionPreHandler;-><init>()V
│  
│      aput-object v2, v0, v1
│  
│      invoke-static {v0}, Ljava/util/Arrays;->asList([Ljava/lang/Object;)Ljava/util/List;
│  
│      move-result-object v0
│  
│ @@ -48,21 +48,21 @@
│  
│  .method public static $load1()Ljava/util/Iterator;
│      .locals 3
│  
│      const/4 v0, 0x1
│  
│      :try_start_0
│ -    new-array v0, v0, [Lkotlinx/coroutines/CoroutineExceptionHandler;
│ +    new-array v0, v0, [Lkotlinx/coroutines/internal/MainDispatcherFactory;
│  
│      const/4 v1, 0x0
│  
│ -    new-instance v2, Lkotlinx/coroutines/android/AndroidExceptionPreHandler;
│ +    new-instance v2, Lkotlinx/coroutines/android/AndroidDispatcherFactory;
│  
│ -    invoke-direct {v2}, Lkotlinx/coroutines/android/AndroidExceptionPreHandler;-><init>()V
│ +    invoke-direct {v2}, Lkotlinx/coroutines/android/AndroidDispatcherFactory;-><init>()V
│  
│      aput-object v2, v0, v1
│  
│      invoke-static {v0}, Ljava/util/Arrays;->asList([Ljava/lang/Object;)Ljava/util/List;
│  
│      move-result-object v0
├── smali/kotlinx/coroutines/CoroutineExceptionHandlerImplKt.smali
│ @@ -16,15 +16,15 @@
│  
│  
│  # direct methods
│  .method static constructor <clinit>()V
│      .locals 2
│  
│      .line 23
│ -    invoke-static {}, L$$ServiceLoaderMethods;->$load1()Ljava/util/Iterator;
│ +    invoke-static {}, L$$ServiceLoaderMethods;->$load0()Ljava/util/Iterator;
│  
│      move-result-object v0
│  
│      const-string v1, "ServiceLoader.load(\n    \u2026.classLoader\n).iterator()"
│  
│      invoke-static {v0, v1}, Lkotlin/jvm/internal/Intrinsics;->checkExpressionValueIsNotNull(Ljava/lang/Object;Ljava/lang/String;)V
├── smali/kotlinx/coroutines/internal/MainDispatcherLoader.smali
│ @@ -88,15 +88,15 @@
│  
│      move-result-object v1
│  
│      goto :goto_0
│  
│      .line 33
│      :cond_0
│ -    invoke-static {}, L$$ServiceLoaderMethods;->$load0()Ljava/util/Iterator;
│ +    invoke-static {}, L$$ServiceLoaderMethods;->$load1()Ljava/util/Iterator;
│  
│      move-result-object v1
│  
│      const-string v2, "ServiceLoader.load(\n    \u2026             ).iterator()"
│  
│      invoke-static {v1, v2}, Lkotlin/jvm/internal/Intrinsics;->checkExpressionValueIsNotNull(Ljava/lang/Object;Ljava/lang/String;)V
```

danielfullmer, 2021-04-06:
Tested again recently and can no longer reproduce.

closed 2021-04-06 by danielfullmer

Atemu, 2021-04-07:
You mean, you can reproduce now ;)