nix-community/robotnix #47: google apps support (e.g. opengapps)

ajs124, 2020-10-19:
microG is pretty cool, but some of us still require the google apps right now.

Do you have any plans to add support for including google apps in rom builds or if not, are you open to PRs that would add support for that?

danielfullmer, 2020-10-19:
I don't personally plan to implement this, but I would be open to PRs.  I would have a couple of potential concerns though:
1) Maintenance burden.  I'd want to have some assurance that it could be regularly updated, and wouldn't be broken for long periods of time. (Since I don't plan to update or use it). Perhaps future robotnix work on automated testing in the emulator could help here.  I've seen some mention on the internet about GTS (GMS test suite), which is perhaps like the CTS (compatibility test suite). GTS sounds nice but I think it is restricted to Android partners.
2) Authenticity of build inputs.  I'm not sure how the opengapps scripts work, but I'd like to have some assurance that the files it produces come from reliable sources.  (Not just prebuilts from some random github/gitlab repo).  LineageOS also has this problem with vendor binaries from `TheMuppets`, but I intend to make the warnings and documentation about this issue more clear.  So, this is not strictly a requirement, but I just want to make the concerns clear to potential users.

If you notice, I actually have a `google` module under `modules/google.nix`.  This just grabs some random files from the upstream `crosshatch` image that I personally wanted for Google Fi.  This will probably get moved out of this repo, however, since it's much lower quality than the rest of the modules--and likely doesn't work on most devices besides my personal `crosshatch`.

ajs124, 2020-10-22:
Cool, then I'll see if I can get around to implementing this.

RE: 1. Assuming I get it working properly, I'm planning on running this on my daily driver phone, so I should be motivated enough to maintain it.
RE: 2. I did a bit of looking around in their various repos and scripts and I _think_ it is possible to download upstream factory rom images and use their scripts and probably some more scripting around it to get all the relevant files.

danielfullmer, 2020-11-23:
Another way to alleviate my concern around authenticity of build inputs would be to check signatures of the Google APKs inside Nix, e.g. https://github.com/danielfullmer/robotnix/pull/59  If this was done, we could theoretically fetch them from anywhere (not just trusted ROM images), and then run a signature check to verify authenticity, and rely on Nix's sha256 checks for build input integrity