nix-community/robotnix #49: Allow adding intermediate products to garbage collector root

petabyteboy, 2020-10-19:
I'm not sure if this is a common use case, but I build robotnix every few days or weeks on my big server, but the SSD space is quite limited and I want to run nix-collect-garbage in between to clean up unrelated things. I would still like to be able to keep the android source files and similar things that will be needed in the next build in some profile.

danielfullmer, 2020-10-19:
One way I think you could do this now is to run `nix-build ... -A build.debugEnterEnv -o enter-env`.  This would produce a shell script that enters an FHS environment and also bind-mounts the source dirs necessary to build android. It's what I use sometimes to check build-related things without doing the normal `repo` checkout.  But, because it has explicit runtime references to all the soruce dirs in `/nix/store`, having the script as a gc root should prevent removing the underlying source dirs.

petabyteboy, 2020-10-19:
Thanks, this is mostly what I was looking for! What about Chromium? I guess the 'pre'built Chromium will be included, but not the sources? Is there a second thing I could build that would include the Chromium build env?

danielfullmer, 2020-10-19:
Yep, I believe you're right that it would only save the prebuilt chromium apk and not the source.  You could save another reference by running: `nix-build ./apks/default.nix -A chromium.src`

petabyteboy, 2020-10-22:
This is solved for me. Documenting debugEnterEnv somewhere in README.md would be nice-to-have.

closed 2020-10-22 by petabyteboy