nix-community/robotnix #53: Build GrapheneOS for Pixel 4a (sunfish)

petabyteboy, 2020-10-22:
I know there is no official GrapheneOS release for sunfish yet, but I would like to get any possible issues out of the way already and also understand the process a bit better. I noticed that sunfish uses a different source.manifest.rev and apv.buildID in the vanilla flavor. How will this work for GrapheneOS? In `modules/pixel` there is no image for sunfish + buildID RP1A.201005.004, but does GrapheneOS rely on blobs from the original software matching the source revision of GrapheneOS?

danielfullmer, 2020-10-22:
So, yes, the `apv.buildID` has to match one available in the `modules/pixel/*.json` files. I believe the tagged releases that GrapheneOS publishes should match the ones used in upstream vanilla android.  If Google's next AOSP release publishes multiple tags then I expect GrapheneOS will match those tags.  We'll basically just reproduce the conditions used in the vanilla flavor for `apv.buildID`.  For reference, the released git tags are published here: https://source.android.com/setup/start/build-numbers and also in the `android-building` mailing list.

We might have to add in some hack for the kernel source like is currently in `flavors/grapheneos/default.nix` for other devices. I'm not anticipating any other significant issues with adding `sailfish` support for GrapheneOS, after they release it.

hmenke, 2020-11-25:
Does this work yet?  I've built GrapheneOS for sunfish and signed it but after `fastboot flashing lock` I can't boot anymore.  It just says

> Can't find valid operating system.
> The device will not start.
> 
> Visit this link on another device:
> g.co/ABH
> 
> ID: 879CD3F1

Do I have to enroll the signing keys somewhere first?

hmenke, 2020-11-25:
Could this be the problem?

https://github.com/danielfullmer/robotnix/blob/2eeeffe134ca0e52476e268eaff292e56f515b1a/flavors/grapheneos/default.nix#L99

hmenke, 2020-11-25:
I've got one step further but this is still rather unsatisfactory.  I manually flashed the AVB custom key into the correct partition:
```
$ fastboot flash avb_custom_key ./keys/sunfish/avb_pkmd.bin
```
Now I am able to boot the OS but before it comes up there is still a warning message

> Your device is loading a different operating system.
>
> Visit this link on another device:
> g.co/ABH
>
> ID: BA135E0F

danielfullmer, 2020-11-25:
@hmenke Glad you found the `avb_custom_key`, and that it's working otherwise.  I'll put that on the TODO list to add to our documentation.   I believe the "Your device is loading a different operating system" message is intended behavior.  It happens on stock GrapheneOS as well. Note the boot flow diagram from https://source.android.com/security/verifiedboot/boot-flow
[image]

closed 2020-11-25 by danielfullmer

hmenke, 2020-11-25:
@danielfullmer Sorry for not reading the official Android documentation better.  I found out about `avb_custom_key` because I downloaded the official GrapheneOS factory image to compare files until I find why it doesn't work and the [`flash-all.sh`](https://github.com/GrapheneOS/device_common/blob/53615b871109b61f433ad2a439f01222b928223d/generate-factory-images-common.sh#L171-L176) of GrapheneOS has
```bash
fastboot erase avb_custom_key
fastboot flash avb_custom_key avb_pkmd.bin
```
Does robotnix already have a documentation format?  I'm more than happy to help and document stuff along the way as I try to get to grips with my sunfish.  Also, is there any other way of supporting robotnix development?  You don't seem to participate in GitHub Sponsors and there is no donation link anywhere.

danielfullmer, 2020-11-25:
@hmenke No need to apologize. Thanks for trying out robotnix!

Ah, I see, the GrapheneOS documentation used to explicitly include instructions on setting the `avb_custom_key`, but it appears they've moved that step into the factory image flashing script.  (Btw, robotnix can produce a factory image like this via the `factoryImg` output)

Currently, the only documentation is in the `README.md`.  Markdown format is fine for now, as having the documentation content available is more important than what format it is in. As I look into doing a documentation pass in the (hopefully) near future, I'll look into if we should change formats.

Thanks for the support! I'll look into setting up GitHub sponsors.