nix-community/robotnix #58: Building LineageOS for a specific device

newkozlukov, 2020-10-25:
Hi! Thanks for the great work you're doing here. Also, your talk at nixcon was cool!

I've been trying to use `robotnix` to build `LineageOS` for Google Nexus 7 2013 (LTE) (named `deb` or `razorg`). I should note that the device isn't (anymore) supported by lineageos team, they don't provide prebuilt images. The page for the device is: https://wiki.lineageos.org/devices/deb/build

The build fails with this message (cropped)

```
...
Which product would you like? [aosp_arm] lineage_deb
Which product would you like? [aosp_arm] lineage_deb
build/make/core/product_config.mk:223: error: Can not locate config makefile for product "lineage_deb".
build/make/core/product_config.mk:223: error: Can not locate config makefile for product "lineage_deb".
21:23:56 dumpvars failed with: exit status 1
21:23:56 dumpvars failed with: exit status 1
** Not a valid product: lineage_deb
** Not a valid product: lineage_deb
...
```

Here are my attempts at fixing this: https://github.com/newkozlukov/robotnix/pull/1

I wasn't sure if I'm running the correct commands (README would benefit from some more details) and I couldn't track the repo that has a script that could output such messages. I just tried running some straightforward stuff, as much as I had spare time for, hoping it could work.
It is possible that I haven't done enough research, but I thought I'd still ask for an advice. 

Thanks

danielfullmer, 2020-10-26:
So, it looks like you're on a good track.  One thing I can say is that the `update-device-dirs.py` script under `flavors/lineageos` actually has support for updating just a particular device, even if that device isn't one of the supported ones.  Running `./update-device-dirs.py asus_deb` should automatically update the `device-dirs.json` with just the directories that `asus_deb` depends on.  However, it's possible that the newer device sources might not work with the existing LineageOS sources in that file, since it looks like the last time I updated it was on 2020-09-15.

You're right that you would need to add your device to `device-metadata.json` if you want the lineageos flavor to automatically automatically use the relevant directories from `device-dirs.json` and `vendor-dirs.json`.

I agree this is an area of where the documentation could be improved (https://github.com/danielfullmer/robotnix/issues/29).  Adding a new device isn't currently "supported" in the documentation.  That said, it is probably a goal we'd like to work towards.

danielfullmer, 2020-10-26:
I tried this myself.  I think the underlying cause is probably the fact that [LineageOS/android_device_asus_deb](https://github.com/LineageOS/android_device_asus_deb) is missing an `AndroidProducts.mk` file.  It doesn't look like it's currently set up to work properly with Android 10.

That repo would at least need a commit like the following one:
https://github.com/LineageOS/android_device_oneplus_bacon/commit/b0cf99e4cd21e68b6701c8c7947ceb8611c67b6c

newkozlukov, 2020-10-29:
> `AndroidProducts.mk`

Quick update -- yup, that worked! Also that repo missed the cm->lineage rebranding so paths weren't valid.

New error message looks like (full version in the [commit message](https://github.com/newkozlukov/robotnix/pull/1/commits/5917acc0d2849d68149f95245f252bf212d650cc)):

```

[100% 140/140] out/soong/.bootstrap/bin/soong_build out/soong/build.ninja
FAILED: out/soong/build.ninja
out/soong/.bootstrap/bin/soong_build -t -l out/.module_paths/Android.bp.list -b out/soong -n out -d out/soong/build.ninja.d -globFile out/soong/.bootstrap/build-globs.ninja -o out/soong/build.ninja Android.bp
error: system/bt/types/Android.bp:9:1: module "libbluetooth-types" variant "linux_glibc_x86_64_static": Path is outside directory: /bluetooth
```

Probably some string interpolation with empty/undeclared variable, but I guess I'll postpone this now until the end of next week. It could be nice shortcut if you had some hints regarding new error though

danielfullmer, 2021-04-15:
The documentation aspect of this issue is a duplicate of https://github.com/danielfullmer/robotnix/issues/29
Although, it looks like https://github.com/LineageOS/android_device_asus_deb does have a `lineage-17.1` branch now.

closed 2021-04-15 by danielfullmer