nix-community/robotnix #74: dex2oatd failure when building on Linux 5.10: Invalid argument

Atemu, 2020-12-18:
https://gist.github.com/Atemu/825405424837ce48205bce3053e57c38

The fail is at

```
Failed to copy oat file to file: out/soong/cheeseburger/dex_apexjars_unstripped/system/framework/arm/apex.oat: Invalid argument
```

I believe.

I run ZFS, so that might be a factor here. It wouldn't be the first time ZFS has thrown a weird error at me because of some rarely used Linux FS feature it doesn't implement quite right.

Is anyone able to reproduce? (on  non-ZFS too?)

```Nix
{
  device="cheeseburger";
  flavor="lineageos";

  apps.fdroid.enable = true;

  microg.enable = true;

  apps.seedvault.enable = true;
}
```

A minimal reproducer (or potential ways of getting one) would also be nice as this happens near the very end of the ginormous build process.

hmenke, 2020-12-18:
Note that OpenZFS is not yet officially compatible with Linux 5.10.

Atemu, 2020-12-18:
Huh, I was sure v2.0.0 included 5.10 compat patches but I can still see some open PRs regarding that. Guess I more reason to get off 5.10 in the meantime.

I'd still like to know whether ZFS is the cause though, so if anyone can try to build LOS on 5.10 (preferably for cheeseburger), that'd be great. (I don't have any non-ZFS storage fast and large enough for an android build dir.)

hmenke, 2020-12-18:
Yes, ZFS 2.0.0 *compiles* with Linux 5.10 but the maxiumum supported version is Linux 5.9.  See also https://github.com/openzfs/zfs/blob/master/META

danielfullmer, 2021-01-28:
Just successfully built a LineageOS image for cheeseburger using Linux 5.10 + OpenZFS 2.0.1 without encountering this problem.  OpenZFS 2.0.1 does have support for 5.10, so perhaps it was fixed upstream in the meantime.  Let me know if you still encounter the issue.

closed 2021-01-28 by danielfullmer