nix-community/robotnix #76: LOS device-dirs update doesn't work with the Google/Asus Flox

Atemu, 2020-12-30:
The update script fails at updating flox because it tries to get device_google_flox because its vendor is google but the actual repo name is device_asus_flox.

This information seems to be recorded in updater/device_deps.json which we don't use at all anymore apparently. Is there a specific reason for that @danielfullmer?

hmenke, 2021-03-17:
On my experimental [lineageos-18.1 branch](https://github.com/hmenke/robotnix/tree/lineageos-18.1) I just added this exception:
```patch
diff --git a/flavors/lineageos/update-device-metadata.py b/flavors/lineageos/update-device-metadata.py
index d95dbc9..be0aa85 100755
--- a/flavors/lineageos/update-device-metadata.py
+++ b/flavors/lineageos/update-device-metadata.py
@@ -47,6 +47,10 @@ def fetch_metadata():
         if data['model'] == 'shamu':
             vendor = 'moto'
 
+        # Workaround google flox source tree inconsistency
+        if data['model'] == 'flox':
+            vendor = 'asus'
+
         metadata[data['model']].update({
             'vendor': vendor,
             'name': data['name'],
```

Atemu, 2021-03-17:
Have you tried building for it? IIRC I also had to declare that quirk in Nix somewhere.

hmenke, 2021-03-17:
I just started a build with this command line
```console
$ nix-build --arg configuration '{ device="flox"; flavor="lineageos"; }' -A img
```
and it's working so far (currently at 48%).

danielfullmer, 2021-03-28:
The exception was included in the update python script in e01d772f66767f5ab5bb2a5c5e8f3aa91aef86ed. Tested building with `{ device="flox"; flavor="lineageos"; }` as well.

closed 2021-03-28 by danielfullmer