nix-community/robotnix #80: attestation-server fails on first boot

danielfullmer, 2021-01-06:
If `passwordFile` is set, I have failures at first boot with the following error message: `invalid user 'attestation-server'`.  I can fix this until next boot using `systemctl restart attestation-server`.  I haven't dug into this yet, but perhaps the race condition is still present?  

CC @hmenke

hmenke, 2021-01-06:
It currently works on my machine but I admit that this is an ugly workaround.  I was hoping that simply ordering a different command before copying the credentials would fix the race condition but it seems your computer is too fast.  Anyway, the next version of systemd addresses this properly with https://github.com/systemd/systemd/pull/16568.

hmenke, 2021-01-06:
System 247 is already in nixpkgs (https://github.com/NixOS/nixpkgs/pull/102355) but not yet in the stable release.

hmenke, 2021-01-16:
Is it actually okay if we wait for the next stable or should I fix it now?  I think something like this would work but this would give the service access to all files and directories owned by group `keys`.

```patch
diff --git a/nixos/attestation-server/module.nix b/nixos/attestation-server/module.nix
index fe7e55f..cb6dfd8 100644
--- a/nixos/attestation-server/module.nix
+++ b/nixos/attestation-server/module.nix
@@ -124,10 +124,11 @@ in
           # first command is run as root the allocation of uid and gid for the
           # service seems to be delayed, so we just run something else first.
           "${pkgs.coreutils}/bin/touch %S/attestation/emailPassword"
-          "+${pkgs.coreutils}/bin/install -m 0600 -o %N -g %N ${passwordFile} %S/attestation/emailPassword"
+          "+${pkgs.coreutils}/bin/install -m 0640 -o %N -g keys ${passwordFile} %S/attestation/emailPassword"
           ''${pkgs.sqlite}/bin/sqlite3 %S/attestation/attestation.db "INSERT OR REPLACE INTO Configuration VALUES ${values}"''
           "${pkgs.coreutils}/bin/rm -f %S/attestation/emailPassword"
         ];
+        SupplementaryGroups = [ "keys" ];
 
         # When sending TERM, e.g. for restart, AttestationServer fails with
         # this exit code.
```

danielfullmer, 2021-01-18:
It'd be nice to have it working before the next major NixOS release, and I aim to not use features from NixOS until it's in a stable release.  If there isn't another reasonable alternative that doesn't require granting permissions to files in the `keys` group, then I could live with this workaround.

PR #83 (2021-01-21): Avoid race condition in attestation-server
Fixes #80
closed 2021-01-21 by danielfullmer