nix-community/robotnix #91: update-device-dirs.py `fatal: write error: No space left on device`

lapp0, 2021-03-01:
```
./update-device-dirs.py oneplus_instantnoodle
device/oneplus/instantnoodle is up to date.
device/oneplus/sm8250-common is up to date.
kernel/oneplus/sm8250 is up to date.
device/oneplus/common is up to date.
Checking out https://github.com/TheMuppets/proprietary_vendor_oneplus refs/heads/lineage-18.1
Initialized empty Git repository in /run/user/1000/git-checkout-tmp-7FnS5p1p/proprietary_vendor_oneplus/.git/
remote: Enumerating objects: 4032, done.
remote: Counting objects: 100% (4032/4032), done.
remote: Compressing objects: 100% (2637/2637), done.
remote: Total 4032 (delta 1399), reused 3819 (delta 1295), pack-reused 0
Receiving objects: 100% (4032/4032), 748.30 MiB | 10.43 MiB/s, done.
Resolving deltas: 100% (1399/1399), done.
From https://github.com/TheMuppets/proprietary_vendor_oneplus
 * branch            lineage-18.1 -> FETCH_HEAD
 * [new branch]      lineage-18.1 -> origin/lineage-18.1
error: unable to write file sm8250-common/proprietary/vendor/lib64/libarcsoft_superportrait.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libarcsoft_tricam_calibration.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libarcsoft_tricam_verification.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libarcsoft_video_fringing_correction.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libbatching.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libbluetooth_audio_session_qti.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libbtnv.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libc2d30_bltlib.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcacertclient.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamera_nn_stub.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamerapostproc.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxexternalformatutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxfacialfeatures.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxfdalgo.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxfdengine.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxifestriping.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamximageformatutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxncs.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxstatscore.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxswprocessalgo.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxtintlessalgo.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcamxtintlessalgo_21.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcdfw.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcdfw_remote_api.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcdsp_default_listener.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcdsprpc.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libchilog.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libchrome.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcne.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcneapiclient.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcneoplookup.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcneqmiutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcom.qti.chinodeutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libconfigdb.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcpion.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcvface_api.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcvp2.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcvp2_hfi.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libcvp_common.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdataitems.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdiag.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdisp-aba.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdisplayqos.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdisplayskuutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdpmqmihal.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdrmfs.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdrmtime.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdsi_netctrl.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdsutils.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdualcam_image_optical_zoom.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdualcam_optical_zoom_control.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libdualcam_video_optical_zoom.so
error: unable to write file sm8250-common/proprietary/vendor/lib64/libfastcvdsp_stub.so
[... 286 lines cut ...]
  File "./update-device-dirs.py", line 144, in main
    vendor_dirs = fetch_vendor_dirs(metadata, 'vendor-dirs.json')
  File "./update-device-dirs.py", line 118, in fetch_vendor_dirs
    fetch_relpath(dirs, relpath, url)
  File "./update-device-dirs.py", line 46, in fetch_relpath
    dirs[relpath] = checkout_git(url, 'refs/heads/' + BRANCH)
  File "./update-device-dirs.py", line 33, in checkout_git
    json_text = subprocess.check_output([ "nix-prefetch-git", "--url", url, "--rev", rev]).decode()
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/nix/store/v72cj06nk69cynckz2s12rhar25k1h7v-python3-3.8.5/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['nix-prefetch-git', '--url', 'https://github.com/TheMuppets/proprietary_vendor_oneplus', '--rev', 'refs/heads/lineage-18.1']' returned non-zero exit status 1.
```

This can be fixed by setting `env` in `checkout_git`: 
```
def checkout_git(url, rev):
    print("Checking out %s %s" % (url, rev))
    env = os.environ.update({'TMPDIR': '/tmp'})
    json_text = subprocess.check_output(["nix-prefetch-git", "--url", url, "--rev", rev], env=env).decode()
    return json.loads(json_text)
```

Related: https://github.com/NixOS/nixpkgs/issues/101140

Atemu, 2021-03-01:
You can also just do `TMPDIR=/tmp ./update-device-dirs.py oneplus_instantnoodle`.

I think a better default TMPDIR might not be a bad idea though.

lapp0, 2021-03-03:
yes, easier to pass the environment variable to the script than define it in the script. Leaving this open because it should either be documented or closed once https://github.com/NixOS/nixpkgs/issues/101140 is resolved IMO

danielfullmer, 2021-03-03:
Maybe, for this and the other scripts, we can check the free disk space and warn the user if it seems too small.

commit bb4a4bdd (2021-08-07): Add free space checks to update scripts

Closes #91
closed 2021-08-16 by danielfullmer