nix-community/robotnix #92: Not a valid product: lineage_instantnoodle

lapp0, 2021-03-01:
Hi, I'm trying to build for a new device, the OnePlus 8 Pro.

I took the following steps

Ran `./update-device-dirs.py oneplus_instantnoodle`

Manually added the following to `device-metadata.json`:

```
  "instantnoodle": {
    "branch": "lineage-18.1",
    "lineage_recovery": true,
    "name": "8 Pro",
    "variant": "userdebug",
    "vendor": "oneplus"
  },
```

Created `instantnoodlep.nix`: 

```
{ config, pkgs, lib, ... }:

with pkgs;
with lib;

{
  device = "instantnoodle";
  flavor = "lineageos";

  ccache.enable = true;

  apps = {
    seedvault.enable = true;
    fdroid.enable = true;
    bromite.enable = true;
  };

  webview.bromite.enable = true;
  microg.enable = true;

  kernel.src = fetchGit {
    url = "git@github.com:OnePlusOSS/android_kernel_oneplus_sm8250.git";
    sha256 = "0nrkagm2rah4savfgqky9wkp128rcvnwi21r1bix2vlsrssrq8c2";
  };

  # TODO: add this once its readyhttps://github.com/danielfullmer/robotnix/pull/90

  hosts = fetchurl { # block ads
    url = "https://raw.githubusercontent.com/StevenBlack/hosts/6aa31cb4ad613ae3d9f886f2b5f418b4083733cb/hosts";
    sha256 = "0nrkagm2rah4savfgqky9wkp128rcvnwi21r1bix2vlsrssrq8c2";
  };
}
```

Ran `nix-build ./robotnix/default.nix --arg configuration ./instantnoodlep.nix -A img --option extra-sandbox-paths /keys="$(pwd)/keys" --show-trace` with output

```
these derivations will be built:
  /nix/store/nbi3n9brq525scdzvn3aklp3apqrymcn-robotnix-lineage_instantnoodle-2020.12.29.06.drv
  /nix/store/8nmhal5pqqx33gi8d1fs3sbblyi84jgm-ota-tools.drv
  /nix/store/snx0pxnlrbch0kcyfqsm8m4pl18j8pqi-instantnoodle-img-2020.12.29.06.zip.drv
building '/nix/store/nbi3n9brq525scdzvn3aklp3apqrymcn-robotnix-lineage_instantnoodle-2020.12.29.06.drv'...
unpacking sources
patching sources
configuring
building
Build type choices are:
     1. release
     2. debug

Which would you like? [1] release

Which product would you like? [aosp_arm] lineage_instantnoodle
device/oneplus/sm8250-common/BoardConfigCommon.mk:197: error: device/qcom/sepolicy_vndr/SEPolicy.mk: No such file or directory
16:00:05 dumpvars failed with: exit status 1
** Not a valid product: lineage_instantnoodle

Variant choices are:
     1. user
     2. userdebug
     3. eng
Which would you like? [eng] userdebug

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=10
TARGET_PRODUCT=aosp_arm
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.10.17-x86_64
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ3A.200805.001
OUT_DIR=out
============================================
builder for '/nix/store/nbi3n9brq525scdzvn3aklp3apqrymcn-robotnix-lineage_instantnoodle-2020.12.29.06.drv' failed with exit code 1
cannot build derivation '/nix/store/snx0pxnlrbch0kcyfqsm8m4pl18j8pqi-instantnoodle-img-2020.12.29.06.zip.drv': 1 dependencies couldn't be built
```

hmenke, 2021-03-01:
You don't need to edit the Robotnix sources to add a new device.  In this example I've added support for Google Nexus 4 (mako) to my local configuration: https://git.henrimenke.de/henri/android/src/branch/master/mako.nix
Likewise you will have to locate the `device`, `kernel`, and `vendor` dirs for you your device.

hmenke, 2021-03-01:
Ah, I see instantnoodle is supported on the Lineage OS 18.1 development target, but Lineage OS 18.1 is not yet available in Robotnix.  You will have to change the branch in the update scripts in the `flavors/lineageos` directory and run them to update the entire source tree to 18.1.

lapp0, 2021-03-01:
Right, I set
```
  "instantnoodle": {
    "branch": "lineage-18.1",
    "lineage_recovery": true,
    "name": "8 Pro",
    "variant": "userdebug",
    "vendor": "oneplus"
  },
```

and ran the update script, which updated `device-metadata.nix`

I assume I'll need to update `repo-lineage-17.1.json` manually for this to work?

Is there anything else I need to change?

hmenke, 2021-03-01:
That's unfortunately not enough.  You will have to update all of Lineage OS in robotnix.  Check the different scripts in the `flavors/lineageos` directory, in particular `update.sh`.  They perform the source checkout and update all the JSON files.

hmenke, 2021-03-01:
You could try again with this branch: https://github.com/hmenke/robotnix/tree/lineageos-18.1

lapp0, 2021-03-01:
Awesome, thanks so much, I was struggling with `update.sh` because a bunch of repos either no longer existing or not having 18.1. I'll try this now.

lapp0, 2021-03-01:
Ok, please let me know when to pull again!

hmenke, 2021-03-01:
The update is now complete.  I've force pushed so the easiest would be to just delete your checkout and clone again.

hmenke, 2021-03-01:
The build starts up fine, but that doesn't mean that it will complete.

hmenke, 2021-03-01:
Just as I wrote that, the kernel build failed.
```
make: Entering directory '/build/kernel/oneplus/sm8250'
make[1]: Entering directory '/build/out/target/product/instantnoodle/obj/KERNEL_OBJ'
  HOSTCC  scripts/basic/fixdep
  GEN     ./Makefile
/build/kernel/oneplus/sm8250/scripts/basic/fixdep.c:97:10: fatal error: 'sys/types.h' file not found
#include <sys/types.h>
         ^~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.host:90: scripts/basic/fixdep] Error 1
make[1]: *** [/build/kernel/oneplus/sm8250/Makefile:478: scripts_basic] Error 2
make[1]: Leaving directory '/build/out/target/product/instantnoodle/obj/KERNEL_OBJ'
make: *** [Makefile:146: sub-make] Error 2
```
Now you can choose, wait for upstream to fix the error or fix it yourself and patch the kernel.  Once you have a working patch, patching the kernel is actually as easy as adding this
```nix
{
  source.dirs."kernel/oneplus/sm8250".patches = [ ./example.patch ] ;
}
```

lapp0, 2021-03-01:
Hah, I found this issue by searching the exact same error.

adrianparvino, 2021-03-03:
Hi, I've been exploring this. From what i'm understanding, it's a difference in the build system.

`master` vs `18.01`
```
[  2% 2696/114122] Building Kernel Config
make: Entering directory '/build/kernel/sony/sdm660'
make[1]: Entering directory '/build/out/target/product/pioneer/obj/KERNEL_OBJ'
  HOSTCC  scripts/basic/fixdep
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
```
vs
```
[  0% 614/116852] Building Kernel Config
FAILED: out/target/product/instantnoodle/obj/KERNEL_OBJ/.config
/bin/bash -c "(PATH=/build/out/host/linux-x86/bin:\$PATH PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/lib64:\$LD_LIBRARY_PATH PATH=/build/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-/bin:\$PATH PATH=/build/prebuilts/tools-lineage/linux-x86/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/tools-lineage/linux-x86/lib:\$LD_LIBRARY_PATH PERL5LIB=/build/prebuilts/tools-lineage/common/perl-base BISON_PKGDATADIR=/build/prebuilts/build-tools/common/bison /build/prebuilts/build-tools/linux-x86/bin/make  -j12 CFLAGS_MODULE=\"-fno-pic\" CPATH=\"/usr/include:/usr/include/x86_64-linux-gnu\" HOSTCFLAGS=\"-fuse-ld=lld\" HOSTLDFLAGS=\"-L/usr/lib/x86_64-linux-gnu -L/usr/lib64 -fuse-ld=lld\" DTC_EXT=/build/prebuilts/misc/linux-x86/dtc/dtc OEM_TARGET_PRODUCT=instantnoodle HOSTCC=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang HOSTCXX=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang++ LEX=/build/prebuilts/build-tools/linux-x86/bin/flex YACC=/build/prebuilts/build-tools/linux-x86/bin/bison M4=/build/prebuilts/build-tools/linux-x86/bin/m4 KBUILD_BUILD_TIMESTAMP=\"Thu Jan  1 00:00:01 UTC 1970\" -C kernel/oneplus/sm8250 O=/build/out/target/product/instantnoodle/obj/KERNEL_OBJ ARCH=arm64 CROSS_COMPILE=\"/build/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-\" CROSS_COMPILE_ARM32=\"/build/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-\" CLANG_TRIPLE=aarch64-linux-gnu- CC=\"/nix/store/c9g3pfp82h6rcpmgp68wcmq81hv3kyv7-ccache-3.7.11/bin/ccache clang\"  VARIANT_DEFCONFIG= SELINUX_DEFCONFIG= vendor/kona-perf_defconfig ) && (if [ ! -z \"\" ]; then                       echo \"Overriding kernel config with ''\";                      echo  >> out/target/product/instantnoodle/obj/KERNEL_OBJ/.config;                  PATH=/build/out/host/linux-x86/bin:\$PATH PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/lib64:\$LD_LIBRARY_PATH PATH=/build/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-/bin:\$PATH PATH=/build/prebuilts/tools-lineage/linux-x86/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/tools-lineage/linux-x86/lib:\$LD_LIBRARY_PATH PERL5LIB=/build/prebuilts/tools-lineage/common/perl-base BISON_PKGDATADIR=/build/prebuilts/build-tools/common/bison /build/prebuilts/build-tools/linux-x86/bin/make  -j12 CFLAGS_MODULE=\"-fno-pic\" CPATH=\"/usr/include:/usr/include/x86_64-linux-gnu\" HOSTCFLAGS=\"-fuse-ld=lld\" HOSTLDFLAGS=\"-L/usr/lib/x86_64-linux-gnu -L/usr/lib64 -fuse-ld=lld\" DTC_EXT=/build/prebuilts/misc/linux-x86/dtc/dtc OEM_TARGET_PRODUCT=instantnoodle HOSTCC=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang HOSTCXX=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang++ LEX=/build/prebuilts/build-tools/linux-x86/bin/flex YACC=/build/prebuilts/build-tools/linux-x86/bin/bison M4=/build/prebuilts/build-tools/linux-x86/bin/m4 KBUILD_BUILD_TIMESTAMP=\"Thu Jan  1 00:00:01 UTC 1970\" -C kernel/oneplus/sm8250 O=/build/out/target/product/instantnoodle/obj/KERNEL_OBJ ARCH=arm64 CROSS_COMPILE=\"/build/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-\" CROSS_COMPILE_ARM32=\"/build/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-\" CLANG_TRIPLE=aarch64-linux-gnu- CC=\"/nix/store/c9g3pfp82h6rcpmgp68wcmq81hv3kyv7-ccache-3.7.11/bin/ccache clang\"  oldconfig;                 fi ) && (PATH=/build/out/host/linux-x86/bin:\$PATH PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/clang/host/linux-x86/clang-r383902b/lib64:\$LD_LIBRARY_PATH PATH=/build/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-/bin:\$PATH PATH=/build/prebuilts/tools-lineage/linux-x86/bin:\$PATH LD_LIBRARY_PATH=/build/prebuilts/tools-lineage/linux-x86/lib:\$LD_LIBRARY_PATH PERL5LIB=/build/prebuilts/tools-lineage/common/perl-base BISON_PKGDATADIR=/build/prebuilts/build-tools/common/bison /build/prebuilts/build-tools/linux-x86/bin/make  -j12 CFLAGS_MODULE=\"-fno-pic\" CPATH=\"/usr/include:/usr/include/x86_64-linux-gnu\" HOSTCFLAGS=\"-fuse-ld=lld\" HOSTLDFLAGS=\"-L/usr/lib/x86_64-linux-gnu -L/usr/lib64 -fuse-ld=lld\" DTC_EXT=/build/prebuilts/misc/linux-x86/dtc/dtc OEM_TARGET_PRODUCT=instantnoodle HOSTCC=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang HOSTCXX=/build/prebuilts/clang/host/linux-x86/clang-r383902b/bin/clang++ LEX=/build/prebuilts/build-tools/linux-x86/bin/flex YACC=/build/prebuilts/build-tools/linux-x86/bin/bison M4=/build/prebuilts/build-tools/linux-x86/bin/m4 KBU [... cut ...] cache-3.7.11/bin/ccache clang\"  KCONFIG_ALLCONFIG=out/target/product/instantnoodle/obj/KERNEL_OBJ/.config alldefconfig;            fi )"
make: Entering directory '/build/kernel/oneplus/sm8250'
make[1]: Entering directory '/build/out/target/product/instantnoodle/obj/KERNEL_OBJ'
  HOSTCC  scripts/basic/fixdep
/build/kernel/oneplus/sm8250/scripts/basic/fixdep.c:97:10: fatal error: 'sys/types.h' file not found
#include <sys/types.h>
         ^~~~~~~~~~~~~
  GEN     ./Makefile
1 error generated.
```

adrianparvino, 2021-03-03:
The former configuration is `{ device="pioneer";    flavor="lineageos"; }` while the latter is the `instantnoodlep` a while back.

I also tried the pioneer configuration on `18.01` with the same error.

adrianparvino, 2021-03-03:
Would it be correct to do this? This changes the error.

```patch
From c61309bca91a7b52232f6f2c06630698995ae170 Mon Sep 17 00:00:00 2001
From: Adrian Parvin Ouano <adrianparvino@gmail.com>
Date: Wed, 3 Mar 2021 17:57:31 +0800
Subject: [PATCH] Add glibc into FHS

---
 modules/base.nix | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/modules/base.nix b/modules/base.nix
index 3f45ee1..cc1051e 100644
--- a/modules/base.nix
+++ b/modules/base.nix
@@ -447,7 +447,7 @@ in
 
       env = pkgs.buildFHSUserEnv {
         name = "robotnix-build";
-        targetPkgs = pkgs: config.envPackages;
+        targetPkgs = pkgs: config.envPackages ++ (with pkgs; [ glibc glibc.dev ]);
         multiPkgs = pkgs: with pkgs; [ zlib ];
       };
     };
-- 
2.30.0
```

danielfullmer, 2021-03-03:
@adrianparvino 
You could set `envPackages = [ pkgs.glibc.dev ];` in your own configuration. 

See also a similar workaround I had to do in the past: https://github.com/danielfullmer/robotnix/blob/ae32ae5ec112bd6e2ead1978ffa1c5ba8ca850b7/flavors/lineageos/default.nix#L127
Other devices (at least for `lineage-17.1`) needed openssl headers as well.  Theoretically, the developers should include all requirements to build, and should not be depending on system headers/libs like this.  However, AOSP doesn't yet enforce this requirement in their build system.

This setting should probably be included by default for `lineageos-18.1`, when we merge it.

hmenke, 2021-03-08:
I've updated my [lineageos-18.1 branch](https://github.com/hmenke/robotnix/tree/lineageos-18.1) with these fixes and
the `img` attribute builds successfully.  I don't have a OnePlus 8, so I can't test it obviously.

lapp0, 2021-03-09:
Thanks for your help @adrianparvino and @hmenke I have working lineageos 18.1 running on my OnePlus8Pro.

I used this configuration:

```
{ config, pkgs, lib, ... }:

with pkgs;
with lib;

{
  device = "instantnoodlep";
  flavor = "lineageos";

  ccache.enable = true;

  apps = {
    fdroid.enable = true;
    bromite.enable = true;
  };

  # https://github.com/danielfullmer/robotnix/issues/92#issuecomment-788300033
  apps.seedvault.enable = true;
  source.dirs."packages/apps/Seedvault".enable = false;

  webview.bromite.enable = true;
  microg.enable = true;

  # TODO: add this once its ready https://github.com/danielfullmer/robotnix/pull/90

  hosts = fetchurl { # block ads
    url = "https://raw.githubusercontent.com/StevenBlack/hosts/6aa31cb4ad613ae3d9f886f2b5f418b4083733cb/hosts";
    sha256 = "0nrkagm2rah4savfgqky9wkp128rcvnwi21r1bix2vlsrssrq8c2";
  };
}
```

I had the following changes to hmenkes branch to make it work (thanks adrian!)

```
modified   flavors/lineageos/device-dirs.json
@@ -1356,9 +1356,9 @@
     ],
     "fetchSubmodules": false,
     "leaveDotGit": false,
-    "path": "/nix/store/nip5kn9097grij5jhns8k4j1s2nqz82n-android_device_oneplus_sm8250-common",
-    "rev": "b710ebaaf3d16797e56b2c1adfa8c5999184eade",
-    "sha256": "19ql7dfib02fn61ric86zxhi2jr0q2vm0vma09q4p6i3vk4rgljz",
+    "path": "/nix/store/lm18gmxghh8dcv4f1qzzffc0kmiwji52-android_device_oneplus_sm8250-common",
+    "rev": "4a242e6a1f5c795425f580b65e7194edf16fa3c9",
+    "sha256": "0cwz84xzw0nxa7cv15x1sicafyr6wb8yj5dclzj4cw3dm9mgy6xs",
     "url": "https://github.com/LineageOS/android_device_oneplus_sm8250-common"
   },
   "device/oppo/common": {
@@ -3517,4 +3517,4 @@
     "sha256": "0adwbml12jc74z544gfp9z4mph7xqd9fz31xgvg2qpifpfv6sql5",
     "url": "https://github.com/LineageOS/android_system_qcom"
   }
-}
\ No newline at end of file
+}

modified   modules/base.nix
@@ -447,8 +447,8 @@ in
 
       env = pkgs.buildFHSUserEnv {
         name = "robotnix-build";
-        targetPkgs = pkgs: config.envPackages;
-        multiPkgs = pkgs: with pkgs; [ zlib ];
+        targetPkgs = pkgs: config.envPackages ++ (with pkgs; [ gcc-unwrapped ]);
+        multiPkgs = pkgs: with pkgs; [ glibc.dev zlib kmod ];
       };
     };
   }];
```

LineageOS kept crashing with `mdm_crash_shutdown`. Adrian recognized that this was a modem firmware failure. 

The modem firmware for OxygenOS android 11 was needed to make this work.

https://oxygenos.oneplus.net/OnePlus8ProOxygen_15.O.21_OTA_021_all_2007160051_downgrade_65eeb8e408d44a1d.zip

```
unzip -j oneplus8prooxygen.zip payload.bin
nix-shell -p python3Packages.bsdiff4 python3Packages.protobuf --run "python3 payload_dumper/payload_dumper.py payload.bin"
nix-shell -p androidenv.androidPkgs_9_0.platform-tools --run "fastboot flash modem output/modem.img"
```

I'm wondering if including the official android modem firmware in the build `result` would be out of scope for this project?

closed 2021-03-10 by danielfullmer

lapp0, 2021-03-10:
Thanks for detailing why it's not possible (or rather why it would be extremely cumbersome). Makes sense!