nix-community/robotnix #107: Can this be used to build the Android SDK from source?

malte-v, 2021-06-22:
Title says it all. If not, I guess it wouldn't be that much work since this project anready handles a lot of the AOSP build system? Please correct me if I'm wrong, I don't know anything about this.

malte-v, 2021-06-22:
I'm a little confused because [this](https://github.com/danielfullmer/robotnix/blob/c65ac9c1ec6d5147c36891e9eb140ed789d7da93/sdk/default.nix) looks like building the SDK from source but the flake pulls in [this](https://github.com/tadfisher/android-nixpkgs) which looks like a set of binary packages to me.

danielfullmer, 2021-06-23:
Yes, it can be used the build the SDK.  The current `sdk` directory was just a proof-of-concept for a particular version of the SDK. The prebuilt SDKs are currently only used in a few places (e.g. building some apps under `apks/`)  In principle, I don't see any reason why we couldn't instead use the source-built SDKs inside robotnix, instead of relying on [tadfisher/android-nixpkgs](https://github.com/tadfisher/android-nixpkgs).  It's definitely on the TODO list.  I was hoping to have a robust automated builder + larger cache working before then, because it effectively makes most robotnix builds depend on additional large compilation step(s) for the SDK(s) used for any apps, and I expect most users would not want to build the SDKs themselves.

malte-v, 2021-06-24:
I was hoping to have a source-built SDK inside nixpkgs: https://github.com/NixOS/nixpkgs/issues/75603#issuecomment-866284005 That would also solve our caching problem since it would be built by Hydra.

tadfisher, 2021-06-27:
I would love for this to happen, btw. If this is something that would be better suited for me to implement in `android-nixpkgs` I would appreciate pointers for porting the sdk bits. If not, I'd love to help out here however I can.