nix-community/robotnix #109: Shell for chromium env crashes on some AMD systems

yu-re-ka, 2021-07-14:
```
building '/nix/store/xhqzrcjc584mmcr6l990jlbnm0h86a80-binutils-wrapper-2.35.1.drv'...
unpacking sources
patching sources
installing
post-installation fixup
shrinking RPATHs of ELF executables and libraries in /nix/store/nqy3mrnphcivvp09assir70kks3sh2pw-binutils-wrapper-2.35.1
checking for references to /build/ in /nix/store/nqy3mrnphcivvp09assir70kks3sh2pw-binutils-wrapper-2.35.1...
find: '/nix/store/0im18a5w70yxiw4w95p2miijsnsss3q9-binutils-wrapper-2.35.1-man': No such file or directory
find: '/nix/store/24x2h1zjada18fnm2bh054ka8xg1zhm1-binutils-wrapper-2.35.1-info': No such file or directory
copying path '/nix/store/v0xkrmsp5rfpkjv0mr1292k3dxiz0rm3-gcc-wrapper-10.3.0' from 'https://cache.nixos.org'...
copying path '/nix/store/ghw5p1k4jxmjpn169x89yn1x5idcnssq-gcc-10.3.0-info' from 'https://cache.nixos.org'...
copying path '/nix/store/ld6m9fr7y8bh7kcnykrpcdzi0hxhfr59-gcc-10.3.0-man' from 'https://cache.nixos.org'...
building '/nix/store/07s5wggyk65102ndm1vmikakj3r0jcfd-gcc-wrapper-10.3.0.drv'...
unpacking sources
patching sources
installing
post-installation fixup
shrinking RPATHs of ELF executables and libraries in /nix/store/zclg8n69gi7sysqhyp19k8xfwy3apzx7-gcc-wrapper-10.3.0
checking for references to /build/ in /nix/store/zclg8n69gi7sysqhyp19k8xfwy3apzx7-gcc-wrapper-10.3.0...
find: '/nix/store/7c1511z04d9wal1nv62f895k4c9g4s06-gcc-wrapper-10.3.0-man': No such file or directory
find: '/nix/store/n05d9x1n5g7j05fifg10q6p38ycgk4b5-gcc-wrapper-10.3.0-info': No such file or directory
substituteStream(): WARNING: pattern '
' doesn't match anything in file '/nix/store/zclg8n69gi7sysqhyp19k8xfwy3apzx7-gcc-wrapper-10.3.0/nix-support/libcxx-cxxflags'
substituteStream(): WARNING: pattern '
' doesn't match anything in file '/nix/store/zclg8n69gi7sysqhyp19k8xfwy3apzx7-gcc-wrapper-10.3.0/nix-support/libcxx-ldflags'
building '/nix/store/hsdyxkhk35fn2nbw8q76sjjaf3y2hbaa-stdenv-linux.drv'...
builder for '/nix/store/hsdyxkhk35fn2nbw8q76sjjaf3y2hbaa-stdenv-linux.drv' failed due to signal 11 (Segmentation fault)
error: build of '/nix/store/hsdyxkhk35fn2nbw8q76sjjaf3y2hbaa-stdenv-linux.drv' failed
```

This is reproducible on my Ryzen 9 5950X and Ryzen 7 PRO 4750G. It works fine on an Intel Core i7-3320M.

danielfullmer, 2021-07-14:
What command did you use to get this error?  Which shell are you referring to?

I have a Ryzen 3970x that I use for all my builds, so it'd be a little surprising if the CPU is the issue.

yu-re-ka, 2021-07-14:
Robotnix checkout: aec5e83873c1b6f92057660358ec3fd7dd022a93

bramble.nix:

```
{ config, pkgs, lib, ... }:

with pkgs;
with lib;
{
  device = "bramble";
  flavor = "grapheneos";
  androidVersion = 11;

  buildDateTime = 1626262359; # Use `date "+%s"` to get the current time

  signing = {
    enable = true;
    keyStorePath = "/var/secrets/android-keys";
  };
  
  apps = {
    fdroid.enable = true;
    seedvault.enable = true;
  };

  microg.enable = true;
}
```

```
./build.sh --arg configuration bramble.nix -A img

danielfullmer, 2021-07-14:
I can't reproduce this issue on my machine.  This also looks like an upstream (nixpkgs) issue with the i686 bootstrap files and your hardware.

I'd expect the following command might fail for you also, since it uses the exact same bootstrapped `bash` while building.  (from `/nix/store/dhcjqsvr0kynlx39mpfqagkydgclcdlf-bootstrap-tools/bin/bash`)
```
$ nix-build https://github.com/nixos/nixpkgs/archive/93963c27b934f24289a94b9e3784d60a9b77e92c.tar.gz -A pkgsi686Linux.bash --check
```

I'd check if you can still reproduce this issue on `pkgsi686Linux.bash` using a newer revision of nixpkgs.  If so, I'd report this upstream.

danielfullmer, 2021-08-03:
Any progress on diagnosing this issue?

Closing for now since I believe the issue is either a hardware problem or a problem with upstream `nixpkgs`, not `robotnix`.

closed 2021-08-03 by danielfullmer

yu-re-ka, 2021-08-12:
This is still an issue for me every time I use robotnix after a garbage collect. I don't know how to reproduce it without robotnix, with nixpkgs only.

yu-re-ka, 2021-08-12:
Actually yes I can reproduce this with just nixpkgs. Opened an issue in nixpkgs: https://github.com/NixOS/nixpkgs/issues/133608