nix-community/robotnix #111: Documentation: How to update GrapheneOS flavor?

yu-re-ka, 2021-08-03:
```
[root@nixos:~/robotnix]# flavors/grapheneos/update.sh
flavors/grapheneos/update.sh: line 20: ../../mk-repo-file.py: No such file or directory

[root@nixos:~/robotnix]# cd flavors/grapheneos/

[root@nixos:~/robotnix/flavors/grapheneos]# ./update.sh
/usr/bin/env: ‘python3’: No such file or directory

[root@nixos:~/robotnix/flavors/grapheneos]# nix-env -iA nixos.python3
installing 'python3-3.8.9'
building '/nix/store/ns0y7345drryymcrsw6izsl4pk49c0sd-user-environment.drv'...
created 258 symlinks in user environment

[root@nixos:~/robotnix/flavors/grapheneos]# ./update.sh
Traceback (most recent call last):
  File "../../mk-repo-file.py", line 192, in <module>
    main()
  File "../../mk-repo-file.py", line 154, in main
    mirrors = dict(mirror.split("=") for mirror in os.environ.get('ROBOTNIX_GIT_MIRRORS', '').split('|'))
ValueError: dictionary update sequence element #0 has length 1; 2 is required

[root@nixos:~/robotnix/flavors/grapheneos]# export ROBOTNIX_GIT_MIRRORS=https://android.googlesource.com=/mnt/cache/mirror

[root@nixos:~/robotnix/flavors/grapheneos]# ./update.sh
Fetching information for https://github.com/GrapheneOS/platform_manifest ../anbox/repo-anbox.json
Traceback (most recent call last):
  File "../../mk-repo-file.py", line 192, in <module>
    main()
  File "../../mk-repo-file.py", line 182, in main
    make_repo_file(args.url, args.ref, filename, ref_type,
  File "../../mk-repo-file.py", line 60, in make_repo_file
    subprocess.check_call(['repo', 'init', f'--manifest-url={url}', f'--manifest-branch=refs/{ref_type.value}/{ref}', *REPO_FLAGS], cwd=tmpdir)
  File "/nix/store/6cfajs6lsy9b4wxp3jvyyl1g5x2pjmpr-python3-3.8.9/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/nix/store/6cfajs6lsy9b4wxp3jvyyl1g5x2pjmpr-python3-3.8.9/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/nix/store/6cfajs6lsy9b4wxp3jvyyl1g5x2pjmpr-python3-3.8.9/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/nix/store/6cfajs6lsy9b4wxp3jvyyl1g5x2pjmpr-python3-3.8.9/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'repo'

[root@nixos:~/robotnix/flavors/grapheneos]# git diff

[root@nixos:~/robotnix/flavors/grapheneos]#

hmenke, 2021-08-03:
You have to `cd` to the flavor directory first.
```console
cd flavors/grapheneos
./update.sh
```

danielfullmer, 2021-08-03:
You can use `nix-shell` in the top-level directory to enter an environment that would have all dependencies required for the updater scripts. You are right, the current scripts are not documented, and must be used from the right directory, etc.  I have a local branch where I'm working on these issues currently.

w.r.t GrapheneOS, I usually have to wait at least a few hours (to a couple days at most) for the tagged release to get pushed upstream to the update channel for at least one device. I'm trying to match the same `BUILD_DATETIME` used upstream, which we only know once at least one build is publicly released.  (See `extract-upstream-params.sh` and how `upstream-params.nix` is used in the flavor).

Closing in favor of https://github.com/danielfullmer/robotnix/issues/55

closed 2021-08-03 by danielfullmer

yu-re-ka, 2021-08-03:
Thanks for the hint with `nix-shell` in the repository root! It would still be nice if this process was a bit easier to understand, but I was able to build today's GrapheneOS beta :)