nix-community/robotnix #112: Why is robotnix building one chromium and two bromites?

Ninlives, 2021-08-12:
So I'm trying to use robotnix to build lineageos, and I've add these settings:
```nix
{ ... }:{
  device = "cmi";
  flavor = "lineageos";
  androidVersion = 11;
  variant = "user";
  signing.enable = true;
  apps.seedvault.enable = true;
  apps.chromium.enable = false;
  apps.bromite.enable = true;
  webview.bromite.enable = true;
  webview.bromite.availableByDefault = true;

  # A list of source.dirs config
}
```

When I trying to build the releaseScript I noticed that nix is building one chromium and two bromite simultaneously. Is this an expected behavior? If not, what's wrong with my setup?

Ninlives, 2021-08-12:
The `webview.chromium.enable` should be turned off as well, and the bromite app and webview are meant to be built separately.

closed 2021-08-12 by Ninlives

danielfullmer, 2021-08-12:
Yes, I normally build the webview and app simultaneously so they can share intermediate build products, but I encountered issues when doing this with bromite:
https://github.com/danielfullmer/robotnix/blob/master/modules/apps/chromium.nix#L104-L108

Ninlives, 2021-08-13:
By the way: Is that possible to split the Android build system into several nix derivations? I see that kernel and webview packages are already separated, is there a guide to do something similar with other components?

Atemu, 2021-08-13:
https://github.com/danielfullmer/robotnix/issues/25

danielfullmer, 2021-08-13:
It's a tough problem, but yes, https://github.com/danielfullmer/robotnix/issues/25 has some discussion of earlier attempts at splitting up the Android build system. The approach with blueprint2nix/soongnix is not something I'm going to continue working on, since it would require a huge maintenance effort, and android is moving to using bazel anyway.  It is something I think about a lot though.