nix-community/robotnix #121: CalyxOS support?

pshirshov, 2021-09-09:
Hi,

Could you add support for Calyx flavour please?

Thanks!

Kranzes, 2021-09-09:
I can try to implement that but i am more interested in what Calyx has to offer that isn't already in something like LineageOS or GrapheneOS?

Kranzes, 2021-09-09:
Afaik? Calyx is just LineageOS with MicroG and fdroid/aurora store pre installed, all of these can already be achieved through robotnix.

pshirshov, 2021-09-09:
They maintain some patches, I don't have enough knowledge in that domain, though I believe they did some improvements in work profile management, preinstalled aurora services, probably something else.

Still, Calyx is a good distro, it would be very nice to have reproducible builds for it.

Kranzes, 2021-09-09:
You have yet to explain why Calyx is any different/good, the stuff that you've mentioned so far can be added to LineageOS using robotnix.

Kranzes, 2021-09-09:
Again, i can try adding Calyx support to Robotnix but from what i have just read in quite a lot of places from the dev themselves is that the only actual difference between LineageOS + MicroG and Calyx is the default preinstalled apps, that as i said before, can all be added to the robotnix built LineageOS through the apps.prebuilt module.

pshirshov, 2021-09-09:
Probably you are right, I guess in case I configure robotnix with all their patches it would be the same, just not sure how much effort would it require. From what I can see they maintain a huge patchset, e.g. they have 20+ patches just for Chromium: https://gitlab.com/CalyxOS/chromium-patches

Though maybe you are right, I'll try to use Robotnix with graphene or lineage flavours together with my patches and see if it would work for me better than calyx.

Kranzes, 2021-09-09:
You can use source.dirs."SOURCEDIR".patches and use fetchpatch, though you can just use bromite instead of chromium which is already more secure and privacy oriented than chromium, there is a bormite module already in robotnix.

pshirshov, 2021-09-09:
>  the only actual difference between LineageOS + MicroG and Calyx is the default preinstalled apps

Nope, there are many minor patches, e.g. they provide a UI option for Cloudflare private DNS (and I extended their patch for couple more privacy-oriented providers). Also they turned off Lineage telemetry and in contrast with lineage they provide full AVB support OoB.

I'm not sure how many such things are there though and how important they are.

In fact I just need a distro with AVB support, MicroG, working NLP, work profiles, without telemetry and with my base system patches allowing SeedVault (that's a Calyx project btw) to take true full system backups.

Kranzes, 2021-09-09:
Right now, you can get MicroG, NLP, Work profiles and Seedvault in LineageOS, I am not sure about the AVB support and telemetry...

Kranzes, 2021-09-09:
I am sure the "telemetry" can also be disabled through robotnix.

chirayudesai, 2021-09-09:
> Afaik, Calyx is just LineageOS with MicroG and fdroid/aurora store pre installed, all of these can already be achieved through robotnix.

No, that is not true.

CalyxOS is based on AOSP, but we add a lot on top.
We also include patches from other projects, including LineageOS, improve them and share them back.

https://calyxos.org/news/2021/09/02/feature-update/ this was the most recent update, we developed a bunch of features in that, improved existing features, and also ported some from other projects. And when possible, we try to contribute the improvements back to them.

pshirshov, 2021-09-09:
From what I understand out of my experience of building lineage imperatively, there is no way to turn on AVB support easily.

Regarding telemetry - it should be a simple change but I didn't dig through their codebase and don't know how to turn it off.

Kranzes, 2021-09-09:
Calys's building instructions look similar to LineageOS's and AOSP in general so it shouldn't be hard to implement it into robotnix.

chirayudesai, 2021-09-09:
> Calys's building instructions look similar to LineageOS's and AOSP in general so it shouldn't be hard to implement it into robotnix.

Indeed. Let me know if you have any questions, either here or on any chat channels (I'm on the robotnix channel, and you're also welcome to join #calyxos-dev:matrix.org)

One thing that we talked about in past was all of the prebuilt APKs we ship, but most of them are optional, and usually unmodified apps downloaded from F-Droid, so that part should be easy to re-create if needed.

pshirshov, 2021-09-10:
Can I assist somehow?

Kranzes, 2021-09-10:
The docs's have basic information about how to add a new flavor you should take a look at it.

pshirshov, 2021-09-10:
Well, that's kinda too basic. I won't handle that.

Kranzes, 2021-09-10:
Honestly just try to mimic vanilla or lineageos

thestinger, 2021-11-11:
> Probably you are right, I guess in case I configure robotnix with all their patches it would be the same, just not sure how much effort would it require. From what I can see they maintain a huge patchset, e.g. they have 20+ patches just for Chromium: https://gitlab.com/CalyxOS/chromium-patches

thestinger, 2021-11-11:
> Probably you are right, I guess in case I configure robotnix with all their patches it would be the same, just not sure how much effort would it require. From what I can see they maintain a huge patchset, e.g. they have 20+ patches just for Chromium: https://gitlab.com/CalyxOS/chromium-patches

This is just a very out-of-date copy of the GrapheneOS Chromium patches which became Vanadium...

I suggest looking at the author of the patches. In many cases, CalyxOS stripped attribution from changes taken from us so that's not a reliable way to see the origin of the code but they didn't do that there. They don't include the MIT license in their distribution of the built code which is an MIT license violation.

danielfullmer, 2021-11-13:
I apologize for the very long delay in responding to this PR.  Besides our hands being forced by https://github.com/danielfullmer/robotnix/issues/137, there is a maintenance concern about adding additional flavors to the base robotnix repo.  I currently already maintain 3 flavors, and I don't have great tooling to accept and maintain additional ones without significantly more work that I'm not currently able to commit to.

However, Nix does provide great mechanisms for including modules/flavors from external sources:
https://docs.robotnix.org/development.html#external-modules

closed 2021-11-13 by danielfullmer