nix-community/robotnix #122: Support for Aurora Services?

pshirshov, 2021-09-10:
Most of the users would like to be able to install and update Play Store apps.

https://gitlab.com/AuroraOSS/AuroraServices provides us a way to have background application updates.

It would be very convenient to have it as a module.

Kranzes, 2021-09-10:
This should be it:

      apps.prebuilt.ExampleApp = {
        apk = (pkgs.fetchurl {
          url = "https://gitlab.com/AuroraOSS/AuroraServices/uploads/c22e95975571e9db143567690777a56e/AuroraServices_v1.1.1.apk";
          sha256 = "sha256-8D83aPGVWQfsMTrkl8/ZK09vWO4cAbazi9C2oqmlBZM=";
        });
        privileged = true;
        privappPermissions = [ "INSTALL_PACKAGES" "DELETE_PACKAGES" ];
        packageName = "com.aurora.services";
      };

pshirshov, 2021-09-10:
Thanks!

I'm trying to add a module to build it from source.  But can't figure out how do you generate gradle-env jsons. Perhaps you use gradle2nix?

Kranzes, 2021-09-10:
Aurora Services seems to be forked from Fdroid privileged-extension, so it should be fairly easy to implement it as a module, though it wont be a big module since it doesnt support adding custom repos for well known reasons like fdroid does. Also no i have never used gradle2nix :(

pshirshov, 2021-09-10:
Well, how do you generate these insane jsons then?

Kranzes, 2021-09-10:
>Well, how do you generate these insane jsons then?

Are you talking specifically about me?

pshirshov, 2021-09-10:
Not sure actually. Anyway, I already figured it out, indeed I have to use gradle2nix.

Kranzes, 2021-09-10:
Are you in the matrix channel, would you be down to share what ever you are doing there?

pshirshov, 2021-09-10:
Nope, I'm not even sure what channel you are talking about :)
I'm just trying to add a module for Aurora Services, will create a P/R in case of success.

Kranzes, 2021-09-10:
https://app.element.io/#/room/#robotnix:nixos.org

pshirshov, 2021-09-10:
Yup, got in there.

pshirshov, 2021-09-10:
Well, I've tried to replicate F-Droid extension, but have at least two issues:

1. I can't understand how do I get the fingerprint from `fdroid.nix`
2. `gradle2nix` fails for me. `gradle2nix -g 6.5` in aurora services project root spits this:

```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=on -Dswing.aatext=true -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel -Dswing.crossplatformlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel
Resolving project model: root project...
Downloading https://services.gradle.org/distributions/gradle-6.5-bin.zip
.................................................................................................

> Configure project :app
WARNING: DSL element 'annotationProcessorOptions.includeCompileClasspath' is obsolete.
It will be removed in version 5.0 of the Android Gradle plugin.
It does not do anything and AGP no longer includes annotation processors added on your project's compile classpath
Settings script
Subproject: :
Subproject: :app
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':app:debugAndroidTestCompileClasspath'.
> The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'. However we cannot choose between the following variants of project :app:
- Configuration ':app:debugApiElements' variant android-base-module-metadata declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-base-module-metadata' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
- Configuration ':app:debugApiElements' variant android-feature-all-metadata declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-feature-all-metadata' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
- Configuration ':app:debugApiElements' variant android-feature-res-ap_ declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-feature-res-ap_' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
- Configuration ':app:debugApiElements' variant android-feature-signing-config declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-feature-signing-config' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
- Configuration ':app:debugApiElements' variant android-java-res declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-java-res' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
- Configuration ':app:debugApiElements' variant android-manifest-metadata declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
- Unmatched attributes:
- Provides attribute 'artifactType' with value 'android-manifest-metadata' but the consumer didn't ask for it
- Provides attribute 'com.android.build.api.attributes.VariantAttr' with value 'debug' but the consumer didn't ask for it
[... cut ...]
at org.gradle.api.internal.artifacts.ivyservice.DefaultLenientConfiguration.getFirstLevelModuleDependencies(DefaultLenientConfiguration.java:173)
at org.gradle.api.internal.artifacts.ivyservice.DefaultLenientConfiguration.getFirstLevelModuleDependencies(DefaultLenientConfiguration.java:290)
at org.gradle.api.internal.artifacts.ivyservice.ErrorHandlingConfigurationResolver$ErrorHandlingLenientConfiguration.getFirstLevelModuleDependencies(ErrorHandlingConfigurationResolver.java:137)
at org.nixos.gradle2nix.ConfigurationResolver.resolve(ConfigurationResolver.kt:81)
... 111 more
```

pshirshov, 2021-09-10:
Well, I've managed to generate the env.

gradle2nix has to be invoked like `gradle2nix -c bundle -c lintVitalRelease`. In order to get a working gradle env I used the following  `shell.nix`:

```nix
{ pkgs ? import <nixpkgs> { config.android_sdk.accept_license = true; } }:

let
  buildToolsVersion = "30.0.3";

  androidComposition = pkgs.androidenv.composeAndroidPackages {
    toolsVersion = "26.1.1";
    platformVersions = [
      "30"
    ];
    buildToolsVersions = [ "30.0.3" ];
    includeExtras = [
    ];
  };

  fhs = pkgs.buildFHSUserEnv {
    name = "android-env";
    targetPkgs = pkgs: with pkgs;
      [
        git
        gitRepo
        gnupg
        python2
        curl
        procps
        openssl
        gnumake
        nettools
        androidenv.androidPkgs_9_0.platform-tools
        pkgs.androidenv.androidPkgs_9_0.androidsdk
        jdk
        schedtool
        utillinux
        m4
        gperf
        perl
        libxml2
        zip
        unzip
        bison
        flex
        lzop
        python3
      ];
    multiPkgs = pkgs: with pkgs;
      [
        zlib
        ncurses5
      ];
    runScript = "bash";
    profile = ''
      export ALLOW_NINJA_ENV=true
      export USE_CCACHE=1
      export ANDROID_JAVA_HOME=${pkgs.jdk.home}
      export LD_LIBRARY_PATH=/usr/lib:/usr/lib32
      export ANDROID_SDK_ROOT=${pkgs.androidenv.androidPkgs_9_0.androidsdk.out}
      export ANDROID_SDK_ROOT="${androidComposition.androidsdk}/libexec/android-sdk";
      export ANDROID_NDK_ROOT="${androidComposition.androidsdk}/libexec/android-sdk/ndk-bundle";
      export GRADLE_OPTS="-Dorg.gradle.project.android.aapt2FromMavenOverride=${androidComposition.androidsdk}/libexec/android-sdk/build-tools/${buildToolsVersion}/aapt2";
    '';
  };
in
pkgs.stdenv.mkDerivation {
  name = "android-env-shell";
  nativeBuildInputs = [ fhs ];
  shellHook = "exec android-env";

}
```

See also: https://github.com/tadfisher/gradle2nix/issues/13

Though that's not everything, the build still fails.

pshirshov, 2021-09-10:
Aha, seems like the right way to invoke that shite is `gradle2nix -c ':app:lintVitalRelease' -c ':app:packageRelease'`