nix-community/robotnix #135: Error when building GrapheneOS with signing enabled

hmenke, 2021-11-03:
Since the switch to triChrome by default the following build command throws an error.
```
nix-build --arg configuration '{
  flavor = "grapheneos";
  device = "sunfish";
  signing = {
    enable = true;
    keyStorePath = "/dev/shm/android-keys";
  };
 }' -A generateKeysScript -o generate-keys --show-trace
```
The error is
```
error (ignored): error: bad archive: input doesn't look like a Nix archive
error: getting status of '/dev/shm/android-keys/vanadium.x509.pem': No such file or directory

       … while evaluating the attribute 'buildCommand' of the derivation 'cert-fingerprint'

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/pkgs/stdenv/generic/make-derivation.nix:201:11:

          200|         // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          201|           name =
             |           ^
          202|             let

       … while evaluating 'certFingerprint'

       at /home/henri/Code/android/robotnix/pkgs/robotnix/default.nix:15:21:

           14|
           15|   certFingerprint = cert: (import (runCommand "cert-fingerprint" {} ''
             |                     ^
           16|     ${openssl}/bin/openssl x509 -noout -fingerprint -sha256 -in ${cert} | awk -F"=" '{print "\"" $2 "\"" }' | sed 's/://g' > $out

       … from call site

       at /home/henri/Code/android/robotnix/modules/apps/prebuilt.nix:180:20:

          179|             else if _config.signing.enable
          180|               then pkgs.robotnix.certFingerprint (putInStore "${evalTimeKeyPath config.certificate}.x509.pem") # TODO: IFD
             |                    ^
          181|             else # !_config.signing.enable

       … while evaluating the attribute 'value.content'

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/lib/modules.nix:708:14:

          707|     { _type = "override";
          708|       inherit priority content;
             |              ^
          709|     };

       … while evaluating the attribute 'value._type'

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/lib/modules.nix:648:73:

          647|       highestPrio = foldl' (prio: def: min (getPrio def) prio) 9999 defs;
          648|       strip = def: if def.value._type or "" == "override" then def // { value = def.value.content; } else def;
             |                                                                         ^
          649|     in {

       … while evaluating anonymous lambda

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/lib/modules.nix:547:19:

          546|           # Avoid sorting if we don't have to.
          547|           if any (def: def.value._type or "" == "order") defs''.values
             |                   ^
          548|           then sortProperties defs''.values

[... 297 lines cut ...]
          297|           g =
          298|             name: value:
             |                   ^
          299|             if isAttrs value && cond value

       … from call site

       … while evaluating the attribute 'buildCommand' of the derivation 'android-key-tools'

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/pkgs/stdenv/generic/make-derivation.nix:201:11:

          200|         // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          201|           name =
             |           ^
          202|             let

       … while evaluating the attribute 'text' of the derivation 'generate_keys.sh'

       at /nix/store/p3z5pr2p8cc5idpq4kfps187ml1y1fn4-source/pkgs/stdenv/generic/make-derivation.nix:201:11:

          200|         // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          201|           name =
             |           ^
          202|             let
```

danielfullmer, 2021-11-03:
I think this error applies to vanilla as well. Even this command tries to pull in information from `signing.keyStorePath`, which it shouldn't:
```
nix-build --arg configuration '{device="crosshatch"; flavor="vanilla"; signing.enable=true; }' -A generateKeysScript
```

hmenke, 2021-11-03:
I think this is the offender:

https://github.com/danielfullmer/robotnix/blob/ab6e5b96ad90725a94d10d0580d642bb369a75b2/modules/apps/chromium.nix#L58

it uses `fingerprint` which probably blow up when used here

https://github.com/danielfullmer/robotnix/blob/ab6e5b96ad90725a94d10d0580d642bb369a75b2/modules/apps/chromium.nix#L93-L104

danielfullmer, 2021-11-03:
@hmenke Since it's lazily evaluated, I don't think that code should be running at all.

A little further debugging shows that just trying to instantiate an entirely unrelated source dir would also trigger this: 
```
nix-instantiate --arg configuration '{device="crosshatch"; flavor="vanilla"; signing.enable=true; }' -A config.source.dirs.development.src
```

PR #136 (2021-11-04): Fix improper dependency of generateKeysScript on keyStorePath
Building `generateKeysScript` with a simple configuration of:

{device="crosshatch"; flavor="vanilla"; signing.enable=true; }

would result in Nix trying to import the key fingerprint for chromium.
This is shouldn't happen because the generateKeysScript should have no
data dependency on the actual APK that would be built.

The chain of dependencies was generateKeysScript -> keyTools ->
config.source.dirs -> nonNullPrebuilts (in modules/apps/prebuilts).

This nonNullPrebuilts would have to further evaluate the apk to decide
if it was [... cut]
commit f78f2cf2 (2021-11-04): Fix improper dependency of generateKeysScript on keyStorePath

Building `generateKeysScript` with a simple configuration of:

{device="crosshatch"; flavor="vanilla"; signing.enable=true; }

would result in Nix trying to import the key fingerprint for chromium.
This is shouldn't happen because the generateKeysScript should have no
data dependency on the actual APK that would be built.

The chain of dependencies was generateKeysScript -> keyTools ->
config.source.dirs -> nonNullPrebuilts (in modules/apps/prebuilts).

This nonNullPrebuilts would have to further evaluate the apk to decide
if it wa [... cut]
closed 2021-11-04 by danielfullmer