nix-community/robotnix #137: GrapheneOS android-prepare-vendor and GrapheneOS trademarks are not available to CalyxOS or their AOSP Alliance members

thestinger, 2021-11-11:
Our changes to android-prepare-vendor before Android 11 were under permissive licensing. For Android 12 and beyond our changes are explicitly not available for use by CalyxOS, their AOSP Alliance group and projects collaborating with them due to severe ongoing harassment, bullying and libel targeting our developers, particularly myself, which has included doxxing and threats mailed to my house. There has also been extensive misinformation repeatedly spread about GrapheneOS by CalyxOS leadership/developers and their inaccurate talking points are propagated by their community to harm the project across multiple platforms. 2 of the moderators in the CalyxOS room are heavily involved in severe bullying and libel along with direct involvement by the leader of the Calyx Institute. We can provide extensive screenshots and logs showing the ongoing harassment and abusive behavior. Message me on Matrix and I'll invite you to a group where that can be provided.

The android-prepare-vendor projects has never been under an open source license. The upstream project had no open source license and only a vague usage grant for the data. We're going to be using a full rewrite with much cleaner/simpler code for the Pixel 6, Pixel 6 Pro and beyond. The rewrite will be under a permissive license like the GrapheneOS code and our other standalone projects. You'll be free to use that even if you remain a member of AOSP Alliance.

CalyxOS used the AOSP Alliance GitHub project to spread libel and attacks on GrapheneOS. The libel spread as part of the AOSP Alliance README repository was rebased out but has been archived and continues to be used by the CalyxOS community to attack GrapheneOS. They've repeatedly done the same with the official chat room. Robotnix is currently at least a symbolic member of AOSP Alliance and is therefore not allowed to use the Android 12 port of android-prepare-vendor by GrapheneOS. It's up to you how you want to address that. If you leave AOSP Alliance, then we have no problem with you using it. You're also welcome to help us with rewriting it as a proper open source project with much cleaner code.

Our trademarks are in the process of being officially registered and are currently under common law protection. Those are also not available for use by CalyxOS or AOSP Alliance either.

You're welcome to use our our android-prepare-vendor code and our trademarks the way you're doing as soon as you leave AOSP Alliance. We have no problem with you, only your association with CalyxOS. We're very open to discussing it on Matrix and providing extensive details on what has been happening.

GrapheneOS is still entirely permissively licensed, but the script for extracting proprietary firmware was never truly open source and now has explicit restrictions for Android 12 and beyond. Going forward, our licensing is likely to include a statement about the attacks on the project by CalyxOS and Techlore without any legal meaning, but which must be included as part of the attribution requirement of the MIT license. This will be our way of deterring further usage by malicious groups harming GrapheneOS.

It's unfortunate that these actions need to be taken but the bullying targeting me and the misinformation being used to harm GrapheneOS by CalyxOS and their community has caused *substantial* harm to the project and disrupted development. Substantial actions are going to be taken against them and against groups collaborating with them. We'll be raising the abusive behavior with any conferences, organizations, open source projects, etc. where CalyxOS has any involvement.

danielfullmer, 2021-11-11:
I'm away on a business trip for the next couple days where I'll have limited access to the internet, but I'll respond to this Saturday/Sunday.

Atemu, 2021-11-13:
I can't speak for the project or drama but a few notes:

> Our trademarks are in the process of being officially registered and are currently under common law protection. Those are also not available for use by CalyxOS or AOSP Alliance either.

We might want to call it by a different name (g6eOS or something?) anyways then to avoid any potential trademark issues in the future.

> The android-prepare-vendor project has never been under an open source license. The upstream project had no open source license and only a vague usage grant for the data.

IANAL but from a legal perspective, that's a license, no matter what the original author thinks it is and a license to the effect of the WTFPL is a free open source one.

With your removal of that license in GrapheneOS/android-prepare-vendor@397adda2d5d0bfddca3f7774cb50b9757d00d697, it has become unfree software (unlike most FOSS licenses, you have no obligations to keep it). Robotnix currently uses a version from *after* the license change, so that needs to be remedied ASAP.

I don't think we should use unfree software in our build process, no matter what github organisations we belong to.

thestinger, 2021-11-13:
> We might want to call it by a different name (g6eOS or something?) anyways then to avoid any potential trademark issues in the future.

That sounds like a blatant trademark violation.

> IANAL but from a legal perspective, that's a license, no matter what the original author thinks it is and a license to the effect of the WTFPL is a free open source one.

There was never a license for the code. There was only a vague usage grant for the configuration data.

> With your removal of that license in GrapheneOS/android-prepare-vendor@397adda, it has become unfree software (unlike most FOSS licenses, you have no obligations to keep it). Robotnix currently uses a version from after the license change, so that needs to be remedied ASAP.

It was never free software. A vague usage grant for the configuration data doesn't make the code open source.

We're likely willing to add back that usage grant for our changes in a few weeks but it won't make the software open source.

> I don't think we should use unfree software in our build process, no matter what github organisations we belong to.

Please don't use derivatives of our code without attribution from CalyxOS / AOSP Alliance either. We're entirely willing to take legal action against projects using CalyxOS due to infringement on our copyright.

The reality is that we intend to deter AOSP Alliance and CalyxOS from using our code due to highly abusive behavior and the usage of the AOSP Alliance branding to cause harm to us. We're entirely capable of specifically mentioning AOSP Alliance in new licenses for future code. We can just as quickly drop that from the licensing to go back to the MIT license once it has the desired effect.

I'm not going to tolerate a group using our code which has been involved in relentless bullying / harassment targeting me largely by making baseless accusations that I'm delusional / crazy / schizophrenic and using that direct harassment towards me from neonazis, Kiwi Farms members, etc. who they welcome in their rooms. It's your choice if you want to be part of a GitHub organization which literally published libel about me in a Git commit to the README listing your project because we dared to call out their abusive behavior. They rebased that out since it was completely dishonest but it still happened and was archived by the people they have harassing us to continue using it. Any members of AOSP Alliance are hostile projects and will be treated as such which means finding ways to take legal action and respond in other ways to the ongoing harassment / attacks targeting us.

I care about this far more than I do about temporary clauses added to our licenses which may make a tiny minority of people upset. It will hardly impact anyone since I can make it extremely well targeted.

My well being and the well being of the other GrapheneOS developers matters more than whether there's a sentence in a text file saying that AOSP Alliance can't use our code. Zero problem with adding that if it's necessary.

Atemu, 2021-11-13:
> That sounds like a blatant trademark violation.

IANAL but from my understanding, it's the exact opposite. Calling it "GrapheneOS™" would obviously be one but something to the effect of g6eOS would not be covered by the trademark as it is a distinct name. A trademark only reserves the rights to a specific name, not the thing it refers to. Pretty sure close variations of a trademarked name that obviously try to mimic the real one (i.e. "GrephaneOS") could be considered a violation too.  
You can't trademark "anything that might refer to x".

> There was only a vague usage grant for the configuration data

That's a license.  

AFAIUI, even a vague "do wtf you want" constitutes a license. The [WTFPL](https://en.wikipedia.org/wiki/WTFPL) isn't much more than that and that's definitely a license, even if it's not intended to be 100% serious.

> It was never free software. A vague usage grant for the configuration data doesn't make the code open source.

The code was and is open source (unless you plan on changing that).  
I can openly access the source -> Open Source.

Whether it's free software depends on the definition but if you take something like the freedoms listed in the OSI's poorly named [open source definition](https://opensource.org/osd), you can see that the scrappy license ticks all the boxes.  
(Fun fact: The OSI didn't actually approve the WTFPL because it's a pretty bad license from a legal perspective and you'd be better off with one of the pseudo-public-domain licenses which have basically the same effect.)

> Please don't use derivatives of our code

Those derivatives would either be just as unfree or the result of a license violation, so that's out of the question anyways.

> We're entirely willing to take legal action against projects using CalyxOS due to infringement on our copyright.

I don't know or care about the drama and all that crap but please know that stuff like that should have no place in the FOSS community. Please don't be a dick.

I greatly appreciate you making us aware of the license violation before any needless escalation though.

> My well being and the well being of the other GrapheneOS developers

As I mentioned, I don't care for the drama one bit but let me give you this advice: Stay calm.

Without knowing (or wanting to know) any of the details, I can already tell that at least a part of the reason any of that happened is fueled by your behaviour. Just chill and reflect a bit perhaps.

danielfullmer, 2021-11-13:
@Atemu Let's try to deescalate this situation and not resort to any name-calling.

danielfullmer, 2021-11-13:
> You're welcome to use our our android-prepare-vendor code and our trademarks the way you're doing as soon as you leave AOSP Alliance. We have no problem with you, only your association with CalyxOS.

OK, I intend to withdraw from the AOSP alliance project.  It was a nice attempt for all these projects to collaborate on shared code, but it doesn't seem to have fulfilled the initial goals we had for it.

closed 2021-11-13 by danielfullmer