nix-community/robotnix #145: LOS 17.1 fails to build with µG: error: unknown library path for "org.apache.http.legacy.impl"

Atemu, 2021-12-30:
```Nix
{
  device = "cheeseburger";
  flavor = "lineageos";
  androidVersion = 10;

  apps.fdroid.enable = true;

  microg.enable = true;
}
```

```
[ 10% 10713/102271] RobotnixGmsCore dexpreopt gen
FAILED: out/target/product/cheeseburger/obj/APPS/RobotnixGmsCore_intermediates/dexpreopt.sh out/target/product/cheeseburger/obj/APPS/RobotnixGmsCore_intermediates/strip.sh
/bin/bash -c "out/soong/host/linux-x86/bin/dexpreopt_gen -global out/target/product/cheeseburger/dexpreopt.config -module out/target/product/cheeseburger/obj/APPS/RobotnixGmsCore_intermediates/dexpreopt.config       -dexpreopt_script out/target/product/cheeseburger/obj/APPS/RobotnixGmsCore_intermediates/dexpreopt.sh -strip_script out/target/
product/cheeseburger/obj/APPS/RobotnixGmsCore_intermediates/strip.sh     -out_dir out"
error: unknown library path for "org.apache.http.legacy.impl"
```

cheriimoya, 2021-12-31:
i also have this problem, if i set `microg.enable = false;` the build passes

danielfullmer, 2021-12-31:
This regression looks to be a result of https://github.com/danielfullmer/robotnix/commit/3a686acd8cf6117d8aab76bbded7a06753c8cf49 , which was a change needed for Android 12.  For older versions, lets just default to our previous behavior. CC @zhaofengli

commit d8f2cc30 (2021-12-31): modules/apps/prebuilt: Don't set USES_LIBRARIES for Android <= 10

Fixes #145
closed 2021-12-31 by danielfullmer