nix-community/robotnix #149: Support for older versions of LineageOS

jtrees, 2022-01-09:
Any chance you could add support for older versions of LineageOS? I'm interested in 15.1, specifically, because that's the latest version that supported [my device](https://wiki.lineageos.org/devices/huashan/).

samueldr, 2022-01-09:
Given LineageOS 15.1 maps to Android 8.1, it will require a bit more work as [there is no `modules/8` at the current time](https://github.com/danielfullmer/robotnix/tree/530b6500ebf69f628aeb29fb20996ff996eb8870/modules).

So, in addition to needing to add support for LineageOS 15.1 itself, it might require adding workarounds and quirks for the Android 8.1 build system.

Atemu, 2022-01-10:
Your device isn't supported on 15.1 either.

There are 16.0, 17.1 and 18.1 branches for it though, you could try to build those: https://github.com/LineageOS/android_device_sony_huashan/

They might just be usable or even stable.

jtrees, 2022-01-12:
> you could try to build those

Good point. I'll give it a go.

jtrees, 2022-01-29:
After fighting with this for a few hours, I'm out of ideas.

[Here](https://0bin.net/paste/6r0HyRbU#-vzL2RPmw20Jw/d9j9DO2Db1UPzbebEESO9RGYAtxa9) is my config. [This](https://0bin.net/paste/PbB6gjst#6aH0rptdfia0-EpiS8Gi3vUjdHyDYIuUFDXMChTqnSo) is the output I'm getting.

The (probably) relevant lines being:

```
[ 99% 475/476] finishing build rules ...
vendor/lineage/build/tasks/kernel.mk:134: warning: ***************************************************************
vendor/lineage/build/tasks/kernel.mk:135: warning: * Using prebuilt kernel binary instead of source              *
vendor/lineage/build/tasks/kernel.mk:136: warning: * THIS IS DEPRECATED, AND IS NOT ADVISED.                     *
vendor/lineage/build/tasks/kernel.mk:137: warning: * Please configure your device to download the kernel         *
vendor/lineage/build/tasks/kernel.mk:138: warning: * source repository to kernel/sony/msm8960t
vendor/lineage/build/tasks/kernel.mk:139: warning: * for more information                                        *
vendor/lineage/build/tasks/kernel.mk:140: warning: ***************************************************************
[100% 476/476] writing build rules ...
out/build-lineage_huashan-package.ninja is missing, regenerating...
[ 99% 477/479] initializing packaging system ...
[ 99% 478/479] including distdir.mk ...
[100% 479/479] writing packaging rules ...
16:32:05 Allowing all environment variables during ninja; incremental builds may be unsafe.
Starting ninja...
FAILED: ninja: 'out/target/product/huashan/kernel', needed by 'out/target/product/huashan/recovery.img', missing and no known rule to make it
16:32:06 ninja failed with: exit status 1
```

So... it's not building the kernel and can't find a pre-built one?

I also tried setting `kernel.enable = true;` but that led me down a whole new rabbit hole of (undocumented) options and requirements I was unable to satisfy (or understand for that matter).

danielfullmer, 2022-01-29:
It looks like you are still missing additional `source.dirs` dependencies. The `lineage.dependencies` file in each repo tells what additional repos need to be included in the build. For example: https://github.com/AdrianDC/android_device_sony_huashan/blob/lineage-18.1/lineage.dependencies
You can manually go add those repos to your config, but I also wanted to point out another script that might be helpful.

The `update_device_dirs.py` script under `flavors/robotnix/lineageos.py` does recursively walk these dependencies to fetch all the required repos. Normally this script will fetch absolutely every supported lineageos device for a particular version, but there is an option to have it fetch sources for a single device:
I tried:
```shell
$ nix-shell
$ cd flavors/lineageos
$ ./update_device_dirs.py --branch lineage-18.1 sony_huashan
```
and it fetched a few additional repos: https://gist.github.com/danielfullmer/e2d15436820b50fa7d688de4c2ee3d4d (the others are already in present in `device-dirs.json`.  However, this script is only set up currently to work with repos from https://github.com/LineageOS, while yours are from https://github.com/AdrianDC/android_device_sony_huashan .  A great improvement would be to make this script work well with repos from other sources.

The `kernel.*` options are not in a great state, Like you noticed, they are undocumented, and they likely only work on the `vanilla` flavor. (with maybe the exception of `kernel.patches`)

jtrees, 2022-01-29:
Thanks for the detailed reply. I was able to add some hacks to the script to get [this `device-dirs.json`](https://0bin.net/paste/V-unVAzV#TSytY1+OpQzRKGyX4IpbeeICQR+j839-JasloPPQ8UP). Still no luck though. The output is the same.

samueldr, 2022-01-29:
Here's an example of a device customizing `source.dirs`, in case it helps anyone.

 - https://github.com/samueldr-android-shenanigans/configuration/blob/e46a1862b3ed38c2c1759dad52a019cc7ca0c7b6/lineageos-begonia.nix#L30-L123

jtrees, 2022-01-29:
Yay! I was able to build something. Thanks to both of you for your support.

I have yet to test this (will probably have to set up signing first) but I'm already thrilled!

Here is my final config:

```nix
{ pkgs, ... }:

let
  fetchRepo = { source, repo, rev, sha256 }: pkgs.fetchgit {
    inherit rev sha256;
    url = "https://github.com/${source}/${repo}.git";
    deepClone = false;
    fetchSubmodules = false;
    leaveDotGit = false;
  };
in {
  device = "huashan";
  flavor = "lineageos";
  variant = "userdebug";
  androidVersion = 11; # lineageos-18.1

  signing.enable = false;

  apps.fdroid.enable = true;
  microg.enable = false;

  source.dirs = {
    # primary huashan repo
    "device/sony/huashan".src = fetchRepo {
      source = "AdrianDC";
      repo = "android_device_sony_huashan";
      rev = "2ed5b4211b80db1b0e6da8fffc830456182e874a";
      sha256 = "0npnag5yhnhn46vcgqb5sxqk26vnbirgc4wxxksb3bf5hfjir3lg";
    };

    # dependencies
    "device/sony/common".src = fetchRepo {
      source = "LineageOS";
      repo = "android_device_sony_common";
      rev = "c5ec35885b267c94c5db9abef675f096a9e6b826";
      sha256 = "0ynq2ql40icgi9cbqkxnk9ygm947g7ps49qs39xcvjps24bkxfvr";
    };
    "hardware/sony/DASH".src = fetchRepo {
      source = "LineageOS";
      repo = "android_hardware_sony_DASH";
      rev = "aa73c0cbf275ed66e1873482c8c3652e7e432171";
      sha256 = "0rawgnnwmhq2nc9wvx63s77974dm81mm7j1gfzkf66hxc36cpij1";
    };
    "hardware/sony/thermanager".src = fetchRepo {
      source = "LineageOS";
      repo = "android_hardware_sony_thermanager";
      rev = "3fb6013a7f4aeb50b297acb57a1500c7cf22f68b";
      sha256 = "1dvzhan93cdq74nlhc0s6nxjpjnpby0p80zg3qf7insplq3cid54";
    };
    "kernel/sony/huashan".src = fetchRepo {
      source = "AdrianDC";
      repo = "android_kernel_sony_msm8960t";
      rev = "d3d2d9afffb771f8c50da45bc9083efdfeecf59d";
      sha256 = "10a04r5c9jvkkqb7758hajjdv9fa4gl9mbmg8nkb50vxc5nmd925";
    };
    "packages/resources/devicesettings".src = fetchRepo {
      source = "LineageOS";
      repo = "android_packages_resources_devicesettings";
      rev = "6b4da8085c71642b11d3db9414b7308e4bbcd3b5";
      sha256 = "08nfzmqnwd4mvlchl57jsc1ky8kgrswhcy8fy09a8an5fdx3gw6w";
    };
    "vendor/sony".src = fetchRepo {
      source = "AdrianDC";
      repo = "proprietary_vendor_sony";
      rev = "9f5a81ba2ec1a3f95e5869678ee7ea0fedbc794a";
      sha256 = "04shfkw0sww83phyhbsifjg9r2ddz6f3m8dz5ig4rhn8dk0cax98";
    };
  };
}
```

I was also using modified `device-dirs.json` and `device-metadata.json` files but I think it should work without those changes. Will give it a go another day.

danielfullmer, 2022-01-29:
Happy to hear it built at least!

closed 2022-01-29 by danielfullmer

jtrees, 2022-01-31:
Follow-up: It runs! Thanks so much.

For reference...

Turns out I _did_ need the following changes. Not exactly sure why, since I thought providing the dirs via `source.dirs` would make it redundant:

```diff
diff --git a/flavors/lineageos/device-metadata.json b/flavors/lineageos/device-metadata.json
index 908f571..e819335 100644
--- a/flavors/lineageos/device-metadata.json
+++ b/flavors/lineageos/device-metadata.json
@@ -671,6 +671,13 @@
     "variant": "userdebug",
     "vendor": "oneplus"
   },
+  "huashan": {
+    "branch": "lineage-18.1",
+    "lineage_recovery": false,
+    "name": "Xperia SP",
+    "variant": "userdebug",
+    "vendor": "sony"
+  },
   "instantnoodle": {
     "branch": "lineage-18.1",
     "lineage_recovery": true,
diff --git a/flavors/lineageos/lineage-18.1/device-dirs.json b/flavors/lineageos/lineage-18.1/device-dirs.json
index 086f7fd..9effc9c 100644
--- a/flavors/lineageos/lineage-18.1/device-dirs.json
+++ b/flavors/lineageos/lineage-18.1/device-dirs.json
@@ -3195,6 +3195,24 @@
     "sha256": "15mcnvkgzkj0dvvvyfsavddrb1jhv8gbg79mhpl0kiis57mhvhjr",
     "url": "https://github.com/LineageOS/android_device_sony_discovery"
   },
+  "device/sony/huashan": {
+    "date": "2021-01-22T23:10:55+01:00",
+    "deepClone": false,
+    "deps": [
+      "device/sony/common",
+      "kernel/sony/msm8960t",
+      "hardware/sony/DASH",
+      "hardware/sony/thermanager",
+      "packages/resources/devicesettings"
+    ],
+    "fetchLFS": false,
+    "fetchSubmodules": false,
+    "leaveDotGit": false,
+    "path": "/nix/store/y010afldlsqfzxfz11c9p8mdvac7imsh-android_device_sony_huashan",
+    "rev": "2ed5b4211b80db1b0e6da8fffc830456182e874a",
+    "sha256": "0npnag5yhnhn46vcgqb5sxqk26vnbirgc4wxxksb3bf5hfjir3lg",
+    "url": "https://github.com/AdrianDC/android_device_sony_huashan"
+  },
   "device/sony/kirin": {
     "date": "2021-04-22T13:30:44+02:00",
     "deepClone": false,
@@ -4289,6 +4307,18 @@
     "sha256": "14pvd205l1p2cklla5nn8m2qyq9x133bii4skdd02kmrb2iy5r8a",
     "url": "https://github.com/LineageOS/android_hardware_samsung_slsi_scsc_wifibt_wpa_supplicant_lib"
   },
+  "hardware/sony/DASH": {
+    "date": "2020-08-13T16:42:21+02:00",
+    "deepClone": false,
+    "deps": [],
+    "fetchLFS": false,
+    "fetchSubmodules": false,
+    "leaveDotGit": false,
+    "path": "/nix/store/2xswnckzhhk96lp93zp3rvvk0i957w29-android_hardware_sony_DASH",
+    "rev": "aa73c0cbf275ed66e1873482c8c3652e7e432171",
+    "sha256": "0rawgnnwmhq2nc9wvx63s77974dm81mm7j1gfzkf66hxc36cpij1",
+    "url": "https://github.com/LineageOS/android_hardware_sony_DASH"
+  },
   "hardware/sony/SonyOpenTelephony": {
     "date": "2021-08-21T04:06:27+03:00",
     "deepClone": false,
@@ -5178,6 +5208,18 @@
     "sha256": "04ph1wbi5gkdjjknkp19hf2lki7n99g5kyws6nmw2m2lhcz10q8p",
     "url": "https://github.com/LineageOS/android_kernel_shift_sdm845"
   },
+  "kernel/sony/msm8960t": {
+    "date": "2021-01-24T22:08:34+01:00",
+    "deepClone": false,
+    "deps": [],
+    "fetchLFS": false,
+    "fetchSubmodules": false,
+    "leaveDotGit": false,
+    "path": "/nix/store/vgplarhglm7f7mcv807x7c2h96l6p35i-android_kernel_sony_msm8960t",
+    "rev": "d3d2d9afffb771f8c50da45bc9083efdfeecf59d",
+    "sha256": "10a04r5c9jvkkqb7758hajjdv9fa4gl9mbmg8nkb50vxc5nmd925",
+    "url": "https://github.com/AdrianDC/android_kernel_sony_msm8960t"
+  },
   "kernel/sony/msm8974": {
     "date": "2021-09-27T23:45:46+02:00",
     "deepClone": false,
```

I built it with this command:

```sh
nix-build --arg configuration ./huashan.nix -A ota
```

... and then flashed it via TWRP (obtained [here](https://dl.twrp.me/huashan/)).

samueldr, 2022-02-01:
It's perfectly fine to flash via TWRP.

If you wanted a fully-build-with-robotnix install flow, first install can use the `bootImg` or `recoveryImg` (depending on A/B or not) output. This will give you the boot image, or the recovery, with which you can boot to recovery. Just like how upstream LineageOS recommends their own recovery :).

jtrees, 2022-02-01:
Thanks. I'll give that a go eventually.