nix-community/robotnix #153: Error while loading shared libraries: libz.so.1

robbins, 2022-01-18:
I initialized and synced the latest AOSP 12 tag: `repo init -u https://android.googlesource.com/platform/manifest -b android-12.0.0_r26`

A build environment was then created with this command:

`nix-build "https://github.com/danielfullmer/robotnix/archive/master.tar.gz" \
    --arg configuration '{ device="raven"; flavor="vanilla"; }' \
    -A config.build.env`
and used `./result/bin/robotnix-build` to enter it.

I then ran `. build/envsetup.sh`, `lunch aosp_x86_64-eng`, and then `make`.

The following error is then printed:
`FAILED: out/target/product/generic_x86_64/obj/EXECUTABLES/ip-up-vpn_intermediates/ip-up-vpn.o
/bin/bash -c "PWD=/proc/self/cwd  prebuilts/clang/host/linux-x86/clang-r416183b1/bin/clang -I frameworks/native/cmds/ip-up-vpn -I out/target/product/generic_x86_64/obj/EXECUTABLES/ip-up-vpn_intermediates -I out/target/product/generic_x86_64/gen/EXECUTABLES/ip-up-vpn_intermediates -Isystem/core/libcutils/include -Isystem/core/libprocessgroup/include -Isystem/logging/liblog/include -Iexternal/libcxx/include -Iexternal/libcxxabi/include -D__LIBC_API__=10000 -Ibionic/libc/async_safe/include -Isystem/logging/liblog/include -Ibionic/libc/system_properties/include -Isystem/core/property_service/libpropertyinfoparser/include -isystem bionic/libc/include -isystem bionic/libc/kernel/uapi -isystem bionic/libc/kernel/android/scsi -isystem bionic/libc/kernel/android/uapi -isystem bionic/libc/kernel/uapi/asm-x86 -D__LIBM_API__=10000 -D__LIBDL_API__=10000 -isystem bionic/libc/include -isystem bionic/libc/kernel/uapi -isystem bionic/libc/kernel/android/scsi -isystem bionic/libc/kernel/android/uapi -isystem bionic/libc/kernel/uapi/asm-x86 -Iexternal/libcxxabi/include   -I system/core/include -I system/logging/liblog/include -I system/media/audio/include -I hardware/libhardware/include -I hardware/libhardware_legacy/include -I hardware/ril/include -I frameworks/native/include -I frameworks/native/opengl/include -I frameworks/av/include  -c  -Werror=implicit-function-declaration -DANDROID -fmessage-length=0 -W -Wall -Wno-unused -Winit-self -Wpointer-arith -Wunreachable-code-loop-increment -no-canonical-prefixes -DNDEBUG -UDEBUG -fno-exceptions -Wno-multichar -O2 -g -fdebug-info-for-profiling -fno-strict-aliasing -Werror=date-time -Werror=pragma-pack -Werror=pragma-pack-suspicious-include -Werror=string-plus-int -Werror=unreachable-code-loop-increment -fdebug-prefix-map=/proc/self/cwd= -D__compiler_offsetof=__builtin_offsetof -faddrsig -fcommon -Werror=int-conversion -fexperimental-new-pass-manager -Wno-reserved-id-macro -Wno-unused-command-line-argument -fcolor-diagnostics -Wno-sign-compare -Wno-defaulted-function-deleted -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-gnu-folding-constant -Wunguarded-availability -D__ANDROID_UNAVAILABLE_SYMBOLS_ARE_WEAK__ -ftrivial-auto-var-init=zero -enable-trivial-auto-var-init-zero-knowing-it-will-be-removed-from-clang -ffunction-sections -fdata-sections -fno-short-enums -funwind-tables -fstack-protector-strong -Wa,--noexecstack -D_FORTIFY_SOURCE=2 -Wstrict-aliasing=2 -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Werror=format-security -nostdlibinc -m64 -march=x86-64 -mssse3 -msse4 -msse4.1 -msse4.2 -mpopcnt -Bprebuilts/gcc/linux-x86/x86/x86_64-linux-android-4.9/x86_64-linux-android/bin   -std=gnu99    -Wall -Werror -fPIE -DANDROID_STRICT -target x86_64-linux-android10000   -Werror=bool-operation -Werror=implicit-int-float-conversion -Werror=int-in-bool-context -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -Werror=string-compare -Werror=xor-used-as-pow -Wno-void-pointer-to-enum-cast -Wno-void-pointer-to-int-cast -Wno-pointer-to-int-cast -Werror=fortify-source -Werror=address-of-temporary -Werror=return-type -Wno-tautological-constant-compare -Wno-tautological-type-limit-compare -Wno-reorder-init-list -Wno-implicit-int-float-conversion -Wno-int-in-bool-context -Wno-sizeof-array-div -Wno-tautological-overlap-compare -Wno-deprecated-copy -Wno-range-loop-construct -Wno-misleading-indentation -Wno-zero-as-null-pointer-constant -Wno-deprecated-anon-enum-enum-conversion -Wno-deprecated-enum-enum-conversion -Wno-string-compare -Wno-enum-enum-conversion -Wno-enum-float-conversion -Wno-pessimizing-move -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -MD -MF out/target/product/generic_x86_64/obj/EXECUTABLES/ip-up-vpn_intermediates/ip-up-vpn.d -o out/target/product/generic_x86_64/obj/EXECUTABLES/ip-up-vpn_intermediates/ip-up-vpn.o frameworks/native/cmds/ip-up-vpn/ip-up-vpn.c"
prebuilts/clang/host/linux-x86/clang-r416183b1/bin/clang.real: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory`

I also experienced this same error https://0bin.net/paste/RFfDBkV-#UTTtin0a-tzryH6rpeQJPy8j3dj+BkJKZt5neJxLzeN with the Paranoid Android ROM, which is also Android 12 (https://github.com/AOSPA/manifest).

I do see libz.so in /usr/lib inside the FHS env.

danielfullmer, 2022-01-18:
Try `export ALLOW_NINJA_ENV=true`. We do not currently (but should) set the
environment variables (`envVars` option) in `config.build.env`.

robbins, 2022-01-18:
That did the trick!

robbins, 2022-01-18:
That did the trick!

closed 2022-01-18 by robbins