nix-community/robotnix #158: Install Nix

Atemu, 2022-01-23:
We obviously build the image with Nix and all but, after that, we "only" have a regular Android image.

Wouldn't it be cool if we could install the Nix package manager to the image? Android has a Linux kernel, filesystem and a shell which is basically all that's required to use Nix.  
There are ways of using Nix on Android (nix-on-droid) but they're very limited and more "proper" solutions would require root access.

If we had `/nix/` and a `nix-daemon`, that'd be *so* useful.

Not only would such an option allow you to use Nix on your phone, it'd allow Nix to become your phone, similar to NixOS. In a future where modular builds are possible, parts of the OS could be interchanged just like an updated package on NixOS. This could allow for OS upgrades with no flashing from the recovery.

hmenke, 2022-01-23:
Related: **Nix-on-Droid**

GitHub: https://github.com/t184256/nix-on-droid
Prebuilt APKs: https://nix-on-droid.unboiled.info/
F-Droid: https://f-droid.org/en/packages/com.termux.nix/

Atemu, 2022-01-23:
> There are ways of using Nix on Android (nix-on-droid) but they're very limited (...)

MagicRB, 2022-02-03:
You can install nix on any rom with

```
 unset LD_PRELOAD
 export USER=$(id -un) HOME=$PWD GROUP=$(id -g)
 
 sudo mount -o remount,rw /
 # do the same for /etc but the specifics vary, mines on tmpfs but only from android 11 and up
 
 sudo mkdir /nix /data/nix
 sudo mount -o bind /data/nix /nix
 sudo chown $(id -u):$(id -g) -R /nix
 
 # create /etc/resolv.conf however you want, i recommend
 su
 echo "nameserver 8.8.8.8" > /etc/resolv.conf
 exit
 
 # remount all back to ro
 
 curl -L https://nixos.org/nix/install | sh
 
 # then to get into the default nix profile
 . nix-profile/etc/profile.d/nix.sh
 
 # also the environment variable edits must be done before sourcing nix.sh or it will silently do nothing
should be all. from here install home manager as normal
```

hmenke, 2022-02-03:
> install nix on any rom
> `sudo`

Yeah, I don't think so.

MagicRB, 2022-02-03:
Well im assuming you have it rooted

Atemu, 2022-02-04:
I'm aware that it's possible to modify the OS imperatively to include Nix but this is about baking Nix into the Android image declaratively.

MagicRB, 2022-02-04:
I realize that, but given that, that script works it's proof that it's possible and it may help someone

CyberShadow, 2024-03-11:
Let's define the problem a little more:

- The goal is to have Nix running directly on top of the Android/Linux kernel.
- So, VMs are out.
- Things like Termux, which run everything as the app user, and emulate root with things like proot, are out.
- Things like Linux Deploy, which run a Linux userspace under a chroot / mount namespace are out.
- We probably want Nix to be accessible from more than just one system user, and not just as root, so single-user mode is out.
- Android does not have an init or service manager which supports socket activation, so that's out.

That gives us the following picture:

- A robotnix module which adds Nix to an image, and starts the Nix daemon in multi-user mode on boot.
- `/nix` is created and bind-mounted to `/data/nix`, to allow the system partition to stay read-only.

What does this give us?

- Access Nix from a terminal (or `adb shell`).
- Sandbox support, if the kernel has the right namespacing features. (We should enable said features in our kernel builds.)
- Hopefully, allow using a good part of nixpkgs.
- Theoretically, build derivations that produce `.apk` files, and install them.
- Maybe allow apps to use Nix, e.g. an F-Droid-like package manager that builds derivations locally instead of downloading opaque `.apk` files.

Open questions:

- How to define who has access to the Nix daemon? Could it be done as a new Android permission?