nix-community/robotnix #162: Documentation on encryption is unclear and encryption options are confusing

Atemu, 2022-02-08:
I want to finally migrate off test-keys now that µG supports SafetyNet and Magisk has a better hide mechanism.

Unfortunately, the signing setup of robotnix isn't very user-friendly. I want the `releaseScript` variant, so I built a `releaseScript` of my config.

This requires `signing.enable = true;` which is sensible but from there on, things are less clear. The first thing robotnix complains about is that `signing.keyStorePath` is undefined. Since I chose the `releaseScript` method, I don't want to sign from inside the drv, so this is extremely confusing.

Through experimentation, I later found out it needs access to the *certificates*, not the (private) signing keys. These should be separate options to reflect that fact. I don't mind putting (public) certificates in the Nix store.

The need for the certs can be alleviated by setting fingerprints on the prebuilt apps which the docs do hint at but they're very unclear on it.

Related: https://github.com/danielfullmer/robotnix/issues/24

hmenke, 2022-02-08:
Currently `signing.keyStorePath` is needed to determine the key fingerprints via IFD. You can do what Daniel and I and probably others are doing and hardcode the public key fingerprints in the Nix file.

https://github.com/danielfullmer/robotnix-personal/blob/ef8cd1f379e153f66aa89e9fc9362a693537187b/flake.nix#L30-L35

Atemu, 2022-02-08:
> The need for the certs can be alleviated by setting fingerprints on the prebuilt apps which the docs do hint at but they're very unclear on it.

mannp, 2022-02-17:
> Currently `signing.keyStorePath` is needed to determine the key fingerprints via IFD. You can do what Daniel and I and probably others are doing and hardcode the public key fingerprints in the Nix [...]

Can you clarify if this fingerprint is fixed for f-droid, or we have to get the fingerprint of the f-droid cert created by the generateKeysScript?

Kranzes, 2022-02-17:
I never specified any fingerprint for fdroid when i used fdroid, i think it just used the one from generateKeysScript

mannp, 2022-02-17:
> I never specified any fingerprint for fdroid when i used fdroid, i think it just used the one from generateKeysScript

I see thanks, so what is the purpose of this if it builds without it?

`apps.prebuilt.F-Droid.fingerprint = lib.mkIf config.signing.enable "440B1449D705B85191E427C1ACF245B48854CACF1240AA358F15E4D022BA4A7F";`

hmenke, 2022-02-17:
You can extract the fingerprint using
```bash
openssl x509 -noout -fingerprint -sha256 -in ./keys/f-droid.x509.pem | cut -d '=' -f 2 | tr -d ':'
```
Once you have hardcoded all fingerprints you can run with `--option allow-import-from-derivation false` which gets you one step closer to pure evaluation mode.

mannp, 2022-02-17:
Thanks, that makes sense now 👍🏻