nix-community/robotnix #163: releaseScript implies the updater URL to be declared

Atemu, 2022-02-08:
You cannot build a `releaseScript` without specifying `apps.updater.url`, even if you don't want to use the updater functionality.

My situation is the same as described in https://github.com/danielfullmer/robotnix/issues/162; I want a signed OTA zip that I can flash without exposing signing keys to a Nix build.

danielfullmer, 2022-02-10:
This is only the case for the `lineageos` updater. CC @ajs124

Incoming commit will fix this to have the releaseScript only output updater metadata if `apps.updater.enable = true;`

closed 2022-02-10 by danielfullmer

ajs124, 2022-02-10:
The fix is in 6c1faa47bede034724aeed6257ea2e91cc6ce1f6, right?

> I want a signed OTA zip that I can flash without exposing signing keys to a Nix build.

should still have been possible, but `apps.updater.enable` needed to be `true`, which is obviously less than ideal. thanks for fixing this @danielfullmer