nix-community/robotnix #166: Unable to build z3c lineageos ota with flake config.

mannp, 2022-02-15:
Tried multiple times to build a lineageos build for the z3c device, which I believe is supported, but the build finishes, but does not appear to produce any ota zip file.

I cannot find a log file? So this is the last lines of the terminal output.

After a number of days trying to get this to work, I'd welcome any help to understand what I am doing wrong.

Thanks in advance.

Config flake.nix below;

```
{
  description = "Robotnix configuration";

  nixConfig.extra-substituters = [ "https://robotnix.cachix.org" ];
  nixConfig.extra-trusted-public-keys = [ "robotnix.cachix.org-1:+y88eX6KTvkJyernp1knbpttlaLTboVp4vq/b24BIv0=" ];

  inputs = {
    robotnix.url = "github:danielfullmer/robotnix";
    hosts = { url = "github:StevenBlack/hosts"; flake = false; };
  };

  outputs = { self, robotnix, nixpkgs, ... }@inputs:
    let
      common = { config, pkgs, lib, ... }: {
        signing.enable = true;
        signing.keyStorePath = "/run/media/local/1000G/Android/keys";
        buildDateTime = 1644861708;

        apps = {
          bromite.enable = false;
          chromium.enable = false;
          updater.enable = true;
          updater.url = "https://android.me.local/android/";
        };

        webview = {
          chromium.enable = false;
          chromium.availableByDefault = false;
          bromite.enable = true;
          bromite.availableByDefault = true;
        };

        microg.enable = true;

        #hosts = inputs.hosts + "/hosts";
        hosts = pkgs.fetchFromGitHub {
            owner = "StevenBlack";
            repo = "hosts";
            rev = "3.9.46";
            sha256 = "sha256-IkEdqJTEHDYW5mE4Nw5xhPuDif39yQeToO7ywO7BG90=";
        } + "/hosts";
      };
    in
    {
      robotnixConfigurations."z3c" = robotnix.lib.robotnixSystem ({ config, pkgs, ... }: {
        imports = [ common ];
        device = "z3c";
        flavor = "lineageos";
        androidVersion = 11;
      });

      defaultPackage.x86_64-linux = self.robotnixConfigurations."z3c".otaDir;
    };
}
```

Terminal output;

```
z3c-signed_target_files>     signing: SimpleDeviceConfig.apk                            (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/platform)
z3c-signed_target_files>     signing: Snap.apk                                          (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/platform)
z3c-signed_target_files>     signing: StorageManager.apk                                (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/releasekey)
z3c-signed_target_files>     signing: SystemUI.apk                                      (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/platform)
z3c-signed_target_files>     signing: ThemePicker.apk                                   (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/releasekey)
z3c-signed_target_files>     signing: TrebuchetQuickStep.apk                            (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/releasekey)
z3c-signed_target_files>     signing: WallpaperCropper.apk                              (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/platform)
z3c-signed_target_files> Rewriting SYSTEM/vendor/build.prop:
z3c-signed_target_files>   replace:  ro.vendor.build.tags=test-keys
z3c-signed_target_files>      with:  ro.vendor.build.tags=release-keys
z3c-signed_target_files> Rewriting SYSTEM/vendor/etc/selinux/vendor_mac_permissions.xml with new keys.
z3c-signed_target_files>     signing: framework-res__auto_generated_rro_vendor.apk      (/dev/shm/robotnix_keys.XXXXyg2r61/z3c/releasekey)
z3c-signed_target_files> META/otakeys.txt has no keys; using /dev/shm/robotnix_keys.XXXXyg2r61/z3c/releasekey.x509.pem for OTA package verification.
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read SYSTEM/etc/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read VENDOR/etc/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read VENDOR/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read PRODUCT/etc/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read PRODUCT/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read SYSTEM_EXT/etc/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read SYSTEM_EXT/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read ODM/etc/build.prop
z3c-signed_target_files> 2022-02-15 19:50:34 - common.py - WARNING : Failed to read ODM/build.prop
z3c-signed_target_files> done.
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read SYSTEM/etc/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read VENDOR/etc/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read VENDOR/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read PRODUCT/etc/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read PRODUCT/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read SYSTEM_EXT/etc/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read SYSTEM_EXT/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read ODM/etc/build.prop
z3c-ota_update> 2022-02-15 19:54:32 - common.py - WARNING : Failed to read ODM/build.prop
z3c-ota_update> Compressing system.new.dat with brotli
```

zhaofengli, 2022-02-15:
I don't see any obvious problem in the log lines you pasted, but perhaps you ran out of memory and the compression process got killed (the kill should be visible in `dmesg`). There should be a line saying `error: builder for '/nix/store/something.drv' failed with exit code X`, and you can then run `nix log /nix/store/something.drv` to get the full log.

mannp, 2022-02-15:
32gig machine doing the build and nothing in dmesg for kill, compress, robotnix, error, brotli .... anything else it might be logged as?

I'd read the lineage ota updater was a little shaky with robotnix, but there was no error message to direct me to any log.

zhaofengli, 2022-02-15:
If it succeeds it should produce a `result` symlink to the store in the current directory.

mannp, 2022-02-15:
There is a result link, but it appears broken.

```
 tree -l result
result [error opening dir]

```

lrwxrwxrwx  1 local users    54 Feb 15 19:57 result -> /nix/store/rk92chirrc6l6s1n3vywp8ybzisb0vvg-z3c-otaDir

```
result
Link (broken) (inode/symlink)
/nix/store/rk92chirrc6l6s1n3vywp8ybzisb0vvg-z3c-otaDir
```

Kranzes, 2022-02-16:
Are you using a different nix store using --store?

closed 2022-04-18 by mannp