nix-community/robotnix #168: Builds failing with fdroid included and fdroid certs creation

mannp, 2022-02-17:
Having trouble with builds including f-droid;

```
z3c-signed_target_files> Missing f-droid key
z3c-signed_target_files> Certain keys were missing from KEYSDIR. Have you run generateKeysScript?
z3c-signed_target_files> Additionally, some robotnix configuration options require that you re-run
z3c-signed_target_files> generateKeysScript to create additional new keys. This should not overwrite
z3c-signed_target_files> existing keys.
```

This is the snippet from my flake config;

```
        signing.enable = true;
        apps = {
          bromite.enable = false;
          chromium.enable = false;
          updater.enable = true;
          updater.url = "https://${myDomain}/android/";

          seedvault.enable = true;
          fdroid.enable = true;

          prebuilt.F-Droid.fingerprint = lib.mkIf config.signing.enable "440B1449D705B85191E427C1ACF245B48854CACF1240AA358F15E4D022BA4A7F";
        };
```

`nix build --store "/run/media/local/1000G/Android/store" .#robotnixConfigurations.z3c.generateKeysScript`

I also get an error in the generate-keys script trying to create a cert for f-droid? 

```

Skipping generating com.android.hotspot2.osulogin key since it is already exists
Skipping generating com.android.wifi.resources key since it is already exists
Generating f-droid key
./generate-keys.sh: line 24: make_key: command not found
Skipping generating microg key since it is already exists
Skipping generating bromitewebview key since it is already exists
```

Kranzes, 2022-02-17:
I believe the command not found is because inside the generateKeysScript it doesn't refer to the store on the other location.

mannp, 2022-02-17:
> I believe the command not found is because inside the generateKeysScript it doesn't refer to the store on the other location.

It's hard coded, as the other certs are created fine using the alternate store...

mannp, 2022-02-17:
Resolved, thanks @Kranzes

closed 2022-02-17 by mannp