nix-community/robotnix #169: Parallelise fetchers

Atemu, 2022-03-18:
I just tried updating LOS but GH is painfully slow at serving the repos sparsely (one kernel source was in the KiB/s...).

The scripts need to be parallelised, especially now that we've got 3 "sub-flavours" to boot.

Is there an easy way to parallelise a for loop in python?

Kranzes, 2022-03-18:
Is this an issue mk-repo or  just the device/vendor fetchers?

Atemu, 2022-03-18:
The metadata fetcher is so fast that it's not worth parallelising. Though it'd be nice if the repo username check could be done away with.

The device/vendor dir updater is the bug issue.

Atemu, 2022-03-28:
Ah, just noticed #175 doesn't actually fix this issue as it "only" parallelises the metadata/repo fetch, not the updating of the LOS repos.

PR #175 (2022-03-25): mk_repo_file: Parallelize fetch
Got impatient waiting for the 12.1 tags to sync. I didn't time the before/after runs but verified that it was indeed parallelized.

Fixes #169.
commit fb09bc99 (2022-03-25): mk_repo_file: Parallelize fetch

Fixes #169.