nix-community/robotnix #174: Auto-start adb with trusted keys

Atemu, 2022-03-22:
So this is a fairly specific problem I'm having but might have general usefulness.

I've got a friend's cheeseburger where the screen is almost entirely broken. Glass shattered, no image and only a portion of the screen registers touch.

Bootloader is unlocked though and I can control the device via adb, fastboot and/or twrp.

However, I cannot trivially control android itself. I need adb to be enabled in order to use use scrcpy to do anything, including enabling adb.  
This can be solved with these steps: https://gist.github.com/niikoo/3f6bd13a69f2d68f3dd51cc667e79bdc (property step is unnecessary)

You basically need to set some system build.props and push your public key; akin to enabling ssh and pushing a trusted key.

Is it possible to build these into the system image for simplicity?

Atemu, 2022-03-22:
Oh, we even have a `variant` option for that! I'll try that tomorrow.

I'd still kind of like to have a regular userdebug build but with adb running.

zhaofengli, 2022-03-22:
I do this to embed my adb vendor keys:

```
  product.extraConfig = ''
    # Uncomment as needed
    #PRODUCT_SYSTEM_PROPERTIES += persist.service.adb.enable=1
    #PRODUCT_SYSTEM_PROPERTIES += persist.service.debuggable=1
    #PRODUCT_SYSTEM_PROPERTIES += persist.sys.usb.config=mtp,adb
    PRODUCT_ADB_KEYS := ${./adbkey.pub}
  '';
```

Atemu, 2022-03-23:
Thanks a bunch @zhaofengli but unfortunately, that doesn't seem to work. My system's build.prop doesn't contain those proerties and adb isn't started.

Atemu, 2022-03-24:
That doesn't work either. Any idea?

I'm using LineageOS, maybe that's why.