nix-community/robotnix #178: OTA updates fail during reboot step

mgidding, 2022-04-01:
Hello, I am new to robotnix and I'm trying to use the OTA updater for the first time. I have managed to make the updater download and apply an OTA update. However once I reboot my phone afterwards it will show the android logo for a few minutes, reboot again and then attempt to download the same update once more (without having updated my phone).

mgidding, 2022-04-02:
I have managed to make OTA updates work by removing the new prebuilt app I had included (with privileged = true;).

zhaofengli, 2022-04-02:
You likely triggered the rollback mechanism because the new update failed to boot. For privileged apps, the system will simply refuse to boot if it's not allowed all privileged permissions that it's declared with (see [Privileged Application Allowlisting](https://source.android.com/devices/tech/config/perms-allowlist)).

mgidding, 2022-04-03:
Thank you for the response! It seems that was indeed my issue.

closed 2022-04-03 by mgidding