nix-community/robotnix #187: Supported Lineage build fails - surnia

mannp, 2022-05-18:
I am unable to get a Motorola surnia build to work, while it appears to be a supported device;

https://github.com/danielfullmer/robotnix/tree/master/flavors/lineageos/lineage-18.1)/device-dirs.json

flake.nix

```
      robotnixConfigurations."surnia" = robotnix.lib.robotnixSystem ({ config, pkgs, ... }: {
        imports = [ common ];
        device = "surnia";
        variant = "eng";
        flavor = "lineageos";
        #androidVersion = 11;
      });
```

Error output

```
Which product would you like? [aosp_arm] lineage_surnia
In file included from build/make/core/config.mk:313:
In file included from build/make/core/envsetup.mk:312:
[1mbuild/make/core/product_config.mk:160: [31merror: [0m[1mCan not locate config makefile for product "lineage_surnia".[0m
11:09:27 dumpvars failed with: exit status 1
** Not a valid product: lineage_surnia
```

Device definition

```
  "device/motorola/surnia": {
    "date": "2021-03-20T07:05:33+00:00",
    "deepClone": false,
    "deps": [
      "device/motorola/msm8916-common"
    ],
    "fetchLFS": false,
    "fetchSubmodules": false,
    "leaveDotGit": false,
    "path": "/nix/store/p84pnbp4jqwszyqysxjk6wq590806xml-android_device_motorola_surnia",
    "rev": "625b09265e175f44f39f9c750ab1da5df1ab74a0",
    "sha256": "10sbsd264mpw7r03czzsj49ydigrppxjf5inrbkh95q3daqcfr48",
    "url": "https://github.com/LineageOS/android_device_motorola_surnia"
  },
```

Thanks.

Atemu, 2022-05-18:
Why are you building for `variant = "eng";`? Try the default, I'm not sure LOS supports anything else.

What's in the `common` module?

mannp, 2022-05-18:
It works for my own non-supported builds, but maybe does nothing. 

Did I miss something from the docs that it is not supported?

Anyway, tried with and without it and with and without androidVersion = 11 and 12, none of them build.

Added common to the original post.

closed 2022-11-04 by mannp