nix-community/robotnix #194: LineageOS guacamole build failed - SHA256 mismatch

0xb1b1, 2022-08-15:
A build for guacamole (LineageOS latest) failed with SHA256 hashsum mistatch.

Output:
```
warning: dumping very large path (> 256 MiB); this may run out of memory
error: hash mismatch in fixed-output derivation '/nix/store/c1cyni7vhky10bdny34djwgvxb2pbl4p-android_kernel_oneplus_sm8150-bd9fe82.drv':
         specified: sha256-+O/+7yjpZ3WHVXFmJFep3UHnaHnPsjmz2S6ivDJyc2U=
            got:    sha256-k0Efiy/c4LGps3F2xHs/UWtUMo1DUUzC7/cM8JCdwNI=
error: 1 dependencies of derivation '/nix/store/3ypj5z7mmqm9539a5hajydz6vw5dj4mh-kernel=oneplus=sm8150-patched.drv' failed to build
error: 1 dependencies of derivation '/nix/store/sc1qin03ygsjrpb5sp8xc9amimfg9hrk-unpack.sh.drv' failed to build
error: 1 dependencies of derivation '/nix/store/600jwqia9s0zgaqxr98bvj7vphy2wqxc-robotnix-lineage_guacamole-2022041111.drv' failed to build
error: 1 dependencies of derivation '/nix/store/1flql0vp81axvvqbqsjd6y3hi2wln4pb-guacamole-img-2022041111.zip.drv' failed to build
```

flake.nix:
```
{
  description = "A basic example robotnix configuration";

  inputs.robotnix.url = "github:danielfullmer/robotnix";

  outputs = { self, robotnix }: {
    # "dailydriver" is an arbitrary user-chosen name for this particular
    # configuration.  Change it to something meaningful for you, perhaps just
    # the device name if you only have one of this kind of device.
    robotnixConfigurations."guacamole" = robotnix.lib.robotnixSystem ({ config, pkgs, ... }: {
      # These two are required options
      device = "guacamole";
      flavor = "lineageos"; # "grapheneos" is another option

      # buildDateTime is set by default by the flavor, and is updated when those flavors have new releases.
      # If you make new changes to your build that you want to be pushed by the OTA updater, you should set this yourself.
      # buildDateTime = 1584398664; # Use `date "+%s"` to get the current time

      # signing.enable = true;
      # signing.keyStorePath = "/home/voxel/Builds/Android/guacamole/secrets/android-keys"; # A _string_ of the path for the key store.

      # Build with ccache
      # ccache.enable = true;
    });

    # This provides a convenient output which allows you to build the image by
    # simply running "nix build" on this flake.
    # Build other outputs with (for example): "nix build .#robotnixConfigurations.dailydriver.ota"
    defaultPackage.x86_64-linux = self.robotnixConfigurations."guacamole".img;
  };
}
```

flake.lock:
```
{
  "nodes": {
    "androidPkgs": {
      "inputs": {
        "devshell": "devshell",
        "flake-utils": "flake-utils",
        "nixpkgs": "nixpkgs"
      },
      "locked": {
        "lastModified": 1638562808,
        "narHash": "sha256-nnGyBugMQo9WweTgpfPbJu0fHnRtxvsPQ9el2D3wPrY=",
        "owner": "tadfisher",
        "repo": "android-nixpkgs",
        "rev": "a191ab6adb019b09d3bb919bb98dca31d83519d5",
        "type": "github"
      },
      "original": {
        "owner": "tadfisher",
        "ref": "stable",
        "repo": "android-nixpkgs",
        "type": "github"
      }
    },
    "devshell": {
      "locked": {
        "lastModified": 1637575296,
        "narHash": "sha256-ZY8YR5u8aglZPe27+AJMnPTG6645WuavB+w0xmhTarw=",
        "owner": "numtide",
        "repo": "devshell",
        "rev": "0e56ef21ba1a717169953122c7415fa6a8cd2618",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "devshell",
        "type": "github"
      }
    },
    "flake-utils": {
      "locked": {
        "lastModified": 1638122382,
        "narHash": "sha256-sQzZzAbvKEqN9s0bzWuYmRaA03v40gaJ4+iL1LXjaeI=",
        "owner": "numtide",
        "repo": "flake-utils",
        "rev": "74f7e4319258e287b0f9cb95426c9853b282730b",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "flake-utils",
        "type": "github"
      }
    },
    "nixpkgs": {
      "locked": {
        "lastModified": 1637841632,
        "narHash": "sha256-QYqiKHdda0EOnLGQCHE+GluD/Lq2EJj4hVTooPM55Ic=",
        "owner": "NixOS",
        "repo": "nixpkgs",
        "rev": "73369f8d0864854d1acfa7f1e6217f7d6b6e3fa1",
        "type": "github"
      },
      "original": {
        "owner": "NixOS",
        "ref": "nixos-unstable",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "nixpkgsUnstable": {
      "locked": {
        "lastModified": 1638376152,
        "narHash": "sha256-ucgLpVqhFnClH7YRUHBHnmiOd82RZdFR3XJt36ks5fE=",
        "owner": "NixOS",
        "repo": "nixpkgs",
        "rev": "6daa4a5c045d40e6eae60a3b6e427e8700f1c07f",
        "type": "github"
      },
      "original": {
        "owner": "NixOS",
        "ref": "nixos-unstable",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "nixpkgs_2": {
      "locked": {
        "lastModified": 1638371214,
        "narHash": "sha256-0kE6KhgH7n0vyuX4aUoGsGIQOqjIx2fJavpCWtn73rc=",
        "owner": "NixOS",
        "repo": "nixpkgs",
        "rev": "a640d8394f34714578f3e6335fc767d0755d78f9",
        "type": "github"
      },
      "original": {
        "owner": "NixOS",
        "ref": "nixos-21.11",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "robotnix": {
      "inputs": {
        "androidPkgs": "androidPkgs",
        "nixpkgs": "nixpkgs_2",
        "nixpkgsUnstable": "nixpkgsUnstable"
      },
      "locked": {
        "lastModified": 1650338467,
        "narHash": "sha256-rpRB1WbmIRFDdUKYQGzWnNzwi/5BNQEp718M0on+GSw=",
        "owner": "danielfullmer",
        "repo": "robotnix",
        "rev": "1075210d1e32e075958cf5e760150ed8ce000cd7",
        "type": "github"
      },
      "original": {
        "owner": "danielfullmer",
        "repo": "robotnix",
        "type": "github"
      }
    },
    "root": {
      "inputs": {
        "robotnix": "robotnix"
      }
    }
  },
  "root": "root",
  "version": 7
}
```

P.S.: Thanks for your work on this project! It's purely amazing.

Waelwindows, 2023-01-04:
Can't reproduce, the build worked just fine on my end. However, that could be because i specified the android version.

Here's my flake and lock just in case...
[robotnix.zip](https://github.com/danielfullmer/robotnix/files/10343926/robotnix.zip)

0xb1b1, 2023-02-10:
> Can't reproduce, the build worked just fine on my end. However, that could be because i specified the android version.
> 
> 
> 
> Here's my flake and lock just in case...
> 
> [robotnix.zip](https://github.com/danielfullmer/robotnix/files/10343926/robotnix.zip)
> 
> 

Thanks for the reference and sorry for replying late. Gotta test my config with updated parameters:)

erahhal, 2024-02-09:
I'm seeing the following with the flake you provided @Waelwindows 

```
error: builder for '/nix/store/9smjswxxx0ys9c681kk09zm6rdh6xg76-proprietary_vendor_oneplus-50b0f23.drv' failed with exit code 1;                                                                                                    
       last 10 log lines:                                                                                                                                                                                                           
       > fatal: early EOF                                                                                                                                                                                                           
       > fatal: fetch-pack: invalid index-pack output                                                                                                                                                                               
       > remote: Enumerating objects: 55138, done.                                                                                                                                                                                  
       > remote: Counting objects: 100% (336/336), done.                                                                                                                                                                            
       > remote: Compressing objects: 100% (177/177), done.                                                                                                                                                                         
       > error: 7961 bytes of body are still expectedMiB | 353.00 KiB/s                                                                                                                                                             
       > fetch-pack: unexpected disconnect while reading sideband packet                                                                                                                                                            
       > fatal: early EOF                                                                                                                                                                                                           
       > fatal: fetch-pack: invalid index-pack output                                                                                                                                                                               
       > Unable to checkout 50b0f237798c654c2b6071c10af6b656ea3c36df from https://github.com/TheMuppets/proprietary_vendor_oneplus.                                                                                                 
       For full logs, run 'nix log /nix/store/9smjswxxx0ys9c681kk09zm6rdh6xg76-proprietary_vendor_oneplus-50b0f23.drv'.                                                                                                             
error: 1 dependencies of derivation '/nix/store/pxjibjbm33ri27z0l302nq50zlrrhg3y-unpack.sh.drv' failed to build                                                                                                                     
error: 1 dependencies of derivation '/nix/store/3b4p2f77rmyaq02x812kciswfxc4gqfj-robotnix-lineage_guacamole-2022041111.drv' failed to build                                                                                         
error: 1 dependencies of derivation '/nix/store/9d92l0ds8lml6qlphh2rvbspnws701mx-guacamole-img-2022041111.zip-env.drv' failed to build   
```

erahhal, 2024-02-09:
@Waelwindows I see this when I build with your flake:

```
[12/21/96 built, 0.0 MiB DL] building tradefederation: Initialized empty Git repository in /nix/store/3j9grykgin8x                                                                                                                  
error: builder for '/nix/store/9smjswxxx0ys9c681kk09zm6rdh6xg76-proprietary_vendor_oneplus-50b0f23.drv' failed with exit code 1;                                                                                                    
       last 10 log lines:                                                                                                                                                                                                           
       > fatal: early EOF                                                                                                                                                                                                           
       > fatal: fetch-pack: invalid index-pack output                                                                                                                                                                               
       > remote: Enumerating objects: 55138, done.                                                                                                                                                                                  
       > remote: Counting objects: 100% (336/336), done.                                                                                                                                                                            
       > remote: Compressing objects: 100% (177/177), done.                                                                                                                                                                         
       > error: 3824 bytes of body are still expected MiB | 368.00 KiB/s                                                                                                                                                            
       > fetch-pack: unexpected disconnect while reading sideband packet                                                                                                                                                            
       > fatal: early EOF                                                                                                                                                                                                           
       > fatal: fetch-pack: invalid index-pack output                                                                                                                                                                               
       > Unable to checkout 50b0f237798c654c2b6071c10af6b656ea3c36df from https://github.com/TheMuppets/proprietary_vendor_oneplus.                                                                                                 
       For full logs, run 'nix log /nix/store/9smjswxxx0ys9c681kk09zm6rdh6xg76-proprietary_vendor_oneplus-50b0f23.drv'.                                                                                                             
error: 1 dependencies of derivation '/nix/store/pxjibjbm33ri27z0l302nq50zlrrhg3y-unpack.sh.drv' failed to build                                                                                                                     
error: 1 dependencies of derivation '/nix/store/isxqwdkzsi2rjhn1qxw49xi7pkfba79w-robotnix-lineage_guacamole-2022041111.drv' failed to build                                                                                         
error: 1 dependencies of derivation '/nix/store/rdyrijb9ssmfv6ls9ry51x82qn85c1pj-guacamole-img-2022041111.zip.drv' failed to build  

```

erahhal, 2024-02-10:
Other problems though:

1. `android-info.txt` is wrong in the generated image. It has `board=msmnile` instead of `board=msm8996`
2. Flashing results in the following:

```
Warning: skip copying boot image avb footer (boot partition size: 67108864, boot image size: 100663296).                                                                                                                                                                                                                                                                     
Sending 'boot' (98304 KB)                          OKAY [  4.264s]                                                                                                                                                                                                                                                                                                           
Writing 'boot'                                     FAILED (remote: 'size too large')     
```