nix-community/robotnix #197: Running out of space during chromium build

phodina, 2022-10-31:
Hello,

I'm running out of space during build of chromium. My tmpfs is 16G. Is it possible to extract the files somewhere else than `/tmp`?

I have 32 GB RAM machine. Is it enough to build the image?

```
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/details-selection.js': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/global-timeline-template.html': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/global-timeline.js': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/helper.js': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/index.html': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/model.js': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/trace-file-reader-template.html': No space left on device
cp: error writing 'chromium-98.0.4758.101-src/src/v8/tools/zone-stats/trace-file-reader.js': No space left on device
do not know how to unpack source archive /nix/store/06khvgx4b1j72kx0nddi7dpv7ai6si20-chromium-98.0.4758.101-src
error: builder for '/nix/store/ydma8aa69i32wd8nw7gsy30dlrn56kka-chromium-98.0.4758.101.drv' failed with exit code 1
       note: build failure may have been caused by lack of free disk space
error: 1 dependencies of derivation '/nix/store/rg0n21z2b3r5xl97l48bajf2s5kvpfyc-aab-universal.apk.drv' failed to build
error: 1 dependencies of derivation '/nix/store/nck3svirx7s4c2jcn12sf3qgxldjm02p-chromium-webview-trichrome-patched.apk.drv' failed to build
error: 1 dependencies of derivation '/nix/store/7ahri8s7vapgz0ajfdpxv8vmshcaix2z-prebuilt_chromiumTrichromeLibrary.drv' failed to build
error: 1 dependencies of derivation '/nix/store/hfhllmgbzrydd8cgchkrhxpnx5idgcyk-unpack.sh.drv' failed to build
error: 1 dependencies of derivation '/nix/store/i6an0cn6q687cxcshh6z2acfhhwibf1p-robotnix-aosp_crosshatch-2022022022.drv' failed to build
error: 1 dependencies of derivation '/nix/store/sdcd5aky5d07jwv10vlpmkpfpnk7y9fa-crosshatch-img-2022022022.zip.drv' failed to build
```

Atemu, 2022-10-31:
Set `$TMPDIR` to somewhere else. Note that for regular user builds, that variable needs to be applied to the daemon, not the nix-buildprocess.

phodina, 2022-10-31:
Thanks. I got same hint on Nix matrix channel. This helps as the build (chromium) now continues.

closed 2022-10-31 by phodina

Atemu, 2022-10-31:
FYI: https://app.element.io/#/room/#robotnix:nixos.org