nix-community/robotnix #199: Emulator is missing a kernel file

phodina, 2022-11-08:
Hello,

I've built the x86_64 emulator image with following command:
```
nix-build --arg configuration '{ device="x86_64"; flavor="vanilla"; }' -A emulator
```

But when I run the `./result` I get this error:
```
$ ./result
INFO    | Android emulator version 31.1.4.0 (build_id 7920983) (CL:N/A)
ERROR   | This AVD's configuration is missing a kernel file! Please ensure the file "kernel-ranchu" is in the same location as your system image.
ERROR   | ANDROID_SDK_ROOT is defined (/nix/store/04x8sf67kc6w1n813r3i11kgla07mw78-fake-sdk) but cannot find kernel file in /nix/store/04x8sf67kc6w1n813r3i11kgla07mw78-fake-sdk/system-images/ sub directories
```

I'm on branch `commit c8a71bcd1236fad6ae7dea846bab37b9f4c7a641 (HEAD, tag: vanilla-2022022022, tag: grapheneos-2022021721`

Any idea what I'm doing wrong or how to fix the error?

phodina, 2022-11-18:
Hi @danielfullmer,

could you please help with this cryptic error?

c2vi, 2024-05-01:
Hi,

I had the same issue.

This nix command in the AOSP build system runns something like `lunch aosp_x86_64-eng`, which does not build the necessary files for an emulator. with `lunch sdk_x86_64-eng` it does.

So setting the productNamePrefix = "sdk_"; option in the config would solve this issue.

Your cmd would then be:
```
nix-build --arg configuration '{ device="x86_64"; flavor="vanilla"; productNamePrefix = "sdk_"; }' -A emulator
```

Shouldn't the emulator output set the productNamePrefix always to "sdk_"? Has a robotnix built emulator ever worked like it is right now?

RankoR, 2024-06-27:
Facing the same while building AOSP (manually on Arch, not with this project).

`So setting the productNamePrefix = "sdk_"; option in the config would solve this issue.`

Seems that it doesn't work anymore for AOSP 14 (or something changed). There are only car-related `sdk` choices, for example `sdk_car_arm64-trunk_staging-userdebug`.  And `aosp_x86_64-trunk_staging-eng` doesn't build kernel.