nix-community/robotnix #204: Can not locate config makefile for product "lineage_dre"

Gooberpatrol66, 2023-04-08:
this is the config i'm using to build an ota update: 
```
{
 device = "dre";
 flavor = "lineageos";
 variant = "user";
 signing.enable = true;
 signing.keyStorePath = "/home/nathan/guix-configs/robotnix/keys";
 apps.fdroid.enable = true;
}
```
"dre" is a valid lineageos device and there are builds available for it: https://download.lineageos.org/devices/dre/builds

```
building '/nix/store/z0j22l5nbi9l9m649hkgvna4xssfk7mp-robotnix-lineage_dre-2022041111.drv'...
unpacking sources
patching sources
building
Build type choices are:
     1. release
     2. debug

Which would you like? [1] release

Which product would you like? [aosp_arm] lineage_dre
In file included from build/make/core/config.mk:313:
In file included from build/make/core/envsetup.mk:312:
build/make/core/product_config.mk:160: error: Can not locate config makefile for product "lineage_dre".
14:13:35 dumpvars failed with: exit status 1
** Not a valid product: lineage_dre

Variant choices are:
     1. user
     2. userdebug
     3. eng
Which would you like? [eng] user

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=12
TARGET_PRODUCT=aosp_arm
TARGET_BUILD_VARIANT=user
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm
TARGET_ARCH_VARIANT=armv7-a-neon
TARGET_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-6.1.19-gentoo-x86_64-x86_64
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=SP2A.220405.004
OUT_DIR=out
PRODUCT_SOONG_NAMESPACES=device/generic/goldfish device/generic/goldfish-opengl hardware/google/camera hardware/google/camera/devices/EmulatedCamera device/generic/goldfish device/generic/goldfish-opengl packages/apps/Bluetooth
============================================
error: builder for '/nix/store/z0j22l5nbi9l9m649hkgvna4xssfk7mp-robotnix-lineage_dre-2022041111.drv' failed with exit code 1;
       last 10 log lines:
       > HOST_OS=linux
       > HOST_OS_EXTRA=Linux-6.1.19-gentoo-x86_64-x86_64
       > HOST_CROSS_OS=windows
       > HOST_CROSS_ARCH=x86
       > HOST_CROSS_2ND_ARCH=x86_64
       > HOST_BUILD_TYPE=release
       > BUILD_ID=SP2A.220405.004
       > OUT_DIR=out
       > PRODUCT_SOONG_NAMESPACES=device/generic/goldfish device/generic/goldfish-opengl hardware/google/camera hardware/google/camera/devices/EmulatedCamera device/generic/goldfish device/generic/goldfish-opengl packages/apps/Bluetooth
       > ============================================
       For full logs, run 'nix log /nix/store/z0j22l5nbi9l9m649hkgvna4xssfk7mp-robotnix-lineage_dre-2022041111.drv'.
error: 1 dependencies of derivation '/nix/store/n82j5wk7y2jip82xcxv8b6lzwbxyhcj0-dre-signed_target_files-2022041111.zip.drv' failed to build
error: 1 dependencies of derivation '/nix/store/skzifxprrg0c8lksyp6dswniw2fs4q25-dre-ota_update-2022041111.zip.drv' failed to build

```

Atemu, 2023-04-09:
LOS is pretty out of date and `dre` wasn't in LOS many moons ago. Try https://github.com/danielfullmer/robotnix/pull/190.

Gooberpatrol66, 2023-04-12:
> LOS is pretty out of date and `dre` wasn't in LOS many moons ago. Try #190.

I get this error with your fork.

```
building '/nix/store/qrdali40q3652148g9xgv5xgbwlz24ll-dre-signed_target_files-2022123019.zip.drv'...
Missing KEYSDIR directory, did you use "--option extra-sandbox-paths /keys=..." ?

```
I don't know if this is an upstream error or an error with your branch or a config error or what.

Atemu, 2023-04-13:
This is related to signing. Have you set it up properly? I don't sign my images, so I can't help you with that. I only know that it works without signing.